Ysurac/openmptcprouter #579: Upload saturate RAM on router

JorgeMoort89, 2019-07-30:
When I do a lot of uploads (backup some servers, FTP, ecc.) the RAM on the router gradually is used causing at some point high latency and low speed on aggregation. I monitor it via SNMP and in that moment even that service, and all the natted service, stop working.
In this case the router have 4GB of RAM, if I do `htop` or `top` via SSH I don't see any process using that amount of memory. I've also tried with newest and older version, same issue.

[image]

## Specifications

  - OpenMPTCProuter version: 0.49
  - OpenMPTCProuter VPS version: 0.996
  - OpenMPTCProuter platform: x86_64

JorgeMoort89, 2019-08-02:
Same problem with version 0.51.5 on the router and version 0.999 on the VPS

Ysurac, 2019-08-02:
I may have too high network buffer, I will make some tests.

JorgeMoort89, 2019-08-02:
This cause sometimes blocks on DNS resolution. I have to refresh one or two times the web page to have the domain resolved.

closed 2019-12-20 by Ysurac