Ysurac/openmptcprouter #580: Redirection port pour serveur mail et d'autres ports

Slivikk-fr, 2019-07-30:
## Actual Behavior
Bonjour,

J'ai un serveur mail qui n'a pas l'air de fonctionner pourtant j'ai les ports qui sont ouvert (en principe) avec les port 19999 et 10000 sur mon autre VPS (Sur mon serveur) comment je peut check ça ?

## Specifications

  - OpenMPTCProuter version: (0.51.4)
  - OpenMPTCProuter VPS version: (0.999)
  - OpenMPTCProuter platform: (RPI3B+)

Ysurac, 2019-07-30:
la doc https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding a été suivi ? Notamment l'ouverture des ports sur le VPS ?

Slivikk-fr, 2019-07-30:
Point de vu VPS rien n'a été touché depuis le système le port pour le serveur web c'est bon

Slivikk-fr, 2019-07-30:
[image]
[image]

Slivikk-fr, 2019-07-30:
Point de vu port 10000 et 19999 c'est bon

Slivikk-fr, 2019-07-30:
C'est bon j'ai réussi finalement :D Merci quand même

closed 2019-07-30 by Slivikk-fr