Ysurac/openmptcprouter #581: port forwarding broken when adding static route

pyjabox, 2019-07-30:
## Expected Behavior

In previous version (0.49), having a port-forwarding was working properly along with a static route

## Actual Behavior
When activating port-forwarding alone, it works
When activating port-forwarding, and then a static route, port-forwarding stops working and static route works
When reloading after that, port-forwarding works again, but static route stops working

## Steps to Reproduce the Problem

  1. see above

## Specifications

  - OpenMPTCProuter version: 0.51.4 (and 0.51.3 as well)
  - OpenMPTCProuter VPS version: 0.999
  - OpenMPTCProuter platform: x86_64

Ysurac, 2019-07-31:
Do you see your route using ```ip route``` via SSH on the router ?

pyjabox, 2019-07-31:
Here is what look like the routing table after a reload (in that state = portforwarding works / static route doesn't work) 

default via 10.255.255.1 dev tun0
default via 192.168.9.254 dev eth1 metric 3
default via 192.168.105.254 dev eth2 metric 4
10.255.255.1 dev tun0 proto kernel scope link src 10.255.255.2
51.91.59.138 via 192.168.9.254 dev eth1 metric 1
51.91.59.138 via 192.168.9.254 dev eth1 metric 3
51.91.59.138 via 192.168.105.254 dev eth2 metric 4
192.168.8.0 via 192.168.105.254 dev eth2
192.168.9.0/24 dev eth1 scope link metric 3
192.168.105.0/24 dev eth2 scope link metric 4

The static route I added before reloading is "192.168.8.0/24 pointing to 192.168.105.254"

But actually the route above has no mask. Then I realized the following :

When I add the static route, the route looks like this in "ip route"
192.168.8.0/24 via 192.168.105.254 dev eth2 proto static metric 4

After a reload, this becomes
192.168.8.0 via 192.168.105.254 dev eth2

As you can see the mask disappears. I just tried to use a /32 route pointing to the server I want to reach instead of /24, and now the route survives a reload because it is a /32 :
before reload 
192.168.8.1 via 192.168.105.254 dev eth2
after reload
192.168.8.1 via 192.168.105.254 dev eth2

closed 2019-07-31 by Ysurac

pyjabox, 2019-07-31:
Why have you closed the issue ?
Expected to change the mask from 24 to 32 after a reload ?

Ysurac, 2019-07-31:
As github said, it's fixed in commit https://github.com/Ysurac/openmptcprouter-feeds/commit/589cfe484c031946c7edeeb5eb1098c2ba552532
So fix will be in next release.

pyjabox, 2019-07-31:
> As github said, it's fixed in commit Ysurac/openmptcprouter-feeds@589cfe4<https://github.com/Ysurac/openmptcprouter-feeds/commit/589cfe484c031946c7edeeb5eb1098c2ba552532>

Ack, thanks