Ysurac/openmptcprouter #583: Remote recv: Connection reset by peer

Randdrick, 2019-08-01:
## Expected Behavior

  VPS stable sans problème de débit

## Actual Behavior
Ce message apparaît très fréquemment : remote recv: Connection reset by peer

## Steps to Reproduce the Problem

Ce problème n'est pas facile à reproduire, mais le message arrive dès qu'il y a un peu de consommation de bande passante (ex : streaming vidéo). 
Il est très visible et régulier depuis la mise à jour du VPS à la version 0.997, et ce quelque ce soit la version du client. (J'ai fais le test en revenant à la version 0.48)

## Specifications

  - OpenMPTCProuter version: 0.51.5
  - OpenMPTCProuter VPS version: 0.999
  - OpenMPTCProuter platform: RPI3

## Note complémentaire  : 
Le problème débit signalé dans un autre post, et ce message d'erreur me semble lié, ou du moins semble lié à la dégradation des performances des shadowsocks du VPS.

Ysurac, 2019-08-01:
L'"erreur" ici signifie que le serveur distant a fermé la connexion TCP avec un RESET, ça peut être du à un timeout ou un soucis/limitation réseau niveau VPS.

Randdrick, 2019-08-01:
J'y ai pensé, cependant je n'ai pas changé de VPS. De plus, je dispose d'une bande passante de 200 mb/s pour mon VPS alors que ma connexion ne dépasse pas les 100 mb/s.

J'ai pensé à un soucis sur mon VPS, mais qui se répète depuis la version 0.997, je me pose la question de savoir si il n'y a pas autre chose qu'une coïncidence. De plus, ce n'est pas une erreur de temps à autre, c'est par paquets.

Ex de logs:
Thu Aug  1 20:16:47 2019 daemon.err /usr/bin/ss-redir[17269]: send: Connection reset by peer
Thu Aug  1 20:16:51 2019 daemon.err /usr/bin/ss-redir[17268]: send: Connection reset by peer
Thu Aug  1 20:16:53 2019 daemon.err /usr/bin/ss-redir[17269]: send: Connection reset by peer
Thu Aug  1 20:16:58 2019 daemon.err /usr/bin/ss-redir[17266]: send: Connection reset by peer
Thu Aug  1 20:17:01 2019 daemon.err /usr/bin/ss-redir[17267]: send: Connection reset by peer
Thu Aug  1 20:17:02 2019 daemon.err /usr/bin/ss-redir[17267]: send: Connection reset by peer
Thu Aug  1 20:17:04 2019 daemon.err /usr/bin/ss-redir[17268]: send: Connection reset by peer
Thu Aug  1 20:17:04 2019 daemon.err /usr/bin/ss-redir[17268]: send: Connection reset by peer
Thu Aug  1 20:17:09 2019 daemon.err /usr/bin/ss-redir[17269]: send: Connection reset by peer
Thu Aug  1 20:17:15 2019 daemon.err /usr/bin/ss-redir[17266]: send: Connection reset by peer
Thu Aug  1 20:17:26 2019 daemon.err /usr/bin/ss-redir[17266]: send: Connection reset by peer
Thu Aug  1 20:17:30 2019 daemon.err /usr/bin/ss-redir[17266]: send: Connection reset by peer
Thu Aug  1 20:17:43 2019 daemon.err /usr/bin/ss-redir[17267]: send: Connection reset by peer

bambuca, 2019-08-01:
Please write in English so even non-French speaking people know what's the issue and don't make duplicates.

Randdrick, 2019-08-01:
In english : 

## Expected Behavior
Stable VPS with no flow problem

## Actual Behavior
This message appears frequently: remote recv: Connection reset by peer

## Steps to Reproduce the Problem
This problem is not easy to reproduce, but the message arrives as soon as there is a bit of bandwidth consumption (eg video streaming).
It is very visible and regular since the update of the VPS to version 0.997, and this is the version of the client. (I did the test by returning to version 0.48)

## Specifications
OpenMPTCProuter version: 0.51.5
OpenMPTCProuter VPS version: 0.999
OpenMPTCProuter platform: RPI3

## Additional note:
The flow problem reported in another post, and this error message seems to me related, or at least seems related to the performance degradation of the shadowsocks of the VPS.

MacGeyer, 2019-08-04:
Same problem here

OpenMPTCProuter version: 0.51.5
OpenMPTCProuter VPS version: 0.999
OpenMPTCProuter platform: ESXi

"ERROR: remote recv: Connection reset by peer" 360 times in 10 hours !

Ysurac, 2019-08-04:
Do you use P2P or anything that can open many connections ?

MacGeyer, 2019-08-04:
No P2P.
Today after 24h, I got 800 "ERROR: remote recv: Connection reset by peer" 360 times in 10 hours !", for 5 GB data ( most is www.npef.com )

xabix, 2019-09-16:
Hello I have the same issue running 0.52 with 1001.

> Sep 16 21:06:49 XabiX ss-server[699]:  2019-09-16 21:06:49 ERROR: server recv: Connection reset by peer
Sep 16 21:06:49 XabiX ss-server[700]:  2019-09-16 21:06:49 ERROR: server recv: Connection reset by peer
Sep 16 21:06:49 XabiX ss-server[700]:  2019-09-16 21:06:49 ERROR: server recv: Connection reset by peer
Sep 16 21:06:49 XabiX ss-server[699]:  2019-09-16 21:06:49 ERROR: server recv: Connection reset by peer
Sep 16 21:06:49 XabiX ss-server[700]:  2019-09-16 21:06:49 ERROR: server recv: Connection reset by peer
Sep 16 21:06:49 XabiX ss-server[700]:  2019-09-16 21:06:49 ERROR: server recv: Connection reset by peer
Sep 16 21:06:49 XabiX ss-server[699]:  2019-09-16 21:06:49 ERROR: server recv: Connection reset by peer
Sep 16 21:06:50 XabiX ss-server[700]:  2019-09-16 21:06:50 ERROR: server recv: Connection reset by peer
Sep 16 21:06:50 XabiX ss-server[700]:  2019-09-16 21:06:50 ERROR: server recv: Connection reset by peer
Sep 16 21:06:50 XabiX ss-server[700]:  2019-09-16 21:06:50 ERROR: server recv: Connection reset by peer

closed 2019-12-20 by Ysurac

CaptPicardFR, 2020-05-15:
> J'y ai pensé, cependant je n'ai pas changé de VPS. De plus, je dispose d'une bande passante de 200 mb/s pour mon VPS alors que ma connexion ne dépasse pas les 100 mb/s.
>
> J'ai pensé à un soucis sur [...]

Randdrick j'ai le même soucis avec la dernière version dispos, tu as une idée de comment résoudre ça ?

Randdrick, 2020-05-15:
Je pense qu'il s'agit d'un problème lié au serveur sur lequel se trouve ta machine virtuelle.
Si tu es chez firstheberg, demandes leur de migrer ton vps sur une autre machine. 
Après la migration, reboot ton VPS car ils font ça un peu à la sauvage.

CaptPicardFR, 2020-05-15:
> 
> 
> Je pense qu'il s'agit d'un problème lié au serveur sur lequel se trouve ta machine virtuelle.
> Si tu es chez firstheberg, demandes leur de migrer ton vps sur une autre machine.
> Après la migration, reboot ton VPS car ils font ça un peu à la sauvage.

Tu as solutionné ton problème toi ? En changeant de VPS ? 
Je suis chez OVH sur les nouveaux VPS c'est peut être ça le soucis.