Ysurac/openmptcprouter #584: Change server via CLI

JorgeMoort89, 2019-08-02:
Hi Ysurac,
I've only SSH access to a router in my house and I'll be not here for a while.
How can I change VPS IP and key by CLI or NAT the web GUI via SSH?

## Specifications

  - OpenMPTCProuter version: 0.49
  - OpenMPTCProuter VPS version: 0.996
  - OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64)

Ysurac, 2019-08-02:
For now it's not a good idea to try to change VPS IP over cli.

To use the web interface over SSH:
```ssh -L 15080:127.0.0.1:80 ipoftherouter```
Then website is available over _http://127.0.0.1:15080/_ on your local computer.

JorgeMoort89, 2019-08-02:
Thanks a lot!
Instead if I wanted to do a NAT services via CLI anyway?
I'm going to nat a lot o services, it will be so much faster

Ysurac, 2019-08-02:
All is available using ```uci```, you can check OpenWrt doc about that.

JorgeMoort89, 2019-08-02:
Thanks, I'll check on it.

closed 2019-08-02 by JorgeMoort89