Ysurac/openmptcprouter #585: Warning Kernel

Randdrick, 2019-08-02:
## Expected Behavior

  No errors or Warning Kernel

## Actual Behavior

## Steps to Reproduce the Problem

This seems to happen when there is a problem with the aggregation of internet connections.
This problem has occurred since version 0.51.
It exists in EXT4 and SquashFs

## Specifications

  - OpenMPTCProuter version: 0.51.5
  - OpenMPTCProuter VPS version: 0.999
  - OpenMPTCProuter platform: RPI3

## Additional note
[15546.573657] ------------[ cut here ]------------
[15546.579936] refcount_t: underflow; use-after-free.
[15546.586364] WARNING: CPU: 2 PID: 0 at lib/refcount.c:187 refcount_sub_and_test_checked+0x58/0xb0
[15546.598226] Modules linked in: rtl8192cu rtl8192c_common rtl_usb rt2800usb rt2800lib qcserial option ipw brcmfmac usb_wwan usb_serial_simple ti_usb_3410_5052 sierra rtlwifi rtl8xxxu rt2x00usb rt2x00lib pppoe ppp_async pl2303 oti6858 nf_nat_pptp nf_conntrack_pptp mt7601u mos7720 mmc_spi mct_u232 mac80211 keyspan iptable_nat ipt_REJECT ipt_MASQUERADE garmin_gps ftdi_sio ebtable_nat ebtable_filter ebtable_broute cypress_m8 cp210x ch341 cfg80211 cdc_mbim belkin_sa bas_gigaset ark3116 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_owner xt_ndpi xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iface xt_hl xt_helper xt_hashlimit xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_condition xt_comment xt_addrtype xt_TRACE xt_TPROXY xt_TCPMSS
[15546.686320]  xt_REDIRECT xt_LOG xt_IPMARK xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY xt_ACCOUNT visor usbserial usbhid usb_gigaset ts_fsm ts_bm sr9700 sierra_net ser_gigaset sch_cake rtl8150 rndis_host r8712u(C) r8152 qmi_wwan pppox ppp_generic pegasus of_mmc_spi nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_proto_gre nf_nat_irc nf_nat_ipv4 nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount mcs7830 macvlan kaweth kalmia iptable_raw iptable_mangle iptable_filter
[15546.776026]  ipt_ECN ipheth ip6table_raw ip_tables huawei_cdc_ncm hso hid_generic gigaset ezusb ethoc ebtables ebt_vlan ebt_stp ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_among ebt_802_3 dm9601 crc7 crc_itu_t crc_ccitt compat_xtables compat cdc_wdm cdc_subset cdc_ncm cdc_ether cdc_eem cdc_acm brcmutil ax88179_178a asn1_decoder asix arptable_filter arpt_mangle arp_tables sch_teql sch_sfq sch_red sch_prio sch_pie sch_multiq sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp act_simple act_police act_pedit act_ipt act_gact act_csum sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred snd_bcm2835(C) hid evdev spi_ks8995 xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport
[15546.865981]  ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat ip6t_NPT ip6t_MASQUERADE nf_nat_ipv6 nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 isdn_bsdcomp dss1_divert isdn slhc ifb sit sctp libcrc32c tunnel4 ip_tunnel veth tun snd_rawmidi snd_seq_device snd_pcm_oss snd_pcm_dmaengine snd_pcm snd_timer snd_mixer_oss snd_hwdep snd_compress snd soundcore mISDN_dsp l1oip mISDN_core nls_utf8 sha256_generic seqiv jitterentropy_rng drbg md5 hmac ghash_generic gf128mul gcm ctr vfat fat nls_iso8859_1
[15546.956645]  nls_cp437 xhci_plat_hcd xhci_hcd ehci_platform ehci_hcd gpio_button_hotplug ptp broadcom bcm_phy_lib pps_core
[15546.971614] CPU: 2 PID: 0 Comm: swapper/2 Tainted: G         C        4.19.57 #0
[15546.982771] Hardware name: Raspberry Pi 3 Model B Rev 1.2 (DT)
[15546.990511] pstate: 40000005 (nZcv daif -PAN -UAO)
[15546.997151] pc : refcount_sub_and_test_checked+0x58/0xb0
[15547.004291] lr : refcount_sub_and_test_checked+0x58/0xb0
[15547.011358] sp : ffffff8008013b40
[15547.016328] x29: ffffff8008013b40 x28: ffffffc03a3b4a38 
[15547.023305] x27: 0000000000000000 x26: 0000000000010600 
[15547.030236] x25: 0000000000000001 x24: ffffffc033ee96df 
[15547.037108] x23: ffffff80088e39c0 x22: ffffffc0329bb480 
[15547.043922] x21: ffffffc033e2e7b8 x20: ffffff80088e8000 
[15547.050674] x19: 0000000000000000 x18: 0000000000000000 
[15547.057387] x17: 0000000000000000 x16: 0000000000000000 
[15547.064019] x15: 0000000000000001 x14: 0720072007200720 
[15547.070590] x13: 0720072007200720 x12: ffffff8009309dd8 
[15547.077111] x11: 00000000ffffffff x10: 0000000000000001 
[15547.083582] x9 : 0000000000000008 x8 : 0000000000000000 
[15547.089988] x7 : 0000000000000008 x6 : 0000000000000269 
[15547.096320] x5 : 0000000000000000 x4 : 0000000000000000 
[15547.102576] x3 : 0000000000000000 x2 : ffffffc03a3b0e98 
[15547.108794] x1 : 0000004031bad000 x0 : 0000000000000026 
[15547.114974] Call trace:
[15547.118194]  refcount_sub_and_test_checked+0x58/0xb0
[15547.123993]  refcount_dec_and_test_checked+0x14/0x20
[15547.129755]  mptcp_mpcb_put+0x18/0x58
[15547.134212]  mptcp_sock_destruct+0xb8/0x118
[15547.139187]  __sk_destruct+0x2c/0x150
[15547.143633]  sk_destruct+0x38/0x40
[15547.147800]  __sk_free+0x30/0x40
[15547.151792]  sk_free+0x28/0x38
[15547.155598]  tcp_v4_rcv+0x894/0xe20
[15547.159851]  ip_local_deliver_finish+0x130/0x1c8
[15547.165263]  ip_local_deliver+0xb0/0xc8
[15547.169894]  ip_rcv_finish+0x40/0x60
[15547.174263]  ip_rcv+0x74/0x98
[15547.177998]  __netif_receive_skb_one_core+0x3c/0x48
[15547.183700]  __netif_receive_skb+0x6c/0x78
[15547.188605]  process_backlog+0xa8/0x170
[15547.193266]  net_rx_action+0x110/0x298
[15547.197857]  __do_softirq+0xe0/0x1dc
[15547.202277]  irq_exit+0x60/0xb8
[15547.206251]  handle_IPI+0xac/0x140
[15547.210467]  bcm2836_arm_irqchip_handle_irq+0x8c/0xb8
[... 31 lines cut ...]
[59436.352307] x3 : 00000000ffffb7f1 x2 : 00000000c64a2f38 
[59436.358530] x1 : 0000000000000001 x0 : 0000000000000001 
[59436.364708] Call trace:
[59436.367930]  tcp_data_queue+0x1e0/0x408
[59436.372551]  tcp_rcv_established+0x480/0x548
[59436.377612]  tcp_v4_do_rcv+0xec/0x208
[59436.382071]  tcp_v4_rcv+0x854/0xe20
[59436.386343]  ip_local_deliver_finish+0x130/0x1c8
[59436.391783]  ip_local_deliver+0xb0/0xc8
[59436.396406]  ip_rcv_finish+0x40/0x60
[59436.400763]  ip_rcv+0x74/0x98
[59436.404484]  __netif_receive_skb_one_core+0x3c/0x48
[59436.410180]  __netif_receive_skb+0x6c/0x78
[59436.415068]  process_backlog+0xa8/0x170
[59436.419700]  net_rx_action+0x110/0x298
[59436.424259]  __do_softirq+0xe0/0x1dc
[59436.428654]  irq_exit+0x60/0xb8
[59436.432616]  handle_IPI+0xac/0x140
[59436.436839]  bcm2836_arm_irqchip_handle_irq+0x8c/0xb8
[59436.442776]  el1_irq+0xb0/0x140
[59436.446772]  arch_cpu_idle+0x10/0x18
[59436.451209]  do_idle+0x10c/0x228
[59436.455290]  cpu_startup_entry+0x20/0x28
[59436.460067]  secondary_start_kernel+0x1ac/0x1c0
[59436.465475] ---[ end trace 183c9833659a11aa ]---

user747, 2019-09-03:
Seeing the same on openmptcprouter v0.51.5 r0+10518-6d61fcfa9e with rpi 3+

`Sun Sep  1 19:30:19 2019 kern.warn kernel: [138620.015474] ------------[ cut here ]------------
Sun Sep  1 19:30:19 2019 kern.warn kernel: [138620.021209] refcount_t: underflow; use-after-free.
Sun Sep  1 19:30:19 2019 kern.warn kernel: [138620.027085] WARNING: CPU: 2 PID: 0 at lib/refcount.c:187 refcount_sub_and_test_checked+0x58/0xb0
Sun Sep  1 19:30:19 2019 kern.warn kernel: [138620.037937] Modules linked in: rtl8192cu rtl8192c_common rtl_usb rt2800usb rt2800lib qcserial option ipw brcmfmac usb_wwan usb_serial_simple ti_usb_3410_5052 sierra rtlwifi rtl8xxxu rt2x00usb rt2x00lib pppoe ppp_async pl2303 oti6858 nf_nat_pptp nf_conntrack_pptp mt7601u mos7720 mmc_spi mct_u232 mac80211 keyspan iptable_nat ipt_REJECT ipt_MASQUERADE garmin_gps ftdi_sio ebtable_nat ebtable_filter ebtable_broute cypress_m8 cp210x ch341 cfg80211 cdc_mbim belkin_sa bas_gigaset ark3116 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_owner xt_ndpi xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iface xt_hl xt_helper xt_hashlimit xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_condition xt_comment xt_addrtype xt_TRACE xt_TPROXY xt_TCPMSS
Sun Sep  1 19:30:19 2019 kern.warn kernel: [138620.120824]  xt_REDIRECT xt_LOG xt_IPMARK xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY xt_ACCOUNT visor usbserial usbhid usb_gigaset ts_fsm ts_bm sr9700 sierra_net ser_gigaset sch_cake rtl8150 rndis_host r8712u(C) r8152 qmi_wwan pppox ppp_generic pegasus of_mmc_spi nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_proto_gre nf_nat_irc nf_nat_ipv4 nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount mcs7830 macvlan kaweth kalmia iptable_raw iptable_mangle iptable_filter
Sun Sep  1 19:30:19 2019 kern.warn kernel: [138620.206575]  ipt_ECN ipheth ip6table_raw ip_tables huawei_cdc_ncm hso hid_generic gigaset ezusb ethoc ebtables ebt_vlan ebt_stp ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_among ebt_802_3 dm9601 crc7 crc_itu_t crc_ccitt compat_xtables compat cdc_wdm cdc_subset cdc_ncm cdc_ether cdc_eem cdc_acm brcmutil ax88179_178a asn1_decoder asix arptable_filter arpt_mangle arp_tables sch_teql sch_sfq sch_red sch_prio sch_pie sch_multiq sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp act_simple act_police act_pedit act_ipt act_gact act_csum sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred snd_bcm2835(C) hid evdev spi_ks8995 xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport
Sun Sep  1 19:30:19 2019 kern.warn kernel: [138620.292917]  ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat ip6t_NPT ip6t_MASQUERADE nf_nat_ipv6 nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 isdn_bsdcomp dss1_divert isdn slhc ifb sit sctp libcrc32c tunnel4 ip_tunnel veth tun snd_rawmidi snd_seq_device snd_pcm_oss snd_pcm_dmaengine snd_pcm snd_timer snd_mixer_oss snd_hwdep snd_compress snd soundcore mISDN_dsp l1oip mISDN_core nls_utf8 sha256_generic seqiv jitterentropy_rng drbg md5 hmac ghash_generic gf128mul gcm ctr vfat fat nls_iso8859_1
Sun Sep  1 19:30:19 2019 kern.warn kernel: [138620.379920]  nls_cp437 xhci_plat_hcd xhci_hcd ehci_platform ehci_hcd gpio_button_hotplug ptp broadcom bcm_phy_lib pps_core
Sun Sep  1 19:30:19 2019 kern.warn kernel: [138620.394278] CPU: 2 PID: 0 Comm: swapper/2 Tainted: G         C        4.19.57 #0
Sun Sep  1 19:30:19 2019 kern.warn kernel: [138620.404883] Hardware name: Raspberry Pi 3 Model B Plus Rev 1.3 (DT)
Sun Sep  1 19:30:19 2019 kern.warn kernel: [138620.412796] pstate: 40000005 (nZcv daif -PAN -UAO)
Sun Sep  1 19:30:19 2019 kern.warn kernel: [138620.419208] pc : refcount_sub_and_test_checked+0x58/0xb0
Sun Sep  1 19:30:19 2019 kern.warn kernel: [138620.426110] lr : refcount_sub_and_test_checked+0x58/0xb0
Sun Sep  1 19:30:19 2019 kern.warn kernel: [138620.432948] sp : ffffff8008013b40
Sun Sep  1 19:30:19 2019 kern.warn kernel: [138620.437738] x29: ffffff8008013b40 x28: ffffffc03a3b4a38
Sun Sep  1 19:30:19 2019 kern.warn kernel: [138620.444494] x27: 0000000000000000 x26: 0000000000010600
Sun Sep  1 19:30:19 2019 kern.warn kernel: [138620.451220] x25: 0000000000000001 x24: ffffffc033c5c0a5
Sun Sep  1 19:30:19 2019 kern.warn kernel: [138620.457894] x23: ffffff80088e39c0 x22: ffffffc0359dd780
Sun Sep  1 19:30:19 2019 kern.warn kernel: [138620.464521] x21: ffffffc034cb18f8 x20: ffffff80088e8000
Sun Sep  1 19:30:19 2019 kern.warn kernel: [138620.471099] x19: 0000000000000000 x18: 0000000000000000
Sun Sep  1 19:30:19 2019 kern.warn kernel: [138620.477632] x17: 0000000000000000 x16: 0000000000000000
Sun Sep  1 19:30:19 2019 kern.warn kernel: [138620.484104] x15: 0000000000000001 x14: 07550720076d076d
[... cut ...]
Sun Sep  1 19:30:19 2019 kern.warn kernel: [138620.633852]  el1_irq+0xb0/0x140
Sun Sep  1 19:30:19 2019 kern.warn kernel: [138620.637759]  arch_cpu_idle+0x10/0x18
Sun Sep  1 19:30:19 2019 kern.warn kernel: [138620.642087]  do_idle+0x10c/0x228
Sun Sep  1 19:30:19 2019 kern.warn kernel: [138620.646067]  cpu_startup_entry+0x20/0x28
Sun Sep  1 19:30:19 2019 kern.warn kernel: [138620.650747]  secondary_start_kernel+0x1ac/0x1c0
Sun Sep  1 19:30:19 2019 kern.warn kernel: [138620.656045] ---[ end trace 6dff8b39daee894e ]---`

closed 2019-12-20 by Ysurac

Tom-T, 2021-02-15:
I'm also seeing this error. Just posting here to show it is still happening. 

```
BusyBox v1.31.1 () built-in shell (ash)

   ___                 __  __ ___ _____ ___ ___              _           
  / _ \ _ __  ___ _ _ |  \/  | _ \_   _/ __| _ \_ _ ___ _  _| |_ ___ _ _ 
 | (_) | '_ \/ -_) ' \| |\/| |  _/ | || (__|  _/ '_/ _ \ || |  _/ -_) '_|
  \___/| .__/\___|_||_|_|  |_|_|   |_| \___|_| |_| \___/\_,_|\__\___|_|  
       |_|                                                               
------------------------------------------------------------------------------
   (r0+15225-bfc433efd4)
------------------------------------------------------------------------------
-----------------------------------------------------
 PACKAGE:     openmptcprouter
 VERSION:     v0.57.3

 BUILD REPO:  https://github.com/ysurac/openmptcprouter
 BUILD DATE:  Tue Jan  5 07:01:40 UTC 2021
-----------------------------------------------------
root@Daficco:~# logread -f
Sun Feb 14 19:05:07 2021 user.notice omr-bypass: Starting OMR-ByPass...
Sun Feb 14 19:05:10 2021 user.notice omr-bypass: Reload dnsmasq...
Sun Feb 14 19:05:11 2021 daemon.info dnsmasq[30342]: read /etc/hosts - 4 addresses
Sun Feb 14 19:05:11 2021 daemon.info dnsmasq[30342]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Sun Feb 14 19:05:11 2021 daemon.info dnsmasq-dhcp[30342]: read /etc/ethers - 0 addresses
Sun Feb 14 19:05:11 2021 user.notice omr-bypass: OMR-ByPass is running
Sun Feb 14 19:05:13 2021 daemon.info glorytun: starting glorytun vpn instance vpn
Sun Feb 14 19:05:20 2021 user.notice post-tracking-post-tracking: omrvpn down because gateway down and ping from 10.255.255.2 error (4.2.2.1,8.8.8.8)
Sun Feb 14 19:05:21 2021 user.notice post-tracking-post-tracking: Glorytun VPN down, restart it
Sun Feb 14 19:05:21 2021 daemon.info glorytun[27350]: STOPPED tun0
Sun Feb 14 19:05:21 2021 daemon.notice netifd: Network device 'tun0' link is down
Sun Feb 14 19:05:21 2021 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Sun Feb 14 19:05:21 2021 daemon.notice netifd: Interface 'omrvpn' is now down
Sun Feb 14 19:05:21 2021 daemon.debug dnsmasq[30342]: stopped listening on tun0(#33): 10.255.255.2 port 53
Sun Feb 14 19:05:21 2021 daemon.info glorytun: starting glorytun vpn instance vpn
Sun Feb 14 19:05:21 2021 user.notice ModemManager: hotplug: remove network interface tun0: event processed
Sun Feb 14 19:05:21 2021 user.notice ModemManager: hotplug: event reported: action=remove, name=tun0, subsystem=net
Sun Feb 14 19:05:21 2021 daemon.info glorytun[3166]: INITIALIZED tun0
Sun Feb 14 19:05:21 2021 daemon.notice netifd: Interface 'omrvpn' is disabled
Sun Feb 14 19:05:21 2021 daemon.notice netifd: Interface 'omrvpn' is enabled
Sun Feb 14 19:05:21 2021 daemon.notice netifd: Interface 'omrvpn' is disabled
Sun Feb 14 19:05:21 2021 daemon.notice netifd: Interface 'omrvpn' is enabled
Sun Feb 14 19:05:21 2021 daemon.notice netifd: Network device 'tun0' link is up
Sun Feb 14 19:05:21 2021 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Sun Feb 14 19:05:21 2021 daemon.notice netifd: Interface 'omrvpn' is setting up now
Sun Feb 14 19:05:21 2021 daemon.notice netifd: Interface 'omrvpn' is now up
Sun Feb 14 19:05:21 2021 user.notice ModemManager: hotplug: add network interface tun0: event processed
Sun Feb 14 19:05:21 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tun0, subsystem=net
Sun Feb 14 19:05:21 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Feb 14 19:05:21 2021 daemon.debug dnsmasq[30342]: listening on tun0(#34): 10.255.255.2 port 53
Sun Feb 14 19:05:21 2021 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Sun Feb 14 19:05:22 2021 daemon.err glorytun[3166]: 192.250.230.117.65001: connected
Sun Feb 14 19:05:22 2021 daemon.info glorytun[3166]: STARTED tun0
Sun Feb 14 19:05:22 2021 kern.warn kernel: [  555.987208] ------------[ cut here ]------------
Sun Feb 14 19:05:22 2021 kern.warn kernel: [  555.996753] refcount_t: underflow; use-after-free.
Sun Feb 14 19:05:22 2021 kern.warn kernel: [  556.006444] WARNING: CPU: 0 PID: 0 at lib/refcount.c:190 refcount_sub_and_test_checked+0xd8/0xe0
Sun Feb 14 19:05:23 2021 kern.warn kernel: [  556.020324] Modules linked in: ath9k ath9k_htc ath9k_common rtl8192cu rtl8192c_common rtl_usb rt2800usb rt2800lib qcserial pppoe ppp_async option ipw brcmfmac ath9k_hw ath usb_wwan usb_serial_simple ti_usb_3410_5052 sierra rtlwifi rtl8xxxu rtl8187 rt2x00usb rt2x00lib pppox ppp_generic pl2303 oti6858 mt7601u mos7720 mmc_spi mct_u232 mac80211 lzo keyspan ipt_REJECT garmin_gps ftdi_sio ebtable_nat ebtable_filter ebtable_broute cypress_m8 cp210x ch341 cfg80211 cdc_mbim belkin_sa ark3116 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_policy xt_pkttype xt_owner xt_ndpi xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iface xt_hl xt_helper xt_hashlimit xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_condition xt_comment xt_addrtype xt_TRACE xt_TPROXY xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_IPMARK xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY xt_ACCOUNT visor via_velocity via_rhine usbserial usbhid tulip ts_fsm ts_bm
Sun Feb 14 19:05:23 2021 kern.warn kernel: [  556.020456]  sr9700 solos_pci smsc75xx slhc sky2 skge sis900 sis190 sierra_net sch_cake rtl8150 rndis_host r8712u(C) r8169 r8152 r6040 qmi_wwan pegasus pcnet32 of_mmc_spi nf_tproxy_ipv6 nf_tproxy_ [... cut ...]
Sun Feb 14 19:07:05 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tun0, subsystem=net
Sun Feb 14 19:07:05 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Feb 14 19:07:05 2021 daemon.debug dnsmasq[11618]: listening on tun0(#35): 10.255.255.2 port 53
Sun Feb 14 19:07:05 2021 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Sun Feb 14 19:07:06 2021 daemon.err glorytun[16591]: 192.250.230.117.65001: connected
Sun Feb 14 19:07:06 2021 daemon.info glorytun[16591]: STARTED tun0

```