Ysurac/openmptcprouter #586: 0.51.5 Image metadata not found when using sysupgrade

xabix, 2019-08-04:
## Expected Behavior

Trying to upgrade from 0.50.2 but the image doesn't seem good or recognized.

## Steps to Reproduce the Problem

  1. sha256sum openmptcprouter-v0.51.5-r010518-6d61fcfa9e-x86-64-combined-ext4-efi.img.gz 
abb94401aae25e4e9984e9f1580c82b616541e8afed920e5c28859c145519edf  openmptcprouter-v0.51.5-r010518-6d61fcfa9e-x86-64-combined-ext4-efi.img.gz

  2. sysupgrade -v openmptcprouter-v0.51.5-r010518-6d61fcfa9e-x86-64-combined-ext4-efi.img.gz 

=>
Image metadata not found
Unable to determine upgrade device
Image check 'platform_check_image' failed.

What can be the issue?

## Specifications

  - OpenMPTCProuter version: 0.50.2
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter platform: x86_64

Ysurac, 2019-08-04:
True, I tested and same problem. I'm not able to update.
I will try to see where is the problem (maybe the updated EFI patch ?)

xabix, 2019-08-04:
Should I get any advantage on a virtualized environment to use the EFI version? Should I then try the non EFI version to see if it better and if I don't need the EFI version?

Ysurac, 2019-08-04:
There is no advantage at all to use EFI version in any case.
EFI version should be used only if no other choice.

xabix, 2019-08-05:
Ok thanks for the feedback. Any idea of the issue? Is there a workaround for me to upgrade? I d like to test this new kernel and see if it solves some latency issues previously reported

Ysurac, 2019-08-05:
There is a bug in UEFI patch and update is not possible. To make update possible, you have to copy this file https://www.openmptcprouter.com/fix/common.sh in _/lib/upgrade/_

xabix, 2019-08-05:
So with the script, the installation starts but for some reason it doesn't work as it reboots in 0.50.2

> root@OpenMPTCProuter:/tmp# sysupgrade -v openmptcprouter-v0.51.5-r010518-6d61fcfa9e-x86-64-combined-ext4.img.gz 
Image metadata not found
Reading partition table from bootdisk...
Reading partition table from image...
Partition layout has changed. Full image will be written.
Saving config files...
etc/config/dhcp
etc/config/dropbear
etc/config/dscp
etc/config/etherwake
etc/config/firewall
etc/config/glorytun
etc/config/glorytun-opkg
etc/config/glorytun_recipes
etc/config/iperf
etc/config/luci
etc/config/mail
etc/config/mlvpn
etc/config/network
etc/config/nginx-ha
etc/config/omr-bypass
etc/config/omr-quota
etc/config/omr-tracker
etc/config/openmptcprouter
etc/config/openvpn
etc/config/rpcd
etc/config/shadowsocks-libev
etc/config/snmpd
etc/config/sqm
etc/config/system
etc/config/ucitrack
etc/config/uhttpd
etc/config/unbound
etc/config/upnpd
etc/config/vnstat
etc/crontabs/root
etc/dropbear/dropbear_rsa_host_key
etc/group
etc/hosts
etc/inittab
etc/iproute2/rt_tables
etc/luci-uploads/.placeholder
etc/luci-uploads/openvpn.key
etc/msmtprc
etc/nginx/fastcgi_params
etc/nginx/koi-utf
etc/nginx/koi-win
etc/nginx/mime.types
etc/nginx/naxsi_core.rules
etc/nginx/nginx.conf
etc/nginx/win-utf
etc/opkg/keys/1035ac73cc4e59e3
etc/opkg/keys/5151f69420c3f508
etc/opkg/keys/6ca8c1647a60741d
etc/opkg/keys/72a57f2191b211e0
etc/opkg/keys/792d9d9b39f180dc
etc/opkg/keys/9ef4694208102c43
etc/opkg/keys/b26f36ae0f4106d
etc/opkg/keys/b5043e70f9a75cde
etc/opkg/keys/c10b9afab19ee428
etc/opkg/keys/dace9d4df16896bf
etc/opkg/keys/dd6de0d06bbd3d85
etc/passwd
etc/profile
etc/rc.local
etc/shadow
etc/shells
etc/sysctl.conf
etc/uhttpd.crt
etc/uhttpd.key
etc/uhttpd.key
etc/uhttpd.crt
Commencing upgrade. Closing all shell sessions.

BusyBox v1.30.1 () built-in shell (ash)

   ___                 __  __ ___ _____ ___ ___              _           
  / _ \ _ __  ___ _ _ |  \/  | _ \_   _/ __| _ \_ _ ___ _  _| |_ ___ _ _ 
 | (_) | '_ \/ -_) ' \| |\/| |  _/ | || (__|  _/ '_/ _ \ || |  _/ -_) '_|
  \___/| .__/\___|_||_|_|  |_|_|   |_| \___|_| |_| \___/\_,_|\__\___|_|  
       |_|                                                               
------------------------------------------------------------------------------
   (r0+10127-3209f5ae3d)
------------------------------------------------------------------------------
-----------------------------------------------------
 PACKAGE:     openmptcprouter
 VERSION:     v0.50.2

 BUILD REPO:  https://github.com/ysurac/openmptcprouter
 BUILD DATE:  Tue Jun  4 18:27:10 UTC 2019
-----------------------------------------------------

Ysurac, 2019-08-05:
True. No idea why...

xabix, 2019-08-05:
Ok closing the issue. I believe my issue was linked to:
root@OpenMPTCProuter:/tmp# gunzip openmptcprouter-v0.51.5-r010518-6d61fcfa9e-x86-64-combined-ext4-efi.img.gz 
gunzip: write: No space left on device

Maybe something to add within the install script to ensure there is enough space before starting.

closed 2019-08-05 by xabix