Ysurac/openmptcprouter #587: On Debian 10, shorewall uses all resourses

tillfr-1, 2019-08-05:
## Behavior

On a Debian 10 installation, shorewall uses up all resourses.

## Steps to Reproduce the Problem

  1. Debian 10 installation (Linux debian 4.19.0-5-amd64 #1 SMP Debian 4.19.37-5+deb10u1 (2019-07-19) x86_64 GNU/Linux)
  2. run "wget -O - http://www.openmptcprouter.com/server/debian10-x86_64.sh | sh"
  3. Debian 10 with OpenMPTCProuter VPS 0.999 (Linux debian 4.19.56-mptcp #1 SMP Sat Jul 6 07:28:10 UTC 2019 x86_64 GNU/Linux) and shorewall is using all resourses.

## Specifications

  - OpenMPTCProuter version: 0.51.5
  - OpenMPTCProuter VPS version: 0.999
  - OpenMPTCProuter platform: RPI3B+

tillfr-1, 2019-08-05:
/var/log/shorewall-init.log

Aug  4 11:28:22 Processing /etc/shorewall/params ...
Aug  4 11:28:22 Processing /etc/shorewall/shorewall.conf...
Aug  4 11:28:22    WARNING: The CHAIN_SCRIPTS configuration option is no longer supported /etc/shorewall/shorewall.conf (line 147)
Aug  4 11:28:22    WARNING: The INLINE_MATCHES configuration option is no longer supported /etc/shorewall/shorewall.conf (line 183)
Aug  4 11:28:22    WARNING: The LOAD_HELPERS_ONLY configuration option is no longer supported /etc/shorewall/shorewall.conf (line 191)
Aug  4 11:28:22    WARNING: The MAPOLDACTIONS configuration option is no longer supported /etc/shorewall/shorewall.conf (line 199)
Aug  4 11:28:22    WARNING: The MODULE_SUFFIX configuration option is no longer supported /etc/shorewall/shorewall.conf (line 205)
Aug  4 11:28:22 Loading Modules...
Aug  4 11:28:23 Compiling /etc/shorewall/zones...
Aug  4 11:28:23 Compiling /etc/shorewall/interfaces...
Aug  4 11:28:23    Interface "net eth0 dhcp,tcpflags,routefilter,nosmurfs,logmartians,sourceroute=0" Validated
Aug  4 11:28:23    Interface "vpn gt-tun0 nosmurfs,routefilter,logmartians,tcpflags" Validated
Aug  4 11:28:23    Interface "vpn gt-udp-tun0 nosmurfs,routefilter,logmartians,tcpflags" Validated
Aug  4 11:28:23    Interface "vpn mlvpn0 nosmurfs,routefilter,logmartians,tcpflags" Validated
Aug  4 11:28:23    Interface "vpn tun0 nosmurfs,routefilter,logmartians,tcpflags" Validated
Aug  4 11:28:23 Determining Hosts in Zones...
Aug  4 11:28:23    fw (firewall)
Aug  4 11:28:23    net (ipv4)
Aug  4 11:28:23       eth0:0.0.0.0/0
Aug  4 11:28:23    vpn (ipv4)
Aug  4 11:28:23       gt-udp-tun0:0.0.0.0/0
Aug  4 11:28:23       mlvpn0:0.0.0.0/0
Aug  4 11:28:23       gt-tun0:0.0.0.0/0
Aug  4 11:28:23       tun0:0.0.0.0/0
Aug  4 11:28:23 Locating Action Files...
Aug  4 11:28:23    ERROR: Unknown Action (Drop) in DROP_DEFAULT setting /usr/share/shorewall/actions.std (EOF)

Ysurac, 2019-08-05:
Should be fixed if you run the VPS script again.

ThibautBascoulergue, 2019-08-06:
VPS script block after shorewall on a new install : 
[image]

MacGeyer, 2019-08-06:
I could rerun the script on my Debian 10, one hour ago

MacGeyer, 2019-08-06:
By the way, with the updated script my server load average dropped from 1.6/1.8 to 0.0 !

Ysurac, 2019-08-06:
I fixed the script.

closed 2019-08-14 by Ysurac