Ysurac/openmptcprouter #588: Kernel Panic

Marctraider, 2019-08-05:
## Expected Behavior
No Kernel Panic

## Actual Behavior
Kernel Panic, when I start openspeedtest or something.

## Steps to Reproduce the Problem

  1. Install openmptcprouter-v0.51.5-r010518-6d61fcfa9e-mvebu-cortexa9-linksys_wrt3200acm-squashfs-sysupgrade.bin or factory i guess.
  2. Install VPS on Debian 9 or Ubuntu 18 LTS 
  3. Link 3 WAN's and set up OpenMPTCProuter etc... Start speedtest and kernel panic on VPS side.

Something triggers it because not always have i had this, I had the config running flawlessly for 3 days but I just tried different setups etc...
[image]

## Specifications

  - OpenMPTCProuter version: v0.51.5-
  - OpenMPTCProuter VPS version: 0.999?
  - OpenMPTCProuter platform: linksys_wrt3200acm

Ysurac, 2019-08-05:
You seems to use mptcp redundant scheduler. This seems to be not really stable.

Marctraider, 2019-08-05:
Correct.

When will there be proper redundant support for lowest latency tho? Thanks!

Marctraider, 2019-08-08:
> 
> 
> You seems to use mptcp redundant scheduler. This seems to be not really stable.

Is there any version which does work OK with redundant? Thank you!

closed 2019-12-20 by Ysurac