Ysurac/openmptcprouter #589: DHCP n'est plus comme avant.

Dragonox77, 2019-08-06:
Bonjour, merci d'être encore très actif sur ta solution.
## Expected Behavior
Pouvoir renseigner une passerelle dans les baux statiques

## Actual Behavior
On ne peut plus renseigner la passerelle.

J'en ai besoin pour faire fonctionner ma TV mais depuis la mise a jour, impossible de trouver une solution depuis cette maj. Je pourrais installer un serveur dhcp manuellement mais le faire directement depuis le raspberry pi est beaucoup mieux.
## Specifications

  - OpenMPTCProuter version: 0.51.5
  - OpenMPTCProuter VPS version: 0.999 4.19.56-mptcp
  - OpenMPTCProuter platform: RPI3 x86_64

Dragonox77, 2019-08-06:
Du coup j'ai réussi en modifiant directement la config ici: /etc/config/dhcp et j'ai rajouté:
option gw '192.168.1.1'
Ca a marché. Du coup si juste pour la prochaine maj tu peux rajouté directement depuis l'interface graphique, ça serait vraiment cool :D

Ysurac, 2019-08-06:
Oui ça a été perdu dans la modification de l'interface, je vais remettre ça pour la prochaine version.

closed 2019-12-20 by Ysurac