Ysurac/openmptcprouter #590: Docker image for router ?

nicoz76, 2019-08-08:
Hello, I did not find a docker image for the router part. is it achievable?

I saw that Openwrt was present on docker, can we rely on this image to then add some packages?

I thank you in advance.

cordially

Ysurac, 2019-08-08:
Possible (this only need to import rootfs) but host kernel need to be patched with MPTCP support and have all needed modules compiled.

nicoz76, 2019-09-07:
Merci pour la réponse.
mais c'est hors de mes compétences :)

closed 2019-12-20 by Ysurac

unixfox, 2019-12-20:
Why closing this issue without any explanation?

Ysurac, 2019-12-20:
Old issue, should have be closed since a long time and nothing to say about that for now.

unixfox, 2019-12-20:
There are issues older than this... From I can see there isn't any Docker support so why should this issue closed? The initial author, me and probably other people are interested into a Docker image for openmptcrouter

Ysurac, 2019-12-20:
As I already said, you can import the rootfs.
Do something like this: https://openwrt.org/docs/guide-user/virtualization/docker_openwrt_image rootfs available here: https://www.openmptcprouter.com/release/4.19/x86_64/targets/x86/64/
But host kernel need to be patched with MPTCP.
But I'm not sure this can work correctly, so I will not support this on router part for now. So I close this.