Ysurac/openmptcprouter #592: Openmptcprouter LXD image

TomH303, 2019-08-09:
Hi,

thanks for this interesting project.

Would it be possible to build an Openmptcprouter LXD image or that you provide instructions how to create it?

Thomas

Ysurac, 2019-08-09:
It's the same problem as docker: https://github.com/Ysurac/openmptcprouter/issues/590

TomH303, 2019-09-25:
Hi,

I can see that openmptcprouter v0.52 builds do contain rootfs img

x86-generic-rootfs-ext4.img.gz
x86-generic-rootfs-squashfs.img.gz
x86-64-generic-rootfs-ext4.img.gz
x86-64-generic-rootfs-squashfs.img.gz

Could you create default rootfs builds for armvirt 32/64 like https://downloads.openwrt.org/snapshots/targets/armvirt/32/openwrt-armvirt-32-default-rootfs.tar.gz too?

Ysurac, 2019-09-25:
For LXD, host kernel need to be patched with MPTCP support.
I don't want to support armvirt (I don't know any usage for that), so you can build image yourself: https://github.com/Ysurac/openmptcprouter/wiki/Create-image-for-unsupported-platform

closed 2019-12-20 by Ysurac