Ysurac/openmptcprouter #593: Problème TCP  (réutilisation d'un port source par le client)

thepaka, 2019-08-09:
Bonjour, 
( désolé de ne pas suivre les guidelines pour la création de ticket mais je pense que ca va faciliter les choses)

J'ai identifié un problème TCP lors de la communication entre un client et OpenMPTCPRouteur .
Pour résumer : 
 * En théorie, les connexions TCP sont distinguées  par le 4-tuple  : ip-source, port-source, ip-destination, port-destination. 
 * Le problème c'est que cela ne fonctionne pas,  quand un client réutilise un port source encore utilisé par une précédente connexion,  le routeur ré-envoi le TCP ACK de la précédente connexion au lieu d'en établire une nouvelle.

Je vois bien le premier paquet de la connexion (le TCP SYN)  passer par la chaine  NAT  'ss_rules_forward'  et  doit donc ensuite être redirigé vers le port 1100 de shadowsocks-libev ( ss-redir). Sauf que, j'ai essayé de debug ss-redir et apparament il ne reçoit pas le paquet.

A savoir que  j'ai rencontré ce problème à l'origine  sur la solution OverTheBox d'OVH (ils  me semblent qu'ils utilisent une version modifié de openmptcprouter) .  J'ai posté un message détaillant le problème chez eux : https://community.ovh.com/t/nombreux-timeouts/17625/9?u=briced3 , mais pas de réponse depuis.

Ysurac, 2019-08-09:
OverTheBox et OpenMPTCProuter utilisent la même base OpenWrt avec un patch MPTCP ainsi que Shadowsocks et glorytun (du moins ils utilisaient, je sais pas si MPTCP est toujours utilisé par OverTheBox), mais pas vraiment de code en commun sinon.

Je dirais soucis au niveau de Shadowsocks, par contre il faut penser que par défaut plusieurs ss-redir sont ancés et écoutent sur le port 1100.
Il faudrait voir si ce n'est pas un soucis avec les paramètres fast-open ou no-delay de ss-redir...

thepaka, 2019-08-12:
En faisant les tests j'ai désactivé le maximum de fonctionnalités de ss-redir , notamment  fast-open, no-delay, port reuse, mptcp, ...
Le fait de désactiver l'option 'port reuse' permet aussi de tester avec une seule instance de ss-redir (car les autres ne peuvent plus se lancer si le port 1100 est déjà utilisé).

Pour le dernier test que j'ai fait , j'ai remplacé  shadowsocks  par  redsocks + ssh . Le problème est toujours présent.  

Je pense maintenant que c'est plutôt un bug kernel avec la gestion du NAT , en faisant quelques recherche j'ai trouvé qu'il existe un problème de race condition avec SNAT et DNAT (et donc le REDIRECT utilisé pour ss-redir).

Je trouve étonnant que personne d'autre n'ait remonté ce problème. Je vais continuer les tests, ...

thepaka, 2019-09-09:
Je pense avoir trouvé une explication (et solution) assez logique : 

Introduction :

 * Les connexions TCP sont distinguées par le 4-tuple : ip-source, port-source, ip-destination, port-destination.
 * Il est possible qu'un client réseau utilise le même port-source pour différente connexions simultanément
 * Cela est même assez courant si le 'client réseau' est un autre routeur utilisant du NAT
 
Problème : 

 * OpenMptcpRouteur utilise un tunnel shadowsocks, qui nécessite de rediriger le trafic entrant vers un service local (ss-redir)
 * Cette redirection se fait par DNAT via la règle iptable REDIRECT, ce qui va modifier le couple ip-destination / port-destination
 * Si le trafic est redirigé vers un unique couple ip-destination / port-destination, alors il devient impossible pour le routeur de distinguer les différents connexions provenant d'un même couple ip-source / port-source.

Solution : 

 * Utiliser plusieurs instances de ss-redir écoutant sur différents ports
 * Modifier la règle REDIRECT pour utiliser ces différents ports ( l'option --to-ports permet de spécifier une plage de ports )

closed 2019-12-20 by Ysurac