Ysurac/openmptcprouter #594: Inproper use of multiple VPS(es) - conflicting behaviour

vistman, 2019-08-11:
## Expected Behavior

Using exclusively VPS set as master in case of multiple VPS.
Also if multiple VPSes are used it is impossible to set ShadowSocks and Glorytun keys for each of them (under Advanced settings ShadowSocks / Glorytun keys) - only one VPS key set is allowed.

## Actual Behavior

Mixing up master and secondary (maybe tertiary and other) VPS ie. showing secondary VPS name and master VPS IP address. Also connection is constantly breaking up.

Need to manually change ShadowSocks and Glorytun keys upon VPS change.

## Steps to Reproduce the Problem

  1. Add new VPS
  2. Set new VPS as master
  3. Set old VPS as master
  4. Check out Status page

## Specifications

  - OpenMPTCProuter version: 0.51.5
  - OpenMPTCProuter VPS version: 0.995 (1st VPS) and 0.999 (2nd VPS)
  - OpenMPTCProuter platform: RPI3 / x86_64

Ysurac, 2019-08-11:
Only VPS admin key is needed when there is multiple VPS: key are then copied to other VPS.
Check log if there is a problem.

Ysurac, 2019-08-11:
Version should also be the same on all VPS...

vistman, 2019-08-12:
> 
> 
> Only VPS admin key is needed when there is multiple VPS: key are then copied to other VPS.
> Check log if there is a problem.

Automatically or should I do it manually?

vistman, 2019-08-12:
> 
> 
> Version should also be the same on all VPS...

Yeah, that is another problem ... there aren't a lot of VPS providers in nearest interconnection node  and they all seem to have (b)locked ports higher than certain number (they weren't keen to share this info with me), so after rebooting VPS I lost SSH connectivity with my VPS.
I'll have to edit shorewall and sshd rules to enable port 22 or some other port which is not being blocked to be able to update VPS script. Maybe at certain point you could implement port selection in VPS setup script to make things easier and this shouldn't be too difficult for you ;-)

Ysurac, 2019-08-12:
You should set VPS admin key only for both VPS. Other keys are then synchronized automatically if admin script on both VPS can be contacted.
I never have a VPS that block high ports, what are the providers ?
Shadowsocks, VPN, admin script,... all use ports higher than 65000.

vistman, 2019-08-12:
> You should set VPS admin key only for both VPS. Other keys are then synchronized automatically if admin script on both VPS can be contacted.
So, just to make sure - when I run installation script on 2nd VPS I do not need to enter newly generated admin key into OpenMPTCPRouter / Settings Wizard / Server Settings?

> I never have a VPS that block high ports, what are the providers?
> Shadowsocks, VPN, admin script,... all use ports higher than 65000.
Only SSH access seems to be the problem, so only SSH port is problematic (for me).
Austrian providers ...

Ysurac, 2019-08-12:
You NEED to enter the admin key but only the admin key for both VPS on OpenMPTCProuter wizard.

Only port 65222 blocked ? this seems really strange :)

vistman, 2019-08-12:
> 
> 
> You NEED to enter the admin key but only the admin key for both VPS on OpenMPTCProuter wizard.

OK, understood. Thanks.

> 
> Only port 65222 blocked ? this seems really strange :)

It seems they tend to use non-standard SSH port (great from generic security point of view but not in this case) and drop SSH traffic on any other ports. So far I had no luck bypassing this on VPS level. I'll try today resolving this issue using fresh installation with modified version of your server installation script to match this SSH port. Hopefully it will work and enable me to retain SSH access.

vistman, 2019-08-12:
I've just upgraded all VPSes to 0.999 and everything seems to work fine but OpenMPTCPRouter is still mixing up VPS server names and IP addresses ie. I have displayed 1st VPS IP and second VPS name ...

Ysurac, 2019-08-12:
This will be fixed in next release.

closed 2019-12-20 by Ysurac