Ysurac/openmptcprouter #595: Speedtest - aggregated vs separate

bambuca, 2019-08-11:
## Actual Behavior
I use `omr-test-speed` cli. I expect that aggregated speed should be approximately the sum of each ISP.
I have tvo ISP providers, one on eth0.3 and second on eth0.4.

Aggregated speed `omr-test-speed` is ca. 15.2 MB/s
ISP1 speed: 'omr-test-speed eth0.3' is ca. 3.7 MB/s
ISP2 speed: 'omr-test-speed eth0.4' is ca. 2.2 MB/s

Sum of ISP1 and ISP2 is then ca. 6 MB/s. Why the aggregated speed is 15.2?

When i try to measure speed with speedtest.com or nperf.com then the speed is 6 MB/s.

The bandwidth of ISP1 is 4 MB/s and for ISP2 is 20 MB/s so the aggregated speed is often over 20 MB/s.

[image]

## Specifications

  - OpenMPTCProuter version: 0.51.5
  - OpenMPTCProuter VPS version: 0.999
  - OpenMPTCProuter platform: RPI4 32b

Ysurac, 2019-08-11:
Maybe a configuration problem. Where this eth0.x are coming from ? Why using vlan instead of default macvlan ?

bambuca, 2019-08-11:
VLANs are configured on my switch. One of my provider use DHCP and when i was using macvlan then the second provider calls me that the DHCP is propagated to his network. I'm not very familiar with macvlan but I think that when there is a problem with mi RPI then the VLAN (through macvlan) is not configured and both providers are connected together by my switch.

Ysurac, 2019-08-14:
You can try to use IP instead of interface for omr-test-speed
How are you connected for nperf test ? via wifi or ethernet ?

bambuca, 2019-08-30:
Hi, I tried it with IP (i used IPs from ISP. The results are the same.

[image]

The first ip (in my previous post eth0.3) look ok - its speed is the speed declared by ISP.
The second ip (eth0.4) is very slow when tested with ip or eth0.4.

nperf when connected by ethernet cable  8,2 MB/s

bambuca, 2019-08-30:
I tried to keep enabled only one WAN.

When I have enabled only WAN 1 then "omr-test-speed" returns approx 7900 kB/s. amd "omr-test-speed eth0.3" returns the same 7600 KB/s.

When I have enabled only WAN 2 then "omr-test-speed" returns approx 12.1 MB/s. amd "omr-test-speed eth0.4" returns only 3210 kB/s.

Ysurac, 2019-08-30:
So you have a problem with "eth0.4". You should check cable, switch,...

bambuca, 2019-08-31:
I swapped both of my providers in the switch, so the one that was originally eth0.3 is now eth0.4 and vice versa. Unfortunately the problem still persists - now on eth0.3. 

If I connect to the provider directly (ie without RPI with OMR), then speed (speedtest.net) corresponds to what returns "omr-test-speed" (without the specified interface).

I tried the installation from scratch, but still the same result. When the new version will be available, it may have been fixed "by the way" :).

Ysurac, 2019-08-31:
It's not a problem of interface, for the router the speed is quite slow, you need to fix this but it's a problem with this provider (modem, cable, switch,...)

closed 2019-12-20 by Ysurac