Ysurac/openmptcprouter #596: Aggregate in cascade

JorgeMoort89, 2019-08-12:
When I try to bond connections in cascade (logical explanation below):
WAN_1 and WAN_2 correctly bonded by OMR_1, OMR_1 as WAN and WAN_3 not bonded by OMR_2. WAN_3 is MTPCP capable, I don't know why OMR_1 is blocking MPTCP.
There are some way to bypass it? I can't bond WAN_3 directly on WAN_1 in this case.

[image]

## Specifications

  - OpenMPTCProuter version: 0.49
  - OpenMPTCProuter VPS version: 0.997
  - OpenMPTCProuter platform: x86_64

Ysurac, 2019-08-12:
Why there is an OMR_1 and an OMR_2 ?

JorgeMoort89, 2019-08-12:
Because I have 2 offices, the 1st one with 2 internet connection aggregated via OMR_1 and the second one with one internet connection and a link from OMR_1, I'm trying to aggregate them with OMR_2. I can't create VLAN or edit the network infrastructure in this moment, I don't know why MPTCP is blocked by OMR router itself

bambuca, 2019-08-13:
What is your goal? You have 3 ISP providers. Two of them are at office 1 and one is at office 2? How are offices connected together?
I can't imagine how a cascade could improve connectivity, so please explain your goal bit deeper.
The final speed and stability is always limited on the connections to VPS.

JorgeMoort89, 2019-08-14:
Exactly, the two offices are connected by a link in the LAN of OMR_1, my goal is to aggregate WAN_3 and OMR_1 with OMR_2. I think that the image posted explain that clearly.

Ysurac, 2019-08-14:
So you need 2 VPS, and I have really no idea if this can work...

JorgeMoort89, 2019-08-14:
I have two different VPS with 2 different public IP but OMR_1 block MPTCP

Ysurac, 2019-08-14:
How do you check that MPTCP is blocked ? with tracebox ?

bambuca, 2019-08-14:
I was thinking about the same thing with my neighbor, but finally we made it a bit different and we are happy with that. We connect WAN_3 into OMR_1 and we both use only OMR_1. Could it be applicable also for you?
Of course your solution should also work (in theory), but then the office 1 has only two providers and office 2 has three and traffic is then also a bit complicated.

JorgeMoort89, 2019-08-14:
I've just checked, omr-mptcp-intf and omr-tracebox-mptcp give MPTCP compatibility on both WANs but it won't aggregate them. In this case I can't connect WAN_3 directly to OMR_3 sadly :( Are there any way to have multiple public IP with only one OMR router?

Ysurac, 2019-08-14:
Not out the box, maybe with some config on the VPS...

JorgeMoort89, 2019-08-14:
What can I do to do this?

Ysurac, 2019-08-15:
Buy multiple IP from your VPS provider, configure them on your output interface, configure shorewall on the VPS to use them the way you want: http://shorewall.org/Shorewall_and_Aliased_Interfaces.html
I don't know why you want to use multiple public IP.

JorgeMoort89, 2019-08-15:
Just to NAT multiple web servers, I've already /30 on the VPS NIC, I just don't know how to manage the traffic. Have I to create a second LAN on OMR? How can I do?

closed 2019-12-20 by Ysurac