Ysurac/openmptcprouter #597: Online.net VPS

ctrlx83, 2019-08-13:
Hi everyone,

I try OMR on a scaleway VPS (based on AMS1), web navigation is OK, next I try OMR on a online.net VPS (based on DC5), ssl navigation is DOWN.

MTU are same : 1500

When I go back to the VPS Scaleway, everything works properly.
In both cases, the IP addresses are correct and all the flags are green.

Thanks

## Specifications

  - OpenMPTCProuter version: v0.51.5
  - OpenMPTCProuter VPS version: 0.999 4.19.56-mptcp
  - OpenMPTCProuter platform: x86_64

Ysurac, 2019-08-13:
There is still online.net VPS ? I can't find them.

ctrlx83, 2019-08-13:
Hi @Ysurac 

My online.net VPS is : https://www.online.net/fr/serveurs-dedies/start-2-s-sata

Ysurac, 2019-08-13:
It's a dedicated server, not a VPS :)
I will need to make more test on Debian 10, else I have no real idea...

ctrlx83, 2019-08-13:
Ok
I will try to downgrade server to Debian 9

feedback soon
thanks

ctrlx83, 2019-08-13:
Downgrade to Debian 9 resolve problem !! 
DNS works OK
SSL navigation OK

For information :
Server at Amsterdam with online.net have 38ms ping
Server at Paris with online.net have 21ms ping

closed 2019-08-14 by ctrlx83