Ysurac/openmptcprouter #598: OMR-Tracker and QMI interface

dnwk, 2019-08-14:
When one of the Ethernet is using QMI (Qualcomm LTE), it get errors like

user.notice omr-tracker: Can't create route to 198.27.92.1 via 10.13.11.201 dev wwan0 src 10.13.11.200. waiting...

The interface will show up on "status" page as gateway down.

However, if I SSH to the server and ran command 
ping -I wwan0 -c 3 1.1.1.1
Ping could go through without a problem. 

So, I am suspecting OMR-tracker could not find the correct interface to monitor.  I named the interface WAN3. When connection established, it creates a virtual interface named WAN3_4 with "Protocol: Virtual dynamic interface (DHCP client)". So I suspect OMR-tracker didn't look at the correct interface. 
However, when I run ifconfig on the command line, I did not see extra interface.

Ysurac, 2019-08-14:
As you can see in log it find the correct interface : wwan0. But it can't create a route, you can try youself: ```ip route add 198.27.92.1 via 10.13.11.201 dev wwan0 src 10.13.11.200```

dnwk, 2019-08-14:
So the omr-tracker error went away. I don't know if it successfully created a route. 
However the gateway down message persisted on "status" page. 
And the IP shows up on IP route is different from IP in ifconfig. So I don't know if it matters. 

wwan0     Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          inet addr:10.181.170.239  P-t-P:10.181.170.239  Mask:255.255.255.224
          inet6 addr: fe80::7db:4bec:6560:b2f6/64 Scope:Link
         

root@OpenMPTCProuter:~# ip route
default via 10.255.255.1 dev tun0
default via 192.168.7.1 dev wan1 metric 3
default via 184.*.*.* dev 3g-wan2 metric 4
default via 10.181.170.240 dev wwan0 metric 7
10.181.170.224/27 dev wwan0 scope link metric 7
10.255.255.1 dev tun0 proto kernel scope link src 10.255.255.2
71.*.*.* via 192.168.7.1 dev wan1 metric 1
71.*.*.* via 192.168.7.1 dev wan1 metric 3
71.*.*.* via 184.*.*.*.* dev 3g-wan2 metric 4
71.*.*.* via 10.181.170.240 dev wwan0 metric 7
127.0.0.0/8 dev lo proto static scope link metric 1
184.*.*.* dev 3g-wan2 scope link metric 4
192.168.7.0/24 dev wan1 scope link metric 3

Ysurac, 2019-08-14:
For the gateway down, can you ping the gateway or not ?
For the ip in route, it's in /27 subnet so .225 to .254, no problem here.

dnwk, 2019-08-14:
As wwan0 in the ifconfig, I don't know what is the gateway IP?

Ysurac, 2019-08-14:
It's 10.181.170.240

dnwk, 2019-08-14:
I cannot ping 240.

Ysurac, 2019-08-14:
So this is why you have the message "gateway down". Not really a problem, it's only information in status page, I may add an option to disable the message...

dnwk, 2019-08-20:
I would suggest to ping VPS instead of gateway. Then you will have a glance of performance of each WAN.

Ysurac, 2019-08-21:
I ping VPS too. But pinging gateway make it able to detect when modem is down for ethernet modem for exemple or when interface is up or down when VPS is down (or when it doesn't answer to ping).

closed 2019-12-20 by Ysurac