Ysurac/openmptcprouter #601: Redundancy, Master, Multipath, Backup, DSCP for routing?

Marctraider, 2019-08-16:
Hi! I was curious how this works exactly, so far my setup is like this:

DSL: Multipath (Master)
Cable: Multipath
4G: Multipath (Backup)

Using Redundant scheduler (Works great now btw), but I'm curious as to how the Backup behaves when both the fixed WAN lines are both having packet loss or something.

The 4G initiates TCP connection even as 'backup', but simply does not send data through it until both the other fixed WAN lines are both 100% down, is this a correct assumption?

In this case, I'm wondering what happens when both fixed lines are subject to high packet loss, but pings still occasionally come through, what happens here? Is this simply an 'All or nothing' approach? It's either up or down? How is this measured, or how is this dealt with. 

I'm fairly sure if both fixed lines are having quality issues, but they are not really 'down', my application/game or whatever is still experiencing bad quality I reckon, because 4G 'Backup' won't kick in?

----------------------

That put's me on the next question;

Windows has this awesome feature where you can basically put any application and tag all network traffic from that app with a specific DSCP value, this is incredibly handy to route specific apps through a router for any purpose, like send it over VPN, (or simply for SQM like layer_cake)

My question is, since I want maximum redundancy and stability, I would like to use all three of my WAN connections for real-time and critical data streams. The problem is, 4G is limited by monthly bandwidth.

Is it possible to somehow do this: 

Enable Multipath TCP on all three WAN's, but for the 3rd WAN only pass through packets market by specific DSCP?

This would basically make all 3 connections available for realtime critical data, without fully saturating the 4G for all other activity.

This makes optimal use of my 4G connection without being 'just a backup'.

Ysurac, 2019-08-18:
Backup is used only when connection are detected as down, omr-tracker can be configured to set a connection as down more often.

Not possible to use all connections for some data and not for others.

Marctraider, 2019-08-19:
Thank you for the explanation.

Seems in this case 'Default scheduler' is best for now, since 4G has highest RTT it barely gets used.

closed 2019-08-19 by Marctraider