Ysurac/openmptcprouter #603: gateway down, multipath current state is off

wackal, 2019-08-17:
## Expected Behavior

## Actual Behavior
[image]

## Steps to Reproduce the Problem

  1.wan1 (master) adsl
  2.wan2 (on) 4g router (work if i connect directly on it)
  3.this config its ok for 3 month and tomorow i have this message "gateway down, multipath current state is off"

## Specifications

  - OpenMPTCProuter version: 0.51.5
  - OpenMPTCProuter VPS version: 0.999
  - OpenMPTCProuter platform: RPI3

Ysurac, 2019-08-17:
It's because gateway of wan2 doesn't answer to ping and omr-tracker can't ping any IP using this interface. So this interface doesn't answer at all for the router.
You can check "System log" to have more info.

You should check cables. You can also try via SSH on the router to ping the gateway "192.168.1.254"

wackal, 2019-08-17:
ping not good on SSH
i check cables and reinstall openmptcprouter on my RPI3 and all is ok now
thanks Ysurac

closed 2019-08-17 by wackal