Ysurac/openmptcprouter #604: vpn

jlabre, 2019-08-17:
[google trad]
I am trying to bypass the vpn for mycanal and netflix which solution thank you

je cherche a contourner le vpn pour mycanal et netflix quelle solution merci

Ysurac, 2019-08-17:
En ajoutant les réseaux suivant dans omr-bypass :
```
194.4.242.0/24
194.4.244.0/23
194.4.246.0/24
```
Et si ça fonctionne j'ajouterais ça dans la liste des "protocoles".

jlabre, 2019-08-17:
re bonjour 
dans IPs et réseaux

jlabre, 2019-08-17:
[image]

Ysurac, 2019-08-17:
Ils utilisent peut être d'autres IPs maintenant...
Sinon pour netflix il suffit de choisir le protocol "netflix".

jlabre, 2019-08-17:
je viens de voir aussi que l agrégation ne donne pas le max de debit de toute les box pourquoi ?

Ysurac, 2019-08-17:
Pour canal, le VPS est chez qui ? Je viens de prendre un abonnement pour tester tout en ayant un VPS chez scaleway et ça semble fonctionner sans bypass ce qui m'aide pas beaucoup pour trouver les IPs ;)

jlabre, 2019-08-17:
online.net

closed 2019-08-17 by jlabre

Ysurac, 2019-08-17:
Donc même fournisseur du coup... Ca m'arrange pas pour les tests ça...

jlabre, 2019-08-17:
oui je comprend pas j ai lui de com sur autre site et il marche eu aussi

Ysurac, 2019-08-17:
Visiblement les vidéos viennent de "akamaihd.net" donc en ajoutant ce domaine ça doit passer.

jlabre, 2019-08-17:
non toujour rien bizzar

jlabre, 2019-08-17:
ok c bon avec un reboot de tout merci

Ysurac, 2019-08-17:
ok je vais vérifier pourquoi ça s'applique pas sinon.