Ysurac/openmptcprouter #605: RPi4 4GB: unable to enumerate USB device

rmoriz, 2019-08-17:
## Expected Behavior

no dmesg output

## Actual Behavior

After boot, dmesg is collecting the following error

````
[  849.061273] usb usb2-port1: unable to enumerate USB device
[  850.211220] usb 2-1: new high-speed USB device number 114 using xhci_hcd
[  850.361466] usb 2-1: device descriptor read/64, error 18
[  850.631435] usb 2-1: device descriptor read/64, error 18
[  850.901216] usb 2-1: new high-speed USB device number 115 using xhci_hcd
[  851.051436] usb 2-1: device descriptor read/64, error 18
[  851.321432] usb 2-1: device descriptor read/64, error 18
[  851.441322] usb usb2-port1: attempt power cycle
[  852.151217] usb 2-1: new high-speed USB device number 116 using xhci_hcd
[  852.181622] usb 2-1: device descriptor read/8, error -61
[  852.331627] usb 2-1: device descriptor read/8, error -61
[  852.601216] usb 2-1: new high-speed USB device number 117 using xhci_hcd
[  852.631646] usb 2-1: device descriptor read/8, error -61
[  852.781639] usb 2-1: device descriptor read/8, error -61
[  852.901263] usb usb2-port1: unable to enumerate USB device
```

(over and over again)

## Steps to Reproduce the Problem

  1. flash `openmptcprouter-v0.51.5-r010518-6d61fcfa9e-brcm2708-bcm2711-rpi-4-ext4-factory.img.gz` to SD
  2. Boot RPI4 4GB with SD
  3. configure ssh access
  4. check `dmesg`via ssh

## Specifications

  - OpenMPTCProuter version: `v0.51.5-r010518-6d61fcfa9e-brcm2708-bcm2711-rpi-4-ext4-factory.img.gz`

Ysurac, 2019-08-17:
It was a bug in RPI kernel, this will be fixed in next kernel (I hope...)

rmoriz, 2019-08-17:
Thanks!

Just found a newer release at https://www.openmptcprouter.com/release/4.19/rpi4/targets/brcm2708/bcm2711/ which also seems to support wireless.

The error is also gone with that build.

closed 2019-08-17 by rmoriz

Ysurac, 2019-08-17:
It's a pre-release, I'm testing it.

rmoriz, 2019-08-18:
Looks like the RPi4 upstream kernel is still buggy in terms of DNA access unrelated to OpenMPTCProuter:

I used the pre-release mentioned above, upgraded all packages with opkg (fixes the lua error), successfully connected to my OpenMPTCProuter VPS install.

I've then connected my Macbook through eth0, a Huawei E5573 via USB (eth1) and added my existing wifi as upstream. As soon as I max out the bandwidth using speedtest.net, the wifi connection drops, showing dmesg:

```
[ 1397.060660] bcm2835-dma fe007000.dma: swiotlb buffer is full (sz: 1536 bytes)
[ 1397.060678] bcm2835-dma fe007000.dma: DMA: Out of SW-IOMMU space for 1536 bytes
[ 1397.060688] mmc-bcm2835 fe300000.mmcnr: dma_map_sg returned zero length
```

I'm pretty sure this is an open kernel issue for Raspberry Pi 4 and related to https://github.com/raspberrypi/linux/issues/3120 and https://github.com/raspberrypi/linux/issues/3032

Ysurac, 2019-08-18:
True, should be fixed but doesn't seems to be the case yet...
You can also try to add "sdhci.debug_quirks=96" in /boot/cmdline.txt