Ysurac/openmptcprouter #606: iPhone USB Tethering

GENERATION2GEEK, 2019-08-19:
## Expected Behavior

Share the connection of my iPhone in USB to RPI for 4G + ADSL aggregation

## Actual Behavior

I can't share the 4G connection from my iPhone to my RaspBerry Pi 3 on OpenMPTCProuter
The iPhone is not even detected on the list of network interfaces and for him, no device is connected to his sharing.
I tried this guide: https://openwrt.org/docs/guide-user/network/wan/smartphone.usb.tethering
But it is impossible to install the packages (Cannot satisfy the following dependencies for...)
Can someone help me?

[image]

## Steps to Reproduce the Problem

  1. Connect the iPhone via USB to the Pi
  2. Enable connection sharing

## Specifications

  - OpenMPTCProuter version: 0.51.5
  - OpenMPTCProuter VPS version: 0.1000
  - OpenMPTCProuter platform: RPI3/x86_64

ArChlOgY, 2019-08-19:
can you post the kernel log after you have connected your Iphone to Pi ?

Ysurac, 2019-08-20:
Packets are already available in OpenMPTCProuter, no need to update them.

GENERATION2GEEK, 2019-08-20:
Okay but how do I make this connection sharing work via USB then and how do I get the kernel logs (I think I know but it's to be sure)?

Ysurac, 2019-08-20:
For kernel log, it's in "Status->Kernel log".
For USB tethering, read OpenWrt doc, I can't help more: I don't do that.

ArChlOgY, 2019-08-20:
Plug you phone (activate connection sharing)

Go to Wizard and add an interface with below params (save & apply)

[image]

Execute from SSH : /usr/sbin/usbmuxd -v

Unlock your phone and you should see the trust-question , use your unlock code to confirm

You should see the blue line a the top of your phone screen confirming that your are sharing internet

Come back on OMR interface and enjoy

GENERATION2GEEK, 2019-08-20:
Thank you very much for your help, I solved my problem by installing the following script: https://github.com/LeJeko/OpenWRT-USB-Tethering (beware the script changes the ip of the Raspberry to 192.168.0.254)

Ysurac, 2019-08-20:
The script, in our case, seems to only change IP and run usbmuxd.

GENERATION2GEEK, 2019-08-20:
Yes, I don't know why but it worked perfectly with me

closed 2019-08-20 by Ysurac