Ysurac/openmptcprouter #608: LTE + VDSL bonding not working

max4186, 2019-08-20:
## Expected Behavior

I am using a RPI 4B running 32bit version on ext4 and essentially, everything is connected and green in the status menu but when downloading it absolutely refuses to go down 2 lines. It switches between both every so often, but generally staying on the master MPTCP wan1 which is my VDSL connection. I am using Ubuntu 18.04 64bit as the VPS and it installed with no issues.

Any advice?

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.51.5 r0+10518-6d61fcfa9e / LuCI Master (git-19.212.68203-0291791)
  - OpenMPTCProuter VPS version: < OpenMPTCProuter VPS 0.1000 >
  - OpenMPTCProuter platform: (RPI4/x86/)

Ysurac, 2019-08-20:
"ss-server" is the name of the binary used by shadowsocks-libev server part.

The latest error on the VPS "ERROR: bind: Address already in" is not good. It's a fresh install ? If yes reboot the VPS.

To test speed, use ```omr-test-speed``` command via SSH on the router (and ```omr-test-speed wan1``` to test wan1, same for wan2).

max4186, 2019-08-20:
> "ss-server" is the name of the binary used by shadowsocks-libev server part.
>
> The latest error on the VPS "ERROR: bind: Address already in" is not good. It's a fresh install ? If yes reboot the [...]

It is a completely fresh install on Ubuntu and I did reboot it already, the shadowsocks-libev service is showing already binded because ss-server is running. I run that command it's only showing the speed of wan1 unfortunately.

Ysurac, 2019-08-20:
You are running the service manually ?

max4186, 2019-08-20:
> You are running the service manually ?

No, I just type 'reboot' and ss-service automatically starts and the actual shadowsocks-libev.service  (as in service shadowsocks-libev status) shows as disabled and the status is already binded.

Ysurac, 2019-08-20:
The service used is "shadowsocks-libev-server@"

max4186, 2019-08-20:
I don't get that:

root@bonding:~# service shadowsocks-libev-server status
Unit shadowsocks-libev-server.service could not be found.

netstat has:

tcp        0      0 0.0.0.0:65101           0.0.0.0:*               LISTEN      466/ss-server

then:

root@bonding:~# service shadowsocks-libev status
● shadowsocks-libev.service - Shadowsocks-libev Default Server Service
   Loaded: loaded (/lib/systemd/system/shadowsocks-libev.service; disabled; vendor preset: enabled)
   Active: inactive (dead)
     Docs: man:shadowsocks-libev(8

Ysurac, 2019-08-20:
it's really "shadowsocks-libev-server@config" and "shadowsocks-libev-server@config2" if 2 cores are available.

I would need the result of ```omr-test-speed wan1```, ```omr-test-speed wan2``` and ```omr-test-speed```

max4186, 2019-08-20:
root@OpenMPTCProuter:~# omr-test-speed wan1
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0 35.2M    0     0  4842k      0  0:36:05  0:00:07  0:35:58 4948k

root@OpenMPTCProuter:~# omr-test-speed wan2
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0 61.2M    0     0  6365k      0  0:27:27  0:00:09  0:27:18 7331k

root@OpenMPTCProuter:~# omr-test-speed
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0 46.2M    0     0  4925k      0  0:35:28  0:00:09  0:35:19 4994k

max4186, 2019-08-20:
I am receiving the following error on the @config status service:

Aug 20 07:23:20 bonding ss-server[466]:  2019-08-20 07:23:20 ERROR: server recv: Connection reset by peer
Aug 20 07:23:22 bonding ss-server[466]:  2019-08-20 07:23:22 ERROR: server recv: Connection reset by peer
Aug 20 07:23:23 bonding ss-server[466]:  2019-08-20 07:23:23 ERROR: server recv: Connection reset by peer
Aug 20 07:23:26 bonding ss-server[466]:  2019-08-20 07:23:26 ERROR: server recv: Connection reset by peer
Aug 20 07:23:26 bonding ss-server[466]:  2019-08-20 07:23:26 ERROR: server recv: Connection reset by peer
Aug 20 07:23:41 bonding ss-server[466]:  2019-08-20 07:23:41 ERROR: server recv: Connection reset by peer
Aug 20 07:24:10 bonding ss-server[466]:  2019-08-20 07:24:10 ERROR: server recv: Connection reset by peer
Aug 20 07:27:08 bonding ss-server[466]:  2019-08-20 07:27:08 ERROR: server recv: Connection reset by peer
Aug 20 07:29:13 bonding ss-server[466]:  2019-08-20 07:29:13 ERROR: remote recv: Connection reset by peer
Aug 20 07:31:54 bonding ss-server[466]:  2019-08-20 07:31:54 ERROR: remote recv: Connection reset by peer

Ysurac, 2019-08-20:
Could you check router system log ?
I would also need the config from "All router settings" without keys and/or VPS IP and screenshot of status page.

max4186, 2019-08-20:
Screenshot: https://i.cyrex.io/ReXkulx.png

dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].quietdhcp='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.cot/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1' '/all/omr_dst_bypass_all,omr6_dst_bypass_all'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.ra_default='0'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].Port='22'
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].Interface='lan'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs5'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs5'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
dscp.@classify[2]=classify
dscp.@classify[2].direction='both'
dscp.@classify[2].proto='tcp'
dscp.@classify[2].class='cs5'
dscp.@classify[2].src_port='53,5353'
dscp.@classify[2].comment='DNS tcp'
dscp.@classify[3]=classify
dscp.@classify[3].direction='both'
dscp.@classify[3].proto='tcp'
dscp.@classify[3].class='cs6'
dscp.@classify[3].dest_port='65001,65301'
dscp.@classify[3].comment='OMR vpn'
dscp.@classify[4]=classify
dscp.@classify[4].direction='both'
dscp.@classify[4].proto='udp'
dscp.@classify[4].class='cs6'
dscp.@classify[4].dest_port='65001'
dscp.@classify[4].comment='OMR vpn'
dscp.@domains[0]=domains
dscp.@domains[0].name='googlevideo.com'
[... 1380 lines cut ...]
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1' 'wan2'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.channel='36'
wireless.radio0.hwmode='11a'
wireless.radio0.path='platform/soc/3f300000.mmc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.htmode='VHT80'
wireless.radio0.disabled='1'
wireless.radio1=wifi-device
wireless.radio1.type='mac80211'
wireless.radio1.channel='36'
wireless.radio1.hwmode='11a'
wireless.radio1.path='platform/soc/3f300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio1.htmode='VHT80'
wireless.radio1.disabled='1'
wireless.default_radio1=wifi-iface
wireless.default_radio1.device='radio1'
wireless.default_radio1.network='lan'
wireless.default_radio1.mode='ap'
wireless.default_radio1.ssid='OpenWrt'
wireless.default_radio1.encryption='none'

max4186, 2019-08-20:
I remember the last time I used this there were options for MLVPN under Network or something like that, I can't see that anymore, was it removed?

Ysurac, 2019-08-20:
MLVPN is removed.
You have many think strange in your configuration... It's an old config imported and modified ?
You shouldn't have routes, DNS or rules in network settings.

max4186, 2019-08-20:
It's a super old version that I upgraded using the sysupgrade feature as I currently am unable to flash a fresh copy, any way to reset all settings or something like that?

Ysurac, 2019-08-20:
Not really. You really have to do a fresh install.

max4186, 2019-08-20:
Ok, thanks for your help. I'll try and get it flashed later and let you know if it solves it.

max4186, 2019-08-20:
I've now done the reinstall and on my wan2 I am getting error which I've never had before:

'Multipath seems to be blocking the connection'.

Any suggestions? @Ysurac

max4186, 2019-08-20:
1: 192.168.8.1 
2: * 
3: 192.168.213.21  [PARTIAL] +PartialTCP IP::DiffServicesCP IP::TTL IP::CheckSum 
4: 192.168.213.22  [PARTIAL] +PartialTCP IP::DiffServicesCP IP::TTL IP::CheckSum 
5: * 
6: * 
7: 63.130.105.130 IP::DiffServicesCP IP::TTL IP::CheckSum 
8: 195.66.226.180 IP::TTL IP::CheckSum 
9: 5.254.107.85  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
10: * 
11: 185.53.57.112 TCP::SrcPort TCP::DstPort TCP::SeqNumber TCP::AckNumber TCP::DataOffset TCP::Flags TCP::WindowsSize TCP::CheckSum IP::DiffServicesCP IP::TotalLength IP::Identification IP::TTL IP::CheckSum IP::SourceIP IP::DestinationIP +TCPOptionMaxSegSize +TCPOptionPad -TCPOptionMPTCPCapable 

Result:
< Ethernet (14 bytes) :: DestinationMAC = b2:d0:ef:5b:8c:52 , SourceMAC = 88:11:96:2e:cb:c9 , Type = 0x800 , >
< IP (20 bytes) :: Version = 4 , HeaderLength = 5 , DiffServicesCP = 10 , ExpCongestionNot = 0 , TotalLength = 56 , Identification = 0x0 , Flags = 2 , FragmentOffset = 0 , TTL = 52 , Protocol = 0x6 , CheckSum = 0x8b48 , SourceIP = 185.53.57.112 , DestinationIP = 192.168.8.2 , >
< TCP (20 bytes) :: SrcPort = 65101 , DstPort = 45808 , SeqNumber = 3760517662 , AckNumber = 1595722161 , DataOffset = 9 , Reserved = 0 , Flags = ( SYN ACK ) , WindowsSize = 28800 , CheckSum = 0x8a6f , UrgPointer = 0 , >
< TCPOptionMaxSegSize (4 bytes) :: Kind = 2 , Length = 4 , MaxSegSize = 1320 , >
< TCPOptionPad (1 bytes) :: Kind = 1 , >
< TCPOptionPad (1 bytes) :: Kind = 1 , >
< TCPOptionPad (1 bytes) :: Kind = 1 , >
< TCPOptionPad (1 bytes) :: Kind = 1 , >
< TCPOptionPad (1 bytes) :: Kind = 1 , >
< TCPOptionPad (1 bytes) :: Kind = 1 , >
< TCPOptionPad (1 bytes) :: Kind = 1 , >
< TCPOptionPad (1 bytes) :: Kind = 1 , >
< TCPOptionPad (1 bytes) :: Kind = 1 , >
< TCPOptionPad (1 bytes) :: Kind = 1 , >
< TCPOptionPad (1 bytes) :: Kind = 1 , >
< TCPOptionPad (1 bytes) :: Kind = 1 , >

max4186, 2019-08-20:
I managed to resolve that error by resetting the LTE modem but it's still just not working in general :(

max4186, 2019-08-20:
So from what I can tell it looks like Vodafone may block MPTCP worldwide, example of this here:

https://twitter.com/nileshapte2107/status/758667741943435264

Can you think of any way how to hide that sort of data from being detectable?

Bestle, 2019-08-20:
I am using Vodafone in the UK and MPTCP seems to work just fine?

max4186, 2019-08-20:
> I am using Vodafone in the UK and MPTCP seems to work just fine?

Then I am super confused, when I put in the Vodafone SIM it just refuses to bond. When I put in the Three SIM it works fine. The Vodafone SIM also works fine in general when browsing on the LTE modem, but when connected up to Multipath it refuses to work.

Bestle, 2019-08-20:
This is Vodafone bonded with Virgin media:
https://www.speedtest.net/result/8496194980

My virgin alone doesn't get close to that, especially on upload.

max4186, 2019-08-20:
> This is Vodafone bonded with Virgin media:
> https://www.speedtest.net/result/8496194980
> 
> My virgin alone doesn't get close to that, especially on upload.

I'm jealous, I only get 70Mbps with Vodafone in my area on Band 20, looks like you get their 2x2 MIMO stuff that's like 300Mbps.

This is mine with 40Mbps BT and a Three SIM on Band 3. 

https://www.speedtest.net/result/8516558347.png

Bestle, 2019-08-20:
That's fairly decent :) I would suggest going with a completely fresh install and seeing if that resolves your vodafone issue. What are you using for the modems?

max4186, 2019-08-20:
This is a completely fresh install of both the pi router and the actual VPS I am using, I've also factory reset the modem. It's a Huawei B525.

Bestle, 2019-08-20:
Ah... I'm using Huawei B525s as well. That is very odd.

max4186, 2019-08-20:
> Ah... I'm using Huawei B525s as well. That is very odd.

Are you using the new Vodafone unlimited deal or a limited one? I've just put the Vodafone SIM back into the B525 and now it's not working at all, as if they've blocked it. I'd ring and ask but I think it'd be a waste of time.

Bestle, 2019-08-20:
I'm using an "Unlimited Max" sim.

max4186, 2019-08-20:
Same one I'm using.... oh well, I give up. I'm moving property next month and Vodafone is meant to have 4G whilst Three isn't which is why I got this SIM. Pray for me that it works as the house only has 20Mbps VDSL!

closed 2019-12-20 by Ysurac