Ysurac/openmptcprouter #609: Expected speed for bonding 2 FIOS links

Jabroni, 2019-08-20:
Good day, 

Im testing your project to bond 2x 200mbit FIOS connections. However im just getting about 250-300mbit output, not sure if im hitting a limitation or this is expected? 

Running speedtest between my WANs to my VPS (ran on a server owned by me with gigabit connection) gives me almost 100% easily (and response time ~9ms)

On my home Im running openmptcprouter v0.51 r0+10518 inside a qemu vm with beafy specs (AMD Epyc), and on my VPS side I assigned 2 vCPU cores, 4GB ram on a Linux VM.

root@OpenMPTCProuter:~# omr-test-speed wan1
ipset v7.1: The set with the given name does not exist
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 17  9.7G   17 1733M    0     0  24.0M      0  0:06:56  0:01:12  0:05:44 23.9M

root@OpenMPTCProuter:~# omr-test-speed wan2
ipset v7.1: The set with the given name does not exist
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  3  9.7G    3  388M    0     0  23.6M      0  0:07:02  0:00:16  0:06:46 23.4M

root@OpenMPTCProuter:~# omr-test-speed
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 22  9.7G   22 2233M    0     0  26.1M      0  0:06:22  0:01:25  0:04:57 20.0M

(I modified the omr-test-speed to use my Datacenter test files instead ovh which is all across the globe)

Latency reported from the status page is ~8ms, both wans have a green checkmark.

Server reports Version 0.999.4.19.56-mptcp

I have other WANs I would like to add to the bonding (got like 6 FIOS links, other are async, these 2 are both 200down/200up) but guess I would first would like to see a stable 350-400mbit link :)

any recommended parameters I should test ?

Thanks

Bestle, 2019-08-21:
What results do you see using a site like speedtest?

omr-tests-speed gives me about 200mbit but speedtest shows 440mbit (which is what they should be combined). My usenet downloads often top 50MB/s so I know they are aggregating properly. Maybe something is wrong with omr-test-speed?

Jabroni, 2019-08-23:
Similar results as with omr-test-speed.. speeds barely exceeding 300mbit download, uploads a bit slower

closed 2019-12-20 by Ysurac