Ysurac/openmptcprouter #610: High latency when I stop using the internet

max4186, 2019-08-22:
## Expected Behavior

For some odd reason, whenever I am not using the internet (going to sleep). By the time I wake up I am having incredibly latency issues. Example below, bare in mind my normal latency is 18-20ms to this IP normally. I originally thought it was my VPS so I got a new VPS in a new location to test again but it happens whenever I stop using the internet.

Rebooting it solves it immediately (the router).

Reply from 1.1.1.1: bytes=32 time=40ms TTL=59
Reply from 1.1.1.1: bytes=32 time=29ms TTL=59
Reply from 1.1.1.1: bytes=32 time=37ms TTL=59
Reply from 1.1.1.1: bytes=32 time=29ms TTL=59
Reply from 1.1.1.1: bytes=32 time=41ms TTL=59
Reply from 1.1.1.1: bytes=32 time=44ms TTL=59
Reply from 1.1.1.1: bytes=32 time=27ms TTL=59
Reply from 1.1.1.1: bytes=32 time=27ms TTL=59
Reply from 1.1.1.1: bytes=32 time=1070ms TTL=59
Reply from 1.1.1.1: bytes=32 time=302ms TTL=59
Reply from 1.1.1.1: bytes=32 time=27ms TTL=59
Reply from 1.1.1.1: bytes=32 time=37ms TTL=59
Reply from 1.1.1.1: bytes=32 time=45ms TTL=59
Reply from 1.1.1.1: bytes=32 time=35ms TTL=59
Reply from 1.1.1.1: bytes=32 time=28ms TTL=59
Reply from 1.1.1.1: bytes=32 time=29ms TTL=59
Reply from 1.1.1.1: bytes=32 time=35ms TTL=59
Reply from 1.1.1.1: bytes=32 time=37ms TTL=59

AFTER RESTART:

Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Reply from 1.1.1.1: bytes=32 time=24ms TTL=59
Reply from 1.1.1.1: bytes=32 time=23ms TTL=59
Reply from 1.1.1.1: bytes=32 time=24ms TTL=59
Reply from 1.1.1.1: bytes=32 time=24ms TTL=59
Reply from 1.1.1.1: bytes=32 time=24ms TTL=59
Reply from 1.1.1.1: bytes=32 time=24ms TTL=59
Reply from 1.1.1.1: bytes=32 time=24ms TTL=59
Reply from 1.1.1.1: bytes=32 time=24ms TTL=59
Reply from 1.1.1.1: bytes=32 time=24ms TTL=59

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.51.5 r0+10518-6d61fcfa9e / LuCI Master (git-19.212.68203-0291791)
  - OpenMPTCProuter VPS version: < OpenMPTCProuter VPS 0.1000 >
  - OpenMPTCProuter platform: (RPI4/x86/)

max4186, 2019-08-22:
Just happened when I was using the internet so must not just be when I stop using it.

Reply from 1.1.1.1: bytes=32 time=2679ms TTL=59
Reply from 1.1.1.1: bytes=32 time=700ms TTL=59
Reply from 1.1.1.1: bytes=32 time=27ms TTL=59
Reply from 1.1.1.1: bytes=32 time=26ms TTL=59
Reply from 1.1.1.1: bytes=32 time=27ms TTL=59
Reply from 1.1.1.1: bytes=32 time=27ms TTL=59
Reply from 1.1.1.1: bytes=32 time=27ms TTL=59
Reply from 1.1.1.1: bytes=32 time=27ms TTL=59
Reply from 1.1.1.1: bytes=32 time=632ms TTL=59
Reply from 1.1.1.1: bytes=32 time=27ms TTL=59
Reply from 1.1.1.1: bytes=32 time=71ms TTL=59
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Reply from 1.1.1.1: bytes=32 time=24ms TTL=59
Reply from 1.1.1.1: bytes=32 time=24ms TTL=59
Reply from 1.1.1.1: bytes=32 time=24ms TTL=59
Reply from 1.1.1.1: bytes=32 time=23ms TTL=59
Reply from 1.1.1.1: bytes=32 time=25ms TTL=59
Reply from 1.1.1.1: bytes=32 time=24ms TTL=59
Reply from 1.1.1.1: bytes=32 time=24ms TTL=59
Reply from 1.1.1.1: bytes=32 time=24ms TTL=59
Reply from 1.1.1.1: bytes=32 time=24ms TTL=59
Reply from 1.1.1.1: bytes=32 time=23ms TTL=59
Reply from 1.1.1.1: bytes=32 time=24ms TTL=59
Reply from 1.1.1.1: bytes=32 time=23ms TTL=59

max4186, 2019-08-22:
It appears to be happening whenever I watch a YouTube video, HARD latency increase. But when I do a simple HTTP download that maxes my connection my latency only goes up to a steady 50ms which is normal as it's going over multipath LTE.

Ysurac, 2019-08-22:
The 1.1.1.1 address is used by omr-tracker to test when wan is up or down. So each wan can be used to reach out this address. So it's not a good test.

max4186, 2019-08-23:
> The 1.1.1.1 address is used by omr-tracker to test when wan is up or down. So each wan can be used to reach out this address. So it's not a good test.

I get the same issue across other ones as well, I've found other issues with GloryTUN TCP that seem to have the same issue (high latency). I will say however that switching to UDP has completely fixed it so I don't really care anymore providing there is no security issues (like no more encryption for example).

max4186, 2019-08-30:
The issue with GloryTUN UDP is that after some time, 6-12 hours, my ping seems to increase to a jitter level of about 2-20 when my normal connection is 0.5 or under. Rebooting the Pi sorts this but I do wonder why it's doing this to begin with.

hochsaison, 2019-09-26:
I have EXACTLY the same behavior bonding two or three LTE connections together. Every single of these have appx. 30-50 MBit down/30-50 MBit up, but Ping times increase drastically during the day. Downloading is no problem at all, but Streaming while watching YouTube or so or simply browsing, while sideloading many assets on a website destroys responsiveness totally. Ping times increase radically to 300 - 3000 ms after some time. Upload bandwidth also decreases with time. Every few minutes or so the connection totally collapses. Rebooting OMR resolves this issue for minutes or hours, this varies a lot.

This happened on Raspberry Pi initially. Because of that I upgraded to an Gigabyte Brix (Atom), then Microserver Gen8 with Ivy Bridge i5, then a Ryzen 7 2700X with dedicated network cards just to eliminate the possibility of any kind of bottleneck. 

I updated from 0.43 directly to 0.51, reverting to 0.43 solved that problem for me.

hochsaison, 2019-09-26:
And yes, all three concurrent connections are from different carriers of different frequencies. ;)

max4186, 2019-09-27:
Interesting. I will say also in previous versions I didn't have this issue either and it has only started in the new one so I will roll back as well and see if it fixes it for me.

hochsaison, 2019-09-27:
I just tested a few things with the new version again.

Downloading and streaming seems to degrade by a huge factor, it does not matter at all which VPN i choose inside the Wizard.

I show you what happens after while running `omr-test-speed` on OMR and on a client behind the router, right after a fresh install:

**After 3 seconds:**

OMR
[image]

Client
[image]

**After 10 seconds:**

OMR
[image]

Client (this varies a lot! sometimes <1M, sometimes 5M, but never the full 12M):
[image]

After 1 hour of uptime browsing and any use beyond simple one-connection-use-cases (downloads) gets worse, in fact is fairly unusable, connections drop all over the place.

hochsaison, 2019-09-27:
Oh, wow, Using WARP VPN from CloudFlare through the OMR VPN seems to sort this out somehow. The bandwidth is capped to 20MBit down, but the connection drops, timeouts and horrible ping times disappear.

hochsaison, 2019-09-27:
I mean connection drops like this one:

[image]

Ysurac, 2019-09-27:
I don't think it's the same issue. Speed seems to be good on OMR, but bad on your client if I understand all ?
How do you connect to OMR ? Your client have an IPv6 or not ?

hochsaison, 2019-09-27:
The internal network is solely on IPv4.

Yes, it seems so... the server is working just fine, but the clients suffer from extreme lag, extremely increased ping times and unstable connections.

I can't really tell in which version this occured the first time, but alternating between the versions seems to work this out.

hochsaison, 2019-09-27:
Erm... and the clients are connected through 1G or 2.5G Ethernet :)

Ysurac, 2019-09-27:
In network->firewall, "Routing/NAT Offloading" and "Software flow offloading" are disabled ?
What is the chipset used by your network card ?

hochsaison, 2019-09-27:
"Routing/NAT Offloading" does not even show a checkbox to enable or disable, "Software flow offloading" is disabled.

Actually, everything is disabled as by default.

Do you mean the chipset of the Router? Right now it's an Aquantia AQtion AQC107, but as already stated I tried this on many devices, it's the same everywhere (Raspi, Brix, Microserver — with internal and also external Intel Dual Gigabit Network card...).

On client side I tested this on various devies. The lag appears from a 5820k@X99 with RTL8139, a MacBook 2015, an iMac Pro, an XBox One, an Amazon FireTV and an AppleTV 4gen HD (not 4k). The behavior is the same everywhere.

closed 2019-12-20 by Ysurac