Ysurac/openmptcprouter #618: git master build problem

bournehlm, 2019-08-25:
## Expected Behavior

  I try to build git master on Ubuntu18.04 or Debian 10, both failed. 

Installing package 'luci-proto-qmi' from luci
Installing package 'simple-obfs' from openmptcprouter
Done
Building openmptcprouter for the target rpi3
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libgnutls', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libopenldap', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libidn2', which does not exist
WARNING: Makefile 'package/boot/kexec-tools/Makefile' has a dependency on 'liblzma', which does not exist
WARNING: Makefile 'package/network/utils/nftables/Makefile' has a dependency on 'jansson', which does not exist
#
# configuration written to .config
#
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libgnutls', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libopenldap', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libidn2', which does not exist
WARNING: Makefile 'package/boot/kexec-tools/Makefile' has a dependency on 'liblzma', which does not exist
WARNING: Makefile 'package/network/utils/nftables/Makefile' has a dependency on 'jansson', which does not exist
time: target/linux/prereq#0.11#0.04#0.15
 make[1] world
 make[2] tools/compile
 make[3] -C tools/flock compile
 make[3] -C tools/xz compile
 make[3] -C tools/sed compile
 make[3] -C tools/patch compile
 make[3] -C tools/tar compile
>>>>Stuck at here.

## Actual Behavior

## Steps to Reproduce the Problem

  1.Install new Ubuntu18.04 or Debain 10 virtual machine.
  2.sudo apt install build-essential git unzip ncurses-dev libz-dev libssl-dev python subversion gettext gawk wget curl rsync perl
  3.git clone https://github.com/ysurac/openmptcprouter.git
  4.cd openmptcprouter
  5.OMR_TARGET="rpi3" ./build.sh

## Specifications

  - OpenMPTCProuter version: (git master)
  - OpenMPTCProuter VPS version: (git master)
  - OpenMPTCProuter platform: (RPI3)

Ysurac, 2019-08-25:
```./build.sh -j1 V=s``` or ```make -j1 V=s``` in the rpi3/source subdirectory to know what he is doing.

bournehlm, 2019-08-25:
After run OMR_TARGET="rpi3" ./build.sh -j1 V=s, still have Errors:

. /home/ubuntu/openmptcprouter/rpi3/source/include/shell.sh; gzip -dc /home/ubuntu/openmptcprouter/rpi3/source/dl/scons-3.1.1.tar.gz | tar -C /home/ubuntu/openmptcprouter/rpi3/source/build_dir/host/scons-3.1.1/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/* /home/ubuntu/openmptcprouter/rpi3/source/build_dir/host/scons-3.1.1

Applying ./patches/001-platform_env.patch using plaintext: 
patching file engine/SCons/Platform/__init__.py
touch /home/ubuntu/openmptcprouter/rpi3/source/build_dir/host/scons-3.1.1/.prepared4b37c6e03f8f14b49fe18e9648ea97f7_6664517399ebbbc92a37c5bb081b5c53
touch /home/ubuntu/openmptcprouter/rpi3/source/build_dir/host/scons-3.1.1/.configured
touch /home/ubuntu/openmptcprouter/rpi3/source/build_dir/host/scons-3.1.1/.built
./files/pywrap.sh /home/ubuntu/openmptcprouter/rpi3/source/build_dir/host/scons-3.1.1/setup.py install --prefix=/home/ubuntu/openmptcprouter/rpi3/source/staging_dir/host
Traceback (most recent call last):
  File "/home/ubuntu/openmptcprouter/rpi3/source/build_dir/host/scons-3.1.1/setup.py", line 26, in <module>
    import distutils.command.build_scripts
ModuleNotFoundError: No module named 'distutils.command'
Makefile:35: recipe for target '/home/ubuntu/openmptcprouter/rpi3/source/staging_dir/host/stamp/.scons_installed' failed
make[3]: *** [/home/ubuntu/openmptcprouter/rpi3/source/staging_dir/host/stamp/.scons_installed] Error 1
make[3]: Leaving directory '/home/ubuntu/openmptcprouter/rpi3/source/tools/scons'
time: tools/scons/compile#0.28#0.06#3.40
tools/Makefile:157: recipe for target 'tools/scons/compile' failed
make[2]: *** [tools/scons/compile] Error 2
make[2]: Leaving directory '/home/ubuntu/openmptcprouter/rpi3/source'
tools/Makefile:155: recipe for target '/home/ubuntu/openmptcprouter/rpi3/source/staging_dir/target-aarch64_cortex-a53_musl/stamp/.tools_compile_yynyynnyyynyyyyynyynnyyyynyyyyyyyyyyyyyyynyynynnyyynnyyy' failed
make[1]: *** [/home/ubuntu/openmptcprouter/rpi3/source/staging_dir/target-aarch64_cortex-a53_musl/stamp/.tools_compile_yynyynnyyynyyyyynyynnyyyynyyyyyyyyyyyyyyynyynynnyyynnyyy] Error 2
make[1]: Leaving directory '/home/ubuntu/openmptcprouter/rpi3/source'
/home/ubuntu/openmptcprouter/rpi3/source/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

Ysurac, 2019-08-25:
```apt-get install python3-distutils``` ?

bournehlm, 2019-08-25:
Yes, that resolve my problem, I can build success now, Thanks for your help!

closed 2019-08-25 by bournehlm