Ysurac/openmptcprouter #619: Speed very very tres slow

yoplaitXYZ, 2019-08-25:
Hello,

I saw several topics.
I think I have a different problem.
I have all that's green. All seem to be working.
But the speed is incredibly slow.
When I test the speed of the boxes independently I have:

WAN 1: 50m FREEMOBILE 4G
WAN 2 : 6m FREE ADSL
WAN 3: 70m NRJMOBILE
[image]

on the raspberry I have

root@OpenMPTCProuter:~# omr-test-speed **wan1**
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0 80.0M    0     0  7759k      0  0:22:31  0:00:10  0:22:21 7802k^C
root@OpenMPTCProuter:~# omr-test-speed **wan2**
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0 7079k    0     0   697k      0  4:10:38  0:00:10  4:10:28  707k^C
root@OpenMPTCProuter:~# omr-test-speed wan3
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0 84.5M    0     0  8562k      0  0:20:24  0:00:10  0:20:14 9628k^C
root@OpenMPTCProuter:~# omr-test-speed **wan3**
^Cipset v7.1: Missing second mandatory argument to command add

et WAN 

root@OpenMPTCProuter:~# omr-test-speed
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0 53.7M    0     0  5184k      0  0:33:42  0:00:10  0:33:32 6481k^C

Very weird no ?
thank you for your help
## Specifications

  - OpenMPTCProuter version: (0.51.5)
  - OpenMPTCProuter VPS version: (0.999)
  - OpenMPTCProuter platform: (RPI3/x86)

Ysurac, 2019-08-25:
What is the speed of the VPS ?

yoplaitXYZ, 2019-08-25:
I have two vps :
 OVH 100m
 an other 
ONETSOLUTION 100m

i have the probleme on both

Ysurac, 2019-08-25:
For test you should spend more than 10s...

max4186, 2019-08-25:
This is similar to the issue I had before, have you done a completely fresh install of everything? Can you also try using GloryTUN UDP rather than TCP?

yoplaitXYZ, 2019-08-26:
I've done a lot of testing.
I noticed that when I cut my adsl connection and reconnect, my connection starts again at 80mbps
but a few hours later she stayed at 6mbps.
I have the impression that ADSL ( freebox) defaults my other two 4G connections.
Have you ever had a similar case?

yoplaitXYZ, 2019-08-26:
> This is similar to the issue I had before, have you done a completely fresh install of everything? Can you also try using GloryTUN UDP rather than TCP?

I tried several installations ,
I will test GLoryTUN UDP.

yoplaitXYZ, 2019-08-26:
I think i found the problem.
you have to active IPv6 option in system/openMPTCrouter
and change next hop ipv6 by fe80::2 in my adsl router
thank you

closed 2019-08-26 by yoplaitXYZ