Ysurac/openmptcprouter #621: GloryTun TCP problem.

Marctraider, 2019-08-27:
## Expected Behavior

Since I assume Glorytun TCP use MPTCP, I expect UDP streams not to fail (or at least quick redundant failover) when turning a random WAN interface down

## Actual Behavior

With 3 WANs up, I disabled WAN3 and the whole UDP streams all stop for 5+ seconds. terminating all UDP connections and ICMP pings.

Seems like subflows not functioning?

Ysurac, 2019-08-27:
You should look in OpenMPTCProuter system log.
And what is the mode you choose for Master interface ?

Marctraider, 2019-08-27:
> 
> 
> You should look in OpenMPTCProuter system log.
> And what is the mode you choose for Master interface ?

What do you mean mode for Master interface, isn't the mode 'Master' ?

Ysurac, 2019-08-27:
In OpenMPTCProuter->Advanced settings tab.

Marctraider, 2019-08-27:
> 
> 
> In OpenMPTCProuter->Advanced settings tab.

If you mean 'Master Interface selection' I have it set to 'No Change'.

Marctraider, 2019-08-27:
`Tue Aug 27 18:02:22 2019 user.notice post-tracking-post-tracking: Multipath eth0.3 switched to off
Tue Aug 27 18:02:24 2019 user.notice post-tracking-post-tracking: Multipath eth0.2 switched to off
Tue Aug 27 18:02:24 2019 user.notice post-tracking-post-tracking: Multipath eth0.1 switched to off
Tue Aug 27 18:02:34 2019 user.notice MPTCP: Multipath not defined for lan set to off
Tue Aug 27 18:02:34 2019 user.notice MPTCP: Multipath not defined for wan1 set to on
Tue Aug 27 18:02:34 2019 user.notice MPTCP: Multipath not defined for wan2 set to on
Tue Aug 27 18:02:34 2019 user.notice MPTCP: Multipath not defined for wan3 set to on
Tue Aug 27 18:02:34 2019 user.notice MPTCP: Multipath not defined for omrvpn set to off
Tue Aug 27 18:02:34 2019 user.notice MPTCP: Multipath not defined for omr6in4 set to off
Tue Aug 27 18:02:34 2019 user.notice MPTCP: No master multipath defined, setting it to eth0.1
Tue Aug 27 18:02:34 2019 user.notice omr-tracker: Launching...
Tue Aug 27 18:02:34 2019 user.notice omr-tracker: Launched
Tue Aug 27 18:02:35 2019 daemon.info glorytun: starting glorytun vpn instance vpn
Tue Aug 27 18:02:35 2019 daemon.info omr-tracker-ss: Shadowsocks is up (can contact 1.0.0.1)
Tue Aug 27 18:02:35 2019 daemon.notice netifd: Interface 'wan3' is now down
Tue Aug 27 18:02:35 2019 kern.info kernel: [60104.346038] IPv6: ADDRCONF(NETDEV_UP): eth0.3: link is not ready
Tue Aug 27 18:02:35 2019 daemon.notice netifd: Interface 'wan3' is disabled
Tue Aug 27 18:02:35 2019 daemon.notice netifd: Interface 'wan3' has link connectivity loss
Tue Aug 27 18:02:35 2019 daemon.notice netifd: Interface 'wan3' has link connectivity
Tue Aug 27 18:02:35 2019 daemon.notice netifd: VLAN 'eth0.3' link is down
Tue Aug 27 18:02:35 2019 daemon.notice netifd: Interface 'wan3' has link connectivity loss
Tue Aug 27 18:02:38 2019 user.notice omr-bypass: Starting OMR-ByPass...
Tue Aug 27 18:02:41 2019 user.notice omr-bypass: Restart dnsmasq...
Tue Aug 27 18:02:41 2019 daemon.info dnsmasq[14910]: exiting on receipt of SIGTERM
Tue Aug 27 18:02:41 2019 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Tue Aug 27 18:02:41 2019 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Tue Aug 27 18:02:41 2019 user.notice dnsmasq: Allowing RFC1918 responses for domain plex.direct
Tue Aug 27 18:02:44 2019 daemon.info dnsmasq[11252]: started, version 2.80 cachesize 150
Tue Aug 27 18:02:44 2019 daemon.info dnsmasq[11252]: DNS service limited to local subnets
Tue Aug 27 18:02:44 2019 daemon.info dnsmasq[11252]: compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Tue Aug 27 18:02:44 2019 daemon.info dnsmasq-dhcp[11252]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Tue Aug 27 18:02:44 2019 daemon.info dnsmasq[11252]: using local addresses only for domain test
Tue Aug 27 18:02:44 2019 daemon.info dnsmasq[11252]: using local addresses only for domain onion
Tue Aug 27 18:02:44 2019 daemon.info dnsmasq[11252]: using local addresses only for domain localhost
Tue Aug 27 18:02:44 2019 daemon.info dnsmasq[11252]: using local addresses only for domain local
Tue Aug 27 18:02:44 2019 daemon.info dnsmasq[11252]: using local addresses only for domain invalid
Tue Aug 27 18:02:44 2019 daemon.info dnsmasq[11252]: using local addresses only for domain bind
Tue Aug 27 18:02:44 2019 daemon.info dnsmasq[11252]: using nameserver 8.8.8.8#53
Tue Aug 27 18:02:44 2019 daemon.info dnsmasq[11252]: using nameserver 127.0.0.1#5353
Tue Aug 27 18:02:44 2019 daemon.info dnsmasq[11252]: using local addresses only for domain lan
Tue Aug 27 18:02:44 2019 daemon.info dnsmasq[11252]: read /etc/hosts - 4 addresses
Tue Aug 27 18:02:44 2019 daemon.info dnsmasq[11252]: read /tmp/hosts/dhcp.cfg01411c - 3 addresses
Tue Aug 27 18:02:44 2019 daemon.info dnsmasq-dhcp[11252]: read /etc/ethers - 0 addresses
Tue Aug 27 18:02:44 2019 user.notice omr-bypass: OMR-ByPass is running
Tue Aug 27 18:02:51 2019 daemon.info vnstatd[16680]: Info: Interface "eth0.3" disabled.
Tue Aug 27 18:04:09 2019 daemon.info glorytun: starting glorytun vpn instance vpn
Tue Aug 27 18:04:09 2019 user.notice omr-tracker: Launching...
Tue Aug 27 18:04:09 2019 user.notice omr-tracker: Launched
Tue Aug 27 18:04:09 2019 user.notice post-tracking-post-tracking: Multipath eth0.2 switched to off
Tue Aug 27 18:04:09 2019 user.notice MPTCP: Multipath not defined for lan set to off
Tue Aug 27 18:04:09 2019 user.notice MPTCP: Multipath not defined for wan1 set to on
Tue Aug 27 18:04:09 2019 user.notice MPTCP: Multipath not defined for wan2 set to on
Tue Aug 27 18:04:09 2019 daemon.info omr-tracker-ss: Shadowsocks is up (can contact 1.0.0.1)
Tue Aug 27 18:04:09 2019 user.notice MPTCP: Multipath not defined for wan3 set to on
Tue Aug 27 18:04:10 2019 user.notice MPTCP: Multipath not defined for omrvpn set to off
Tue Aug 27 18:04:10 2019 user.notice MPTCP: Multipath not defined for omr6in4 set to off
Tue Aug 27 18:04:10 2019 user.notice MPTCP: No master multipath defined, setting it to eth0.1
Tue Aug 27 18:04:11 2019 daemon.notice netifd: Interface 'wan3' is enabled
Tue Aug 27 18:04:11 2019 daemon.notice netifd: Interface 'wan3' is setting up now
Tue Aug 27 18:04:11 2019 daemon.notice netifd: Interface 'wan3' is now up
Tue Aug 27 18:04:11 2019 daemon.notice netifd: VLAN 'eth0.3' link is up
[... cut ...]
2019-08-27 18:02:39.383: From 1.1.1.1: bytes=60 seq=0f82 TTL=57 ID=8b34 time=86.729ms
2019-08-27 18:02:40.358: From 1.1.1.1: bytes=60 seq=0f83 TTL=57 ID=8ecc time=60.938ms
2019-08-27 18:02:41.366: From 1.1.1.1: bytes=60 seq=0f84 TTL=57 ID=9196 time=69.428ms
2019-08-27 18:02:42.360: From 1.1.1.1: bytes=60 seq=0f85 TTL=57 ID=9488 time=63.483ms
2019-08-27 18:02:43.359: From 1.1.1.1: bytes=60 seq=0f86 TTL=57 ID=9803 time=62.785ms
2019-08-27 18:02:44.361: From 1.1.1.1: bytes=60 seq=0f87 TTL=57 ID=9a34 time=63.048ms`

Ysurac, 2019-08-27:
There is no problem here : you turn off a network interface via enable/disable in network, every change in network reload MPTCP configuration. I don't think it's really a problem.
You ping 1.1.1.1 and if you didn't remove it, it's an IP used to check if interface are up or down in omr-tracker, so ping using this IP can (will) fail.

closed 2019-08-27 by Ysurac

Marctraider, 2019-08-27:
So its impossible to disable interface without screwing all up?

All UDP streams disconnected when doing this, the ping log just shows part of it...

Ysurac, 2019-08-27:
Disabling an interface via the graphical interface yes, this reset all connections. It's the same when we change any network settings.