Ysurac/openmptcprouter #622: Master setting weirdness

Marctraider, 2019-08-27:
## Expected Behavior

I set my 1st WAN as Master, other two Multipath TCP. I disabled WAN3 and all WAN interfaces seem to reset to Multipath TCP.

## Actual Behavior

WAN1 Sticks at Master when turning WAN3 (or WAN2) interface off. 

  - OpenMPTCProuter version: 0.51 
  - OpenMPTCProuter VPS version: 0.999
  - OpenMPTCProuter platform: WRT3200ACM

Ysurac, 2019-08-27:
Update to latest release.

Marctraider, 2019-08-27:
Sorry typo, im on last version.

Ysurac, 2019-08-27:
How do you disable WAN interface ?

Marctraider, 2019-08-27:
> 
> 
> How do you disable WAN interface ?

Network -> Interfaces -> Stop / Connect

Ysurac, 2019-08-27:
It's a know issue: settings are removed when an interface is stopped. No idea why for now.

Marctraider, 2019-08-27:
Ok thanks.

closed 2019-08-27 by Marctraider