Ysurac/openmptcprouter #623: Wrong log server block natted service

JorgeMoort89, 2019-08-28:
Setting up wrong log server block slow down a lot all the natted services and broken the aggregation.

## Specifications

  - OpenMPTCProuter version: 0.49
  - OpenMPTCProuter VPS version: 0.996
  - OpenMPTCProuter platform: x86_64

Ysurac, 2019-08-28:
I don't understand the problem.

JorgeMoort89, 2019-08-28:
If you set an IP in the remote logging server and on that IP there is no working logging server, it will constantly fail to send the log generating a lot of issues, all the natted service stop working/slow down a lot and the router will not aggregate more.

Ysurac, 2019-08-29:
It's the problem when using a TCP logging server.
This may be better by replacing default logd by syslog-ng but I will not do that for now.

closed 2019-08-29 by Ysurac