Ysurac/openmptcprouter #624: VPS error

vistman, 2019-08-28:
## Expected Behavior
System -> OpenMPTCProuter -> Settings -> VPS status - OK

## Actual Behavior
System -> OpenMPTCProuter -> Settings -> VPS status
No output
No VPS IP address, No WAN IP address
Can't get public IP address from ShadowSocks

I haven't changed anything in my config and for few days now I get error on VPS status. Funny thing is everything seems to be working ie. I have Internet access. Since I have several VPSes I've tried all of them and I get the same error on all of them.

## Steps to Reproduce the Problem

  1. Huh, don't know ...

## Specifications

  - OpenMPTCProuter version: 0.51.5
  - OpenMPTCProuter VPS version: 0.999
  - OpenMPTCProuter platform: RPI3/x86_64

ArChlOgY, 2019-08-28:
I had similar issue , check that in OpenMPTCProuter system config the Time is correctly synchronized .

vistman, 2019-08-28:
Thanks. Time was offset indeed. But unfortunately this hasn't resolved my issue.
Found the issue - I've installed stubby for DNS-over-TLS as someone mentioned here and set under Network -> DHCP and DNS -> DNS forwardings to 127.0.0.1#5453 which is obviously blocking OpenMPTCP router to retrieve VPS status properly.

ArChlOgY, 2019-08-28:
Sorry I cant help on that but coming back on Time issue @Ysurac are you able to force something on reboot ?  It happened when I had suddenly an electic shutdown at home for strange reason I had to force synchronizing it to fix reported issue

closed 2019-12-20 by Ysurac