Ysurac/openmptcprouter #626: High throughput ss-server blocks all requests from one IP

d3dx9, 2019-08-28:
## Expected Behavior

ss-server not blocking requests from own IP.

## Actual Behavior

ss-server block own client ip when using high speeds.
Aug 28 15:54:42 openmptcp ss-server[435]:  2019-08-28 15:54:42 ERROR: block all requests from 46.140.xxx.xxx

The other IP doesn't give this out. The connection which gets this errors has 600/60mbit and the other has 60/24mbit.

## Steps to Reproduce the Problem

  1. Use much speed.

## Specifications

  - OpenMPTCProuter version: 0.51.5
  - OpenMPTCProuter VPS version: 0.10000 4.19-56.mptcp
  - OpenMPTCProuter platform: x86_64

d3dx9, 2019-08-28:
> This happen after many handshake problems on shadowsocks.
> I will try to auto whitelist IPs from OpenMPTCProuter clients in a next
> release.

Any Method to do this manually?

Aug. 2019, 19:25:

Ysurac, 2019-08-28:
This happen after many handshake problems on shadowsocks.
I will try to auto whitelist IPs from OpenMPTCProuter clients in a next release.

dur3x, 2019-08-29:
Sometimes I got the same problem.
Like Ysurac said the problem started with some handshake problems and then block all requests.
````
Aug 26 19:35:34 openmptcprouter-vps-55599c7878-cf2p2 ss-server[37]:  2019-08-26 19:35:34 ERROR: failed to handshake with 
10.X.X.X: invalid address type
Aug 26 19:35:44 openmptcprouter-vps-55599c7878-cf2p2 ss-server[37]:  2019-08-26 19:35:44 ERROR: failed to handshake with 
10.X.X.X: invalid host name
Aug 26 19:35:49 openmptcprouter-vps-55599c7878-cf2p2 ss-server[37]:  2019-08-26 19:35:49 ERROR: failed to handshake with 
10.X.X.X: invalid address type
Aug 26 19:35:55 openmptcprouter-vps-55599c7878-cf2p2 ss-server[37]:  2019-08-26 19:35:55 ERROR: failed to handshake with 
10.X.X.X: invalid address type
Aug 26 19:36:01 openmptcprouter-vps-55599c7878-cf2p2 ss-server[37]:  2019-08-26 19:36:01 ERROR: block all requests from 1
0.X.X.X
````
Currently when I detect that I just restart ss-server on the server itself:
````
systemctl restart shadowsocks-libev-server@config.service
systemctl restart shadowsocks-libev-server@config4.service
````

d3dx9, 2019-08-29:
Problem this happens to me everytime after 10-15min of usage.

Ysurac, 2019-08-29:
Time of the router and the VPS are synchronized ?

d3dx9, 2019-08-29:
VM says 21:33 CEST, OpenMPTCPRouter says 19:33 UTC. So I think yes.

Ysurac, 2019-08-30:
This will be solved in next release by using ACL to whitelist IPs. But This still seems strange...

closed 2019-12-20 by Ysurac