Ysurac/openmptcprouter #629: FEC Tunnel Support (Suggestion)

lourensventer, 2019-08-29:
## Expected Behavior

To maybe include support for FEC (Forward Error Correction) via an implementation of UDPSpeeder or TinyFECTunnel for RealTime application optimization over potentially lossy WAN links (Voice/Video would benefit tremendously).

## Actual Behavior

Not currently supported. 

## Steps to Reproduce the Problem

Tried to run TinyFECTunnel alongside OMR VPNs but something causes SIP not to route via the new tunnel interface (even if destination IP is bypassed in OMR rules. Some help/guidance might be appreciated here to fix this issue easily??)

## Specifications

Latest versions of OMR and VPS on x86 servers

suyuan168, 2019-08-30:
DPSpeeder does not support MPTCP

lourensventer, 2019-08-31:
> DPSpeeder does not support MPTCP

I know it doesn't support it, I'm asking for a way to spin up a FEC tunnel to another server (separate from SS/GT), then route SIP/RTP through it.

closed 2019-12-20 by Ysurac