Ysurac/openmptcprouter #630: Utiliser une routeur TP-link ?

Pokpok-sbo, 2019-08-30:
Hello, I would like to test OpenMPTCProuter. I have a TP-link router, model TL-R480 and I would like to know if I can install OpenMPTCProuter on it.
Here is the manufacturer's link: https://www.tp-link.com/fr/business-networking/load-balance-router/tl-r480t+/

Thanks

suyuan168, 2019-08-30:
> Hello, I would like to test OpenMPTCProuter. I have a TP-link router, model TL-R480 and I would like to know if I can install OpenMPTCProuter on it.
> Here is the manufacturer's link: https://www.tp-link.com/fr/business-networking/load-balance-router/tl-r480t+/
> 
> Thanks

His flash is too small to fit openmptcprouter

suyuan168, 2019-08-30:
Moreover, its TL-R480T+ CPU memory configuration is too low, can not run openmptcprouter very well. At least his flash can not be installed,

Ysurac, 2019-08-30:
True: not supported by OpenWrt, slow CPU, small flash, small memory. This can't be supported.

closed 2019-08-30 by Ysurac

Pokpok-sbo, 2019-08-31:
> True: not supported by OpenWrt, slow CPU, small flash, small memory. This can't be supported.

That's a real shame! Thank you for your answers