Ysurac/openmptcprouter #633: PiHole DNS over HTTPS (DoH)

zicochaos, 2019-08-31:
Any chance to add this as option when installing PiHole script 

https://docs.pi-hole.net/guides/dns-over-https/

Thanks 
Sebastian

vistman, 2019-09-01:
This is actually the same request as mine mentioned here - https://github.com/Ysurac/openmptcprouter/issues/624. It would be a nice thing to have implemented. Thanks.

zicochaos, 2019-09-01:
if you follow guide from the link PiHole and everything works fine. DOH works great :) Just few manual steps are needed.

vistman, 2019-09-02:
Thanks for your feedback. Still, DoT as an official standard would be a better option.

zicochaos, 2019-09-19:
> And this procedure needs to be repeated upon each fw upgrade which is not very practical.

you don't need to repeat the procedure. All works after upgrading VPS :)

user747, 2019-11-05:
Another option
https://github.com/pi-hole/pi-hole/wiki/DNSCrypt-2.0

closed 2019-12-20 by Ysurac