Ysurac/openmptcprouter #635: Random reboots with pi 3b+

ragefan, 2019-09-01:
## Expected Behavior

No reboot during normal operation. 

## Actual Behavior

Every few days, random reboot occurs. 
I don't know how to collect error logs. 

## Steps to Reproduce the Problem

  1. Install omr with pi3b+ 
  2. Setup 1 adsl and 1 4g connection
  3.wait

## Specifications

  - OpenMPTCProuter version: 0.51.5
  - OpenMPTCProuter VPS version:. 0.999
  - OpenMPTCProuter platform: RPI3b+

Ysurac, 2019-09-01:
With default configuration ?

ragefan, 2019-09-01:
yes. How to collect some logs ?

Ysurac, 2019-09-01:
Via serial port. But there is no know issues with default configuration.
On current release there is some issue with BLEST scheduler that are fixed in next release.

ragefan, 2019-09-01:
I don't use blest. Need to find out how to log the console on the rpi3+.

I am using default conf as I did not change any scheduler option.

I actually noticed it because the uptime was low. It is hard to detect as reboot is fast.

ragefan, 2019-10-20:
Is there a way to log the crash dump when it happens? Connecting serial is not so easy

closed 2019-12-20 by Ysurac