Ysurac/openmptcprouter #637: Agregation ne fonctionne pas

ivamp, 2019-09-03:
Bonjour,
Pour reprendre la discution sur le forum lafibre voici les test omr speed,
la conf a été fait directement sur le wizard
[image]

Ysurac, 2019-09-03:
Il me faudrait la configuration dispo dans Systèmes->OpenMPTCProuter puis l'onglet "Voir tous les paramètres" en supprimant au moins l'IP du VPS.

ivamp, 2019-09-03:
`dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].enable_tftp='1'
dhcp.@dnsmasq[0].tftp_root='/'
dhcp.@dnsmasq[0].dhcp_boot='PXEClient'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].server='/asc.local/192.168.10.220' '/asccomputer.fr/192.168.10.220' '127.0.0.1#5353'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.cot/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.leasetime='12h'
dhcp.lan.limit='199'
dhcp.lan.dns='192.168.10.220'
dhcp.lan.force='1'
dhcp.lan.dhcp_option='6,192.168.10.220'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dhcp.@host[0]=host
dhcp.@host[0].name='UBIQUITI'
dhcp.@host[0].dns='1'
dhcp.@host[0].mac='18:E8:29:E9:FF:41'
dhcp.@host[0].ip='192.168.0.16'
dhcp.@host[1]=host
dhcp.@host[1].name='ASCSYNO'
dhcp.@host[1].dns='1'
dhcp.@host[1].mac='00:11:32:12:03:63'
dhcp.@host[1].ip='192.168.0.222'
dhcp.@host[2]=host
dhcp.@host[2].name='PCSYLVAIN'
dhcp.@host[2].dns='1'
dhcp.@host[2].mac='60:A4:4C:22:30:61'
dhcp.@host[2].ip='192.168.0.15'
dhcp.@host[3]=host
dhcp.@host[3].name='UNIFI-CLOUDKEY'
dhcp.@host[3].dns='1'
dhcp.@host[3].mac='74:83:C2:19:1E:2B'
dhcp.@host[3].ip='192.168.10.200'
dhcp.@host[4]=host
dhcp.@host[4].name='PLANET'
dhcp.@host[4].dns='1'
dhcp.@host[4].mac='00:30:4F:9B:9E:80'
dhcp.@host[4].ip='192.168.10.210'
dhcp.@host[5]=host
dhcp.@host[5].name='DELL'
dhcp.@host[5].dns='1'
dhcp.@host[5].mac='5C:26:0A:D5:15:AE'
dhcp.@host[5].ip='192.168.0.101'
dhcp.@host[6]=host
dhcp.@host[6].name='DC'
dhcp.@host[6].dns='1'
dhcp.@host[6].mac='00:0C:29:68:E7:60'
[... 1247 lines cut ...]
upnpd.config.enabled='0'
upnpd.config.enable_natpmp='1'
upnpd.config.enable_upnp='1'
upnpd.config.secure_mode='1'
upnpd.config.log_output='0'
upnpd.config.download='1024'
upnpd.config.upload='512'
upnpd.config.internal_iface='lan'
upnpd.config.port='5000'
upnpd.config.upnp_lease_file='/var/run/miniupnpd.leases'
upnpd.config.igdv1='0'
upnpd.@perm_rule[0]=perm_rule
upnpd.@perm_rule[0].action='allow'
upnpd.@perm_rule[0].ext_ports='1024-65535'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1' 'wan2'`

ivamp, 2019-09-03:
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].enable_tftp='1'
dhcp.@dnsmasq[0].tftp_root='/'
dhcp.@dnsmasq[0].dhcp_boot='PXEClient'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].server='/asc.local/192.168.10.220' '/asccomputer.fr/192.168.10.220' '127.0.0.1#5353'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.cot/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.leasetime='12h'
dhcp.lan.limit='199'
dhcp.lan.dns='192.168.10.220'
dhcp.lan.force='1'
dhcp.lan.dhcp_option='6,192.168.10.220'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dhcp.@host[0]=host
dhcp.@host[0].name='UBIQUITI'
dhcp.@host[0].dns='1'
dhcp.@host[0].mac='18:E8:29:E9:FF:41'
dhcp.@host[0].ip='192.168.0.16'
dhcp.@host[1]=host
dhcp.@host[1].name='ASCSYNO'
dhcp.@host[1].dns='1'
dhcp.@host[1].mac='00:11:32:12:03:63'
dhcp.@host[1].ip='192.168.0.222'
dhcp.@host[2]=host
dhcp.@host[2].name='PCSYLVAIN'
dhcp.@host[2].dns='1'
dhcp.@host[2].mac='60:A4:4C:22:30:61'
dhcp.@host[2].ip='192.168.0.15'
dhcp.@host[3]=host
dhcp.@host[3].name='UNIFI-CLOUDKEY'
dhcp.@host[3].dns='1'
dhcp.@host[3].mac='74:83:C2:19:1E:2B'
dhcp.@host[3].ip='192.168.10.200'
dhcp.@host[4]=host
dhcp.@host[4].name='PLANET'
dhcp.@host[4].dns='1'
dhcp.@host[4].mac='00:30:4F:9B:9E:80'
dhcp.@host[4].ip='192.168.10.210'
dhcp.@host[5]=host
dhcp.@host[5].name='DELL'
dhcp.@host[5].dns='1'
dhcp.@host[5].mac='5C:26:0A:D5:15:AE'
dhcp.@host[5].ip='192.168.0.101'
dhcp.@host[6]=host
dhcp.@host[6].name='DC'
dhcp.@host[6].dns='1'
dhcp.@host[6].mac='00:0C:29:68:E7:60'
[... 1247 lines cut ...]
upnpd.config.enabled='0'
upnpd.config.enable_natpmp='1'
upnpd.config.enable_upnp='1'
upnpd.config.secure_mode='1'
upnpd.config.log_output='0'
upnpd.config.download='1024'
upnpd.config.upload='512'
upnpd.config.internal_iface='lan'
upnpd.config.port='5000'
upnpd.config.upnp_lease_file='/var/run/miniupnpd.leases'
upnpd.config.igdv1='0'
upnpd.@perm_rule[0]=perm_rule
upnpd.@perm_rule[0].action='allow'
upnpd.@perm_rule[0].ext_ports='1024-65535'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1' 'wan2'

Ysurac, 2019-09-03:
Je ne vois pas de soucis.
Rien dans le "journal système" d'OpenMPTCProuter ?
Tout est vert dans le status ?
Le VPS et le routeur sont à l'heure ?

ivamp, 2019-09-03:
Merci de ton retour,
Donc dans les logs j'ai quelques déconnexion qui apparaissent:
`Tue Sep  3 22:01:31 2019 user.notice post-tracking-post-tracking: Get status and settings for wan2... Done
Tue Sep  3 22:01:42 2019 user.notice post-tracking-post-tracking: Set server vps (93.115.97.38) route via 192.168.3.254 metric 4
Tue Sep  3 22:02:23 2019 authpriv.info dropbear[3765]: Child connection from 61.177.172.158:20296
Tue Sep  3 22:02:36 2019 user.notice post-tracking-post-tracking: wan2 switched off because ping from 192.168.3.253 error (9.9.9.9,1.1.1.1,1.0.0.1)
Tue Sep  3 22:02:36 2019 user.notice post-tracking-post-tracking: Delete vps (93.115.97.38) route via 192.168.3.254 dev wan2
Tue Sep  3 22:02:46 2019 user.notice post-tracking-post-tracking: Get status and settings for wan1...
Tue Sep  3 22:02:46 2019 user.notice post-tracking-post-tracking: Multipath wan2 switched to on
Tue Sep  3 22:02:46 2019 user.notice post-tracking-post-tracking: Get status and settings for wan2...
Tue Sep  3 22:02:59 2019 user.notice post-tracking-post-tracking: Get status and settings for wan1... Done
Tue Sep  3 22:03:26 2019 user.notice post-tracking-post-tracking: Get status and settings for wan2... Done
Tue Sep  3 22:03:33 2019 user.notice post-tracking-post-tracking: Set server vps (93.115.97.38) route via 192.168.3.254 metric 4
Tue Sep  3 22:04:27 2019 authpriv.info dropbear[30934]: Exit before auth: Timeout before auth
Tue Sep  3 22:04:29 2019 user.notice post-tracking-post-tracking: wan2 switched off because ping from 192.168.3.253 error (80.67.169.12,80.67.169.40,4.2.2.1)
Tue Sep  3 22:04:29 2019 user.notice post-tracking-post-tracking: Delete vps (93.115.97.38) route via 192.168.3.254 dev wan2
Tue Sep  3 22:04:33 2019 user.notice post-tracking-post-tracking: Multipath wan2 switched to on
Tue Sep  3 22:04:33 2019 user.notice post-tracking-post-tracking: Get status and settings for wan2...
Tue Sep  3 22:05:10 2019 user.notice post-tracking-post-tracking: Get status and settings for wan2... Done`
Date et heure OK et routeur au vert
[image]
[image]

ivamp, 2019-09-03:
Merci de ton retour,
Donc dans les logs j'ai quelques déconnexion qui apparaissent:
`Tue Sep  3 22:01:31 2019 user.notice post-tracking-post-tracking: Get status and settings for wan2... Done
Tue Sep  3 22:01:42 2019 user.notice post-tracking-post-tracking: Set server vps (X.X.X.X) route via 192.168.3.254 metric 4
Tue Sep  3 22:02:23 2019 authpriv.info dropbear[3765]: Child connection from 61.177.172.158:20296
Tue Sep  3 22:02:36 2019 user.notice post-tracking-post-tracking: wan2 switched off because ping from 192.168.3.253 error (9.9.9.9,1.1.1.1,1.0.0.1)
Tue Sep  3 22:02:36 2019 user.notice post-tracking-post-tracking: Delete vps (X.X.X.X) route via 192.168.3.254 dev wan2
Tue Sep  3 22:02:46 2019 user.notice post-tracking-post-tracking: Get status and settings for wan1...
Tue Sep  3 22:02:46 2019 user.notice post-tracking-post-tracking: Multipath wan2 switched to on
Tue Sep  3 22:02:46 2019 user.notice post-tracking-post-tracking: Get status and settings for wan2...
Tue Sep  3 22:02:59 2019 user.notice post-tracking-post-tracking: Get status and settings for wan1... Done
Tue Sep  3 22:03:26 2019 user.notice post-tracking-post-tracking: Get status and settings for wan2... Done
Tue Sep  3 22:03:33 2019 user.notice post-tracking-post-tracking: Set server vps (X.X.X.X) route via 192.168.3.254 metric 4
Tue Sep  3 22:04:27 2019 authpriv.info dropbear[30934]: Exit before auth: Timeout before auth
Tue Sep  3 22:04:29 2019 user.notice post-tracking-post-tracking: wan2 switched off because ping from 192.168.3.253 error (80.67.169.12,80.67.169.40,4.2.2.1)
Tue Sep  3 22:04:29 2019 user.notice post-tracking-post-tracking: Delete vps (X.X.X.X) route via 192.168.3.254 dev wan2
Tue Sep  3 22:04:33 2019 user.notice post-tracking-post-tracking: Multipath wan2 switched to on
Tue Sep  3 22:04:33 2019 user.notice post-tracking-post-tracking: Get status and settings for wan2...
Tue Sep  3 22:05:10 2019 user.notice post-tracking-post-tracking: Get status and settings for wan2... Done`
Date et heure OK et routeur au vert
[image]
[image]

ivamp, 2019-09-05:
Bonjour,
L'agrégation fonctionne sans modification de ma part.
Donc impec pour le moment :)

ivamp, 2019-09-05:
Il m'arrive d'avoir des erreurs lorsque je charge des pages web depuis l'agrégation avec comme erreur PR_END_OF_FILE_ERROR
Il y a une option pour empêcher l'erreur ?

Ysurac, 2019-09-05:
Non, normalement c'est quand lors d'une connexion en https le fichier arrive pas en entier. Ca arrive sur un site particulier ou pas ? As-tu quelque chose de particulier dans le journal système à ce moment là ?

bsjroch, 2019-09-06:
J'ai le même problème ici.
J'ai 2 modules 4G usb branchés. Un sur orange et l'autre sur SFR. 
Quand je fais un speedtest ça utilise seulement un des modules. Et si je désactive le module utilisé alors il utilise l'autre mais jamais les 2 en même temps...

closed 2019-12-20 by Ysurac