Ysurac/openmptcprouter #641: Build error

JorgeMoort89, 2019-09-04:
Trying to build I receive this error:
`openmptcprouter/x86_64/source/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 1`
after ` make[3] -C package/system/urngd compile`
How can I fix it?

Ysurac, 2019-09-04:
Use ```./build.sh -j1 V=s``` or ```make -j1 V=s``` in the x86_64/source directory to get more log.

JorgeMoort89, 2019-09-05:
Done, I receive this error:
`go: honnef.co/go/tools@v0.0.0-20180728063816-88497007e858: git -c protocol.version=0 fetch --unshallow -f https://github.com/dominikh/go-tools refs/heads/*:refs/heads/* refs/tags/*:refs/tags/* in /home/jorge/openmptcprouter/x86_64/source/build_dir/target-x86_64_musl/v2ray-plugin-2e56b2b7/.go_work/build/pkg/mod/cache/vcs/115e616ee9610971a19aceed42604a0b4a01c5a45cee3ebdd0caf4410edc0615: exit status 128:
        fatal: unable to access 'https://github.com/dominikh/go-tools/': gnutls_handshake() failed: Error in the pull function.
go: error loading module requirements

Makefile:43: recipe for target '/home/jorge/openmptcprouter/x86_64/source/build_dir/target-x86_64_musl/v2ray-plugin-2e56b2b7/.built' failed
make[3]: *** [/home/jorge/openmptcprouter/x86_64/source/build_dir/target-x86_64_musl/v2ray-plugin-2e56b2b7/.built] Error 1
make[3]: Leaving directory '/home/jorge/openmptcprouter/feeds/openmptcprouter/shadowsocks-v2ray-plugin'
time: package/feeds/openmptcprouter/shadowsocks-v2ray-plugin/compile#20.43#8.52#311.16
package/Makefile:111: recipe for target 'package/feeds/openmptcprouter/shadowsocks-v2ray-plugin/compile' failed
make[2]: *** [package/feeds/openmptcprouter/shadowsocks-v2ray-plugin/compile] Error 2
make[2]: Leaving directory '/home/jorge/openmptcprouter/x86_64/source'
package/Makefile:107: recipe for target '/home/jorge/openmptcprouter/x86_64/source/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/jorge/openmptcprouter/x86_64/source/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/jorge/openmptcprouter/x86_64/source'
/home/jorge/openmptcprouter/x86_64/source/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
`

Ysurac, 2019-09-05:
Visiblement il arrive pas à télécharger une dépendance de v2ray-plugin lié à une erreur tls. C'est sous quelle distribution ?

JorgeMoort89, 2019-09-06:
Debian 9

JorgeMoort89, 2019-09-08:
Tried anothet time, no work, any news?

Ysurac, 2019-09-08:
no, it's working for me and on CircleCI, so this seems to be a problem on your config, nothing that I can do.

Ysurac, 2019-09-08:
You can try to git clone https://github.com/dominikh/go-tools/ and see if it's working or not. This seems to be a TLS problem.

JorgeMoort89, 2019-09-09:
How can I build it on CircleCI?

Ysurac, 2019-09-09:
Every commit is build on CircleCI and available.
For a custom build you need an account on CircleCI and configure it, but I don't think it's a good solution for only a custom build for a private usage.

JorgeMoort89, 2019-09-10:
I want to try and improve my skills on it, how can I setup it on CircleCI? I've already an account

Ysurac, 2019-09-10:
The configuration file is on ```openmptcprouter-feeds/.circle/```, then you need to read the doc of CircleCI.

JorgeMoort89, 2019-09-10:
`make -r package/compile: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
/home/circleci/openmptcprouter/x86_64/source/include/toplevel.mk:216: recipe for target 'package/compile' failed
make: *** [package/compile] Error 1
make: Leaving directory '/home/circleci/openmptcprouter/x86_64/source'
Exited with code 2`
Same problem on CircleCI

JorgeMoort89, 2019-09-10:
I've just forked your repo

JorgeMoort89, 2019-09-11:
Same problem on Ubuntu 16.04 server

JorgeMoort89, 2019-09-12:
Which language do you use to build?

Ysurac, 2019-09-12:
Language ?

JorgeMoort89, 2019-09-12:
[image]

Ysurac, 2019-09-12:
I would say other, even if I don't know where this screen come from.

JorgeMoort89, 2019-09-15:
Nothing, same error any way, how do you build the feeds exactly?

Ysurac, 2019-09-15:
I use ```.circleci/config.yml``` nothing more.

Ysurac, 2019-09-15:
You can see the result here: https://circleci.com/gh/Ysurac/openmptcprouter-feeds

JorgeMoort89, 2019-09-15:
Where I can find the built image?

Ysurac, 2019-09-15:
Here: https://www.openmptcprouter.com/release/4.19/
The link is at the end of the download page. But no support on these images, this can (and will) not work.

closed 2019-12-20 by Ysurac