Ysurac/openmptcprouter #642: Debian dependencies in  readme.md not complete

marcels2017, 2019-09-06:
Hi Yannick,

I just compiled the source code in a ubuntu 16.04 environment and I noticed if the package python-dev (or python3-dev for python3) isn't installed, it fails. Could you please add both packages to the prerequisites in the Readme.md?

Thanks,
Marcel

Ysurac, 2019-09-06:
Should work without python3-dev but with python3 installed. At least it's the case on CircleCI.

closed 2019-09-08 by Ysurac

marcels2017, 2019-09-09:
Hi,

Sorry for the late reply. I've started my first run from a freshly installed Ubuntu 16.04 (Python3 is already included in the base system) and it fails with this error message:

``
  CHK     include/generated/asm-offsets.h
  UPD     include/generated/asm-offsets.h
  SHIPPED scripts/dtc/pylibfdt/libfdt.i
  PYMOD   scripts/dtc/pylibfdt/_libfdt.so
scripts/dtc/pylibfdt/libfdt_wrap.c:147:21: fatal error: Python.h: No such file or directory
compilation terminated.
error: command 'gcc' failed with exit status 1
scripts/dtc/pylibfdt/Makefile:27: recipe for target 'scripts/dtc/pylibfdt/_libfdt.so' failed
make[7]: *** [scripts/dtc/pylibfdt/_libfdt.so] Error 1
scripts/Makefile.build:434: recipe for target 'scripts/dtc/pylibfdt' failed
make[6]: *** [scripts/dtc/pylibfdt] Error 2
scripts/Makefile.build:434: recipe for target 'scripts/dtc' failed
make[5]: *** [scripts/dtc] Error 2
Makefile:508: recipe for target 'scripts' failed
make[4]: *** [scripts] Error 2
make[4]: Leaving directory '/home/marcel/openmptcprouter/bpi-r2/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/u-boot-mt7623n_bpir2/u-boot-2019.01'
Makefile:58: recipe for target '/home/marcel/openmptcprouter/bpi-r2/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/u-boot-mt7623n_bpir2/u-boot-2019.01/.built' failed
make[3]: *** [/home/marcel/openmptcprouter/bpi-r2/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/u-boot-mt7623n_bpir2/u-boot-2019.01/.built] Error 2
make[3]: Leaving directory '/home/marcel/openmptcprouter/bpi-r2/source/package/boot/uboot-mediatek'
time: package/boot/uboot-mediatek/mt7623n_bpir2/compile#6.94#1.99#22.83
package/Makefile:111: recipe for target 'package/boot/uboot-mediatek/compile' failed
make[2]: *** [package/boot/uboot-mediatek/compile] Error 2
make[2]: Leaving directory '/home/marcel/openmptcprouter/bpi-r2/source'
package/Makefile:107: recipe for target '/home/marcel/openmptcprouter/bpi-r2/source/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/home/marcel/openmptcprouter/bpi-r2/source/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/marcel/openmptcprouter/bpi-r2/source'
/home/marcel/openmptcprouter/bpi-r2/source/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
``
With an installed python3-dev it went through.

Marcel

commit 819aabbf (2019-09-09): Fix #642