Ysurac/openmptcprouter #644: Aggregation does not seem to work

Ausstriken, 2019-09-07:
Hello,

I'm french guy so my english can be bad XD

Material:
- Box free (DELTA)
- 4G Router (TL-MR6400)
- WIFI Switch (TL-MR3420)

The RP3, Freebox, Router 4G are all in RJ45 on the wifi switch.

After carefully following the install documentation of RPI3, VPS, etc.
In system / openmptcprouter / status everything is "green".
but when I do a speedtest on my pc, i feel that i use only my wan1 (box free) which is 25 mb / s.
However when i do the test on nperf, i have the aggregation of my wan1 and wan2 = 80mb / s.
When I download a test file (1GO), my flow is 25mb / s ...

I also tested with my smarthphone (connected to the wifi switch) and the flow is equal to wan1 (box free) 25mb/s

It's been 3 weeks that I try to find the issue ...

Helps please :)

You will find attached all parameters
[Parameters.txt](https://github.com/Ysurac/openmptcprouter/files/3586795/Parameters.txt)

## Specifications

  - OpenMPTCProuter version: (v0.51.5)
  - OpenMPTCProuter VPS version: (0.995)
  - OpenMPTCProuter platform: (RP3)

Ysurac, 2019-09-07:
The Freebox give an IPv6 to all computer, so by default it's used in speed test that  support IPv6.
You should try to disable IPv6, should work by setting fe80::2 as next hop

Ausstriken, 2019-09-07:
Hello Ysurac,

Thank you! It work now!

closed 2019-09-08 by Ysurac