Ysurac/openmptcprouter #645: Closing Gateway with 4G Network Card Status Display

hljsykj, 2019-09-08:
[image]
Using 4G network card, the MAC address appears to be 0.
[image]
Status display gateway down

Ysurac, 2019-09-08:
Yes, because with QMI gateway seems to doesn't answer to ping.

hljsykj, 2019-09-08:
What causes this?

hljsykj, 2019-09-08:
It's normal to use this network card to test the network at present.

Ysurac, 2019-09-08:
I don't know if it's an issue with QMI or if it's GSM network that block ICMP to the gateway, but it's often the case but it's not a real issue for OpenMPTCProuter, this only display a warning.

hljsykj, 2019-09-08:
Aggregation does not work

Ysurac, 2019-09-08:
This is more an issue ;)
Else everything is green on the status page ?
MPTCP kernel is used on the VPS side ?
I would need the config (in System->OpenMPTCProuter->Show All settings, remove the VPS IP).
I also need the version of OpenMPTCProuter used, the version of the VPS script, the platform used.

Ysurac, 2019-09-08:
And the result of ```omr-test-speed```, ```omr-test-speed wan1``` (for each interface), via SSH on the router.

hljsykj, 2019-09-08:
[image]
Raspberry pie 3B+

hljsykj, 2019-09-08:
[image]
[image]
[image]

Ysurac, 2019-09-08:
I should say to use it at least 2 minutes

hljsykj, 2019-09-08:
[image]

Ysurac, 2019-09-08:
The best speed is with omr-test-speed, so I would say that it's working, even if it's still really slow... (it's in ko/s, so 113ko*8=>904kb/s = 0.9Mb/s)

hljsykj, 2019-09-08:
Yes, in the actual test, this PCI-E 4G network card does not use too much traffic, but the use of other usb4g network cards does not have this problem. I think it's a QMI protocol issue. Is there a better solution? Thank you very much.

closed 2019-09-10 by hljsykj