Ysurac/openmptcprouter #646: bpi-r2 build failed

bournehlm, 2019-09-08:
## Expected Behavior

  Latest master source build bpi-r2 failed, errors:
/home/ubuntu/openmptcprouter/bpi-r2/source/tmp/cc5je2Ow.o: In function `main':
extract-cert.c:(.text.startup+0x1e): undefined reference to `OPENSSL_add_all_algorithms_noconf'
extract-cert.c:(.text.startup+0x23): undefined reference to `ERR_load_crypto_strings'
collect2: error: ld returned 1 exit status
scripts/Makefile.host:90: recipe for target 'scripts/extract-cert' failed
make[6]: *** [scripts/extract-cert] Error 1
Makefile:1061: recipe for target 'scripts' failed
make[5]: *** [scripts] Error 2
make[5]: Leaving directory '/home/ubuntu/openmptcprouter/bpi-r2/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-mediatek_mt7623/linux-4.19.68'
Makefile:19: recipe for target '/home/ubuntu/openmptcprouter/bpi-r2/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-mediatek_mt7623/linux-4.19.68/.modules' failed
make[4]: *** [/home/ubuntu/openmptcprouter/bpi-r2/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-mediatek_mt7623/linux-4.19.68/.modules] Error 2
make[4]: Leaving directory '/home/ubuntu/openmptcprouter/bpi-r2/source/target/linux/mediatek'
Makefile:13: recipe for target 'compile' failed
make[3]: *** [compile] Error 2
make[3]: Leaving directory '/home/ubuntu/openmptcprouter/bpi-r2/source/target/linux'
time: target/linux/compile#21.30#6.34#19624.75
target/Makefile:23: recipe for target 'target/linux/compile' failed
make[2]: *** [target/linux/compile] Error 2
make[2]: Leaving directory '/home/ubuntu/openmptcprouter/bpi-r2/source'
target/Makefile:18: recipe for target '/home/ubuntu/openmptcprouter/bpi-r2/source/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.target_compile' failed
make[1]: *** [/home/ubuntu/openmptcprouter/bpi-r2/source/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/openmptcprouter/bpi-r2/source'
/home/ubuntu/openmptcprouter/bpi-r2/source/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

## Actual Behavior

## Steps to Reproduce the Problem

  1.git clone https://github.com/ysurac/openmptcprouter.git
  2.cd openmptcprouter
  3.OMR_TARGET="bpi-r2" ./build.sh -j1 V=s

## Specifications

  - OpenMPTCProuter version: (Latest master)
  - OpenMPTCProuter platform: (BPI-R2)

Ysurac, 2019-09-08:
I can only say that you have a problem with openssl

bournehlm, 2019-09-16:
Install openssl-1.0-dev fix that problem, thanks.

closed 2019-09-16 by bournehlm

Jasonlixiaojian, 2020-09-29:
How to install ubuntu, I will get an error when installing: [sudo] xiaojian's password:
Reading package list... Done
Analyzing the dependency tree of the package
Reading status information... Done
E: Unable to locate the package openssl-1.0-dev
E: Cannot find any software package according to glob ‘openssl-1.0-dev’
E: Unable to find any software package according to the regular expression openssl-1.0-dev