Ysurac/openmptcprouter #648: Maximum speed is not reached

ccmks, 2019-09-09:
## Expected Behavior

Getting the maximum speed during download, at least faster than single connection. I have 2 different ISP

ISP A - Fiber line with 20 Mbps down and 4 Mbps up
ISP B - LTE Router with LAN connect to WAN port of OpenMPTCPRouter with 7 Mbps down and 3 Mbps up

Should be getting faster than ISP A when aggregation happen

## Actual Behavior

Download slower than single ISP. When I download the file test from here: http://www.ovh.net/files/10Gio.dat

I only get approximately 15 Mbps of aggregation speed. I saw the aggregation happening on the bandwidth graph

However, when I try to connect to ISP A or B directly, I am getting maximum speed of downloading file respective ISP

Already test the VPS speed and the VPS can download the file from OVH with 100 Mbps with no issue

Also when visiting speedtest.net from client web browser, the speed only shows connection up to 5 Mbps download and 0.8 Mbps upload.

## Steps to Reproduce the Problem

 Use default configuration and connect to VPS using IP address and Key on OpenMPTCPRouter wizard

All ISP is using private IP which is passed through DHCP on each their respective router.

## Specifications

  - OpenMPTCProuter version:  openmptcprouter-v0.51.5-r010518-6d61fcfa9e
  - OpenMPTCProuter VPS version: openmptcprouter-v0.51.5-r010518-6d61fcfa9e (Tried using IndoVirtue and OVH-Singapore Public Cloud Instance Region)
  - OpenMPTCProuter platform: Hyper-V running under Windows 10 Pro Mini PC with bunch of USB network adapter plugged in

Marctraider, 2019-09-09:
Have you bench marked all USB Adapters individually and see if this could cause problem?

ccmks, 2019-09-09:
How can I benchmark?

Marctraider, 2019-09-09:
iPerf maybe? Any power saving stuff on those thumb devices? I know regular adapters usually have plenty options etc.

Ysurac, 2019-09-09:
```omr-test-speed``` to get aggregated speed, ```omr-test-speed wan1``` to get speed via wan1.

ccmks, 2019-09-11:
Finally found the temporary workaround. I had to disable the Shadowsocks encryption and just use non-encrypted to tunnel to VPS and the speed is good now! What is the best encryption for the minimum resource? My processor is not that fast, only Intel Atom X3-Z8350 which running OpenMPTCPRouter under Hyper-V instance with single core provisioned

Ysurac, 2019-09-11:
If the processor support AES hardware encryption then "aes-256-gcm" is  a good choice, else chacha20 should be the best choice.

xabix, 2019-09-16:
Hello, Not sure where you change this encryption but through the wizard I can't access : VPN -> Gloryvpn

[image]

Any idea of the issue and if this is were I can change the encryption to aes-256-gcm

MERCI

ccmks, 2019-09-18:
> It's disabled on current release, will be available again in next release but it doesn't a good result.

Hello,

I have upgraded the client and server site to v0.52.3 and 1.000.1 respectively. I saw the MLVPN option is there. I tried to disable the shadowsocks and enable the MLVPN to see if I can get better speed. However, in the status page, I only see the "VPN is not running" on the router. How can I fix it?

Thank you

closed 2019-12-20 by Ysurac