Ysurac/openmptcprouter #651: DNS lags

mycanaletto, 2019-09-10:
Bonjour,
Suite au passage sur le DNS sur OMR, je viens de m’apercevoir qu'il ne rafraîchit pas très bien.

J'ai fait des changement de DNS chez Gandi il y a 6 heures et OMR me revoit toujours les anciens enregistrements. Si j'ai bien compris on ne peut pas choisir son DNS et il s'appuie sur les roots ? Possibilité de faire un flush ?

  - OpenMPTCProuter version: 0.51.5
  - OpenMPTCProuter VPS version: 0.999.4.19.56
  - OpenMPTCProuter platform: VM ESX

Marctraider, 2019-09-15:
Can you PLEASE write in english so others can also understand whats going on?

mycanaletto, 2019-09-15:
> Can you PLEASE write in english so others can also understand whats going on?

Hello, Following the passage of my DNS on OMR (for the Bypass), I just realized that it does not refresh very well: I made DNS changes at Gandi 6 hours ago and OMR see me again always old records. If I understand correctly we can not choose his DNS and OMR he relies on the roots? Possibility of a flush?

https://translate.google.com

closed 2019-12-20 by Ysurac