Ysurac/openmptcprouter #653: OMR All data encrypted ?

ctrlx83, 2019-09-11:
Hi everyone,

I use OMR since two months and it works perfectly !

But I have a question, are the connections flows are encrypted ?
I want a real encryption similar as a private and encrypted VPN.

The idea is to create a dedicated seedbox behind OMR.

Thanks

## Specifications
OpenMPTCProuter version: v0.51.5
OpenMPTCProuter VPS version: 0.999 4.19.56-mptcp
OpenMPTCProuter platform: x86_64

Ysurac, 2019-09-11:
All connections are encrypted by default using chacha20 or aes-256-gcm if there is AES hardware support.

ctrlx83, 2019-09-11:
Hi Ysurac,

ok thanks for the answer, how I can verify if datas are correctly encrypted and which algorithm is running ?

client hardware are a old 'OVH Overthebox', hardware are normaly support AES encryption.

Ysurac, 2019-09-11:
In shadowsocks (distant server tab) and glorytun interface (by edit if chacha20 is not enabled it's AES).

ctrlx83, 2019-09-11:
The sss0 is not disabled, the encryption is chacha20

ctrlx83, 2019-09-11:
ok thanks
on shell I have :

root@OpenMPTCProuter:~# cat /proc/cpuinfo | grep aes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology tsc_reliable nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 cx16 xtpr pdcm sse4_1 sse4_2 movbe popcnt tsc_deadline_timer aes rdrand lahf_lm 3dnowprefetch epb pti tpr_shadow vnmi flexpriority ept vpid tsc_adjust smep erms dtherm arat

On Intel KB, the processor is compatible with aes : https://ark.intel.com/content/www/fr/fr/ark/products/78476/intel-atom-processor-e3815-512k-cache-1-46-ghz.html

Ysurac, 2019-09-11:
So you can enable it.

ctrlx83, 2019-09-11:
I did it.
And I don't see any difference on surfing.

That's all ?

Ysurac, 2019-09-11:
Yes, this can make changes with high internet speed, else not a lot.

ctrlx83, 2019-09-11:
Many websites doesn't work
Timeout...

Ysurac, 2019-09-11:
You should check status page.

ctrlx83, 2019-09-11:
Can't get public IP address from shadowsocks

Ysurac, 2019-09-11:
Valid again the wizard this should solve the issue.

ctrlx83, 2019-09-11:
Reboot router and wait 2 minutes after reboot had solve the issue...
I do speedtest and test navigation on many websites...

ctrlx83, 2019-09-11:
I can't go on https://proof.ovh.net or speedtestNet doesn't work
No error on status page

Ysurac, 2019-09-11:
Check system log.

ctrlx83, 2019-09-11:
I have many daemon.err /usr/bin/ss-redir[8616]: remote recv: Connection reset by peer

ctrlx83, 2019-09-11:
when I rollback to chacha20 and re active chacha20 on glorytun, after reboot all is working fine

Marctraider, 2019-09-14:
Sorry for hijack, but does WRT3200ACS support hardware AES?

xabix, 2019-09-18:
Hello,

I am trying this on 0.52.3 (enabled aes-256 and disable of chacha) but I see:

> Wed Sep 18 09:31:54 2019 daemon.info omr-tracker-ss: Shadowsocks is down (can't contact 1.1.1.1, 1.0.0.1, 212.27.48.10)
Wed Sep 18 09:31:54 2019 user.notice Shadowsocks: Rules DOWN

On the VPS, I get can't get public ip address from ShadowSocks it goes on and off

> Wed Sep 18 09:31:54 2019 user.notice Shadowsocks: Rules DOWN
Wed Sep 18 09:32:52 2019 user.notice post-tracking-post-tracking: Get status and settings for wan4... Done
Wed Sep 18 09:33:02 2019 user.notice post-tracking-post-tracking: Set server vps (149.91.88.x) route via 192.168.9.1 metric 5
Wed Sep 18 09:33:52 2019 daemon.info omr-tracker-ss: Shadowsocks is up (can contact 104.16.1.1)
Wed Sep 18 09:33:52 2019 daemon.info omr-tracker-ss: Reload Shadowsocks rules
Wed Sep 18 09:33:52 2019 user.notice Shadowsocks: Rules UP
Wed Sep 18 09:33:53 2019 user.notice Shadowsocks: Reload omr-bypass rules
Wed Sep 18 09:33:53 2019 user.notice omr-bypass: Starting OMR-ByPass...
Wed Sep 18 09:33:53 2019 user.notice omr-bypass: Reload dnsmasq...
Wed Sep 18 09:33:53 2019 daemon.info dnsmasq[19231]: read /etc/hosts - 4 addresses
Wed Sep 18 09:33:53 2019 daemon.info dnsmasq[19231]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Wed Sep 18 09:33:53 2019 daemon.info dnsmasq-dhcp[19231]: read /etc/ethers - 0 addresses
Wed Sep 18 09:33:53 2019 user.notice omr-bypass: OMR-ByPass is running

can it be linked to the wan4 (4G Orange) going up and down with No Server ping response after 1s?

Thanks
Xavier

closed 2019-12-20 by Ysurac