Ysurac/openmptcprouter #654: Déconnexion journalier (plus aucune connexion)

Slivikk-fr, 2019-09-12:
## Actual Behavior
Bonjour, j'ai l'impression d'avoir des déconnexion chaque jour je ne sais pas pourquoi les deux connexion fonctionnent (enfin une sur deux la seconde (box orange) veut pas ce connecté), j'ai une carte réseau spécialement pour OpenMPTCP, quand la connexion saute la carte est toujours connecté. Ça viendrais du routeur ou de la carte réseau ?

## Steps to Reproduce the Problem

  1. Attendre le jour suivant (en général)
  2. Plus de connexion
  3.

## Specifications

  - OpenMPTCProuter version: (0.51.3)
  - OpenMPTCProuter VPS version: (0.xxx (je me souviens plus de la version))
  - OpenMPTCProuter platform: (x86_64 on ESXI 6.7)

Ysurac, 2019-09-12:
Il faut voir les journaux systèmes.

Slivikk-fr, 2019-09-12:
Ou ça ? 😅

Ysurac, 2019-09-12:
Dans Etat->Journal Système

Slivikk-fr, 2019-09-14:
Le problème à l'air de s'être résolu tout seul merci quand même

closed 2019-09-14 by Slivikk-fr