Ysurac/openmptcprouter #660: Update to v 0.52.1

jerome-info, 2019-09-16:
## Expected Behavior

  In English please.
update failed 

## Actual Behavior
Image : openmptcprouter-v0.52.1-r011006-713561a10b-x86-64-combined-ext4-efi.img.gz
then Ecriture de l'image

 The uploaded image file does not contain a supported format. Make sure that you choose the generic image format for your platform. Select 'Force upgrade' to flash the image even if the image format check fails. Use only if you are sure that the firmware is correct and meant for your device!
Image
P

## Steps to Reproduce the Problem

  1. Backup config
  2. Upload new firmware
  3. Force Upgrade

## Specifications

  - OpenMPTCProuter version: (0.51.5)
  - OpenMPTCProuter VPS version: 
Version 0.1001 4.19.56-mptcp

  - OpenMPTCProuter platform: (x86_64)

Ysurac, 2019-09-17:
https://www.openmptcprouter.com/download -> "Update with EFI images is not working for now."

closed 2019-09-17 by Ysurac

jerome-info, 2019-09-17:
i tried all dowloads efi ext4 etc.... even with 0.52.2 not working yet

Ysurac, 2019-09-17:
There is no know issue without EFI. You tested to upgrade from a no EFI image to a no EFI image ?