Ysurac/openmptcprouter #661: DSVPN not reconnecting after disconnect (Apply certain stuff like tracker, or just in general)

Marctraider, 2019-09-17:
## Expected Behavior

DSVPN reobtaining connection after disconnect.

  In English please.

## Actual Behavior

DSVPN not reconnecting after disconnect (Apply certain stuff like tracker, or just in general)
Need to manually restart service, after 30 sec or so, it reconnects.

## Steps to Reproduce the Problem

  1. Have DSVPN running for UDP/ICMP data
  2. Apply something like tracker OMS page, make some changes
  3. DSVPN stops responding and never reconnects

## Specifications

  - OpenMPTCProuter version: 0.52.1
  - OpenMPTCProuter VPS version: 0.1001
  - OpenMPTCProuter platform: (WRT3200ACM)

Ysurac, 2019-09-17:
True, I have the same problem.

Ysurac, 2019-09-19:
Should be better in 0.52.3 (or not?)

Marctraider, 2019-09-19:
I have not tested since then. I need UDP solution like Glorytun UDP or MLVPN realistically for best efficiency :-)

I can try to test later today... Im not running any critical applications at the moment.

closed 2019-12-20 by Ysurac