Ysurac/openmptcprouter #664: MLVPN not working with wizard, need manual start with command.

Marctraider, 2019-09-18:
## Expected Behavior

MLVPN functioning properly through luci ui.

## Actual Behavior

No way this will work out of the box, need to do this for it to work:
/usr/sbin/mlvpn --config "/tmp/etc/mlvpn0.conf" --user nobody

Otherwise mlvpn process is never executed.

## Steps to Reproduce the Problem

  1. Change wizard UDP/ICMP vpn to MLVPN
  2. Observe in interface statis, OMRVPN is down, no data transmitted.

## Specifications

  - OpenMPTCProuter version: 0.52.2 
  - OpenMPTCProuter VPS version: 0.1001
  - OpenMPTCProuter platform: WRT3200ACM

Ysurac, 2019-09-18:
I'm not able to reproduce the issue.

Ysurac, 2019-09-18:
You can check in system log to see what is the problem.

Marctraider, 2019-09-18:
It just keeps looping the same thing over and over again. Luci shows:

Error: Unknown error (DEVICE_CLAIM_FAILED) and switches back and forth to Error: Network device is not present

Wed Sep 18 13:35:12 2019 user.notice omr-bypass: OMR-ByPass is running
Wed Sep 18 13:35:12 2019 daemon.info dnsmasq[15343]: started, version 2.80 cachesize 150
Wed Sep 18 13:35:12 2019 daemon.info dnsmasq[15343]: DNS service limited to local subnets
Wed Sep 18 13:35:12 2019 daemon.info dnsmasq[15343]: compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dmpfile
Wed Sep 18 13:35:12 2019 daemon.info dnsmasq-dhcp[15343]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Wed Sep 18 13:35:12 2019 daemon.info dnsmasq[15343]: using local addresses only for domain test
Wed Sep 18 13:35:12 2019 daemon.info dnsmasq[15343]: using local addresses only for domain onion
Wed Sep 18 13:35:12 2019 daemon.info dnsmasq[15343]: using local addresses only for domain localhost
Wed Sep 18 13:35:12 2019 daemon.info dnsmasq[15343]: using local addresses only for domain local
Wed Sep 18 13:35:12 2019 daemon.info dnsmasq[15343]: using local addresses only for domain invalid
Wed Sep 18 13:35:12 2019 daemon.info dnsmasq[15343]: using local addresses only for domain bind
Wed Sep 18 13:35:12 2019 daemon.info dnsmasq[15343]: using nameserver 127.0.0.1#5353
Wed Sep 18 13:35:12 2019 daemon.info dnsmasq[15343]: using local addresses only for domain lan
Wed Sep 18 13:35:12 2019 daemon.info dnsmasq[15343]: read /etc/hosts - 4 addresses
Wed Sep 18 13:35:12 2019 daemon.info dnsmasq[15343]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Wed Sep 18 13:35:12 2019 daemon.info dnsmasq-dhcp[15343]: read /etc/ethers - 0 addresses
Wed Sep 18 13:35:12 2019 daemon.info mlvpn[15378]: new password set
Wed Sep 18 13:35:12 2019 daemon.info mlvpn[15378]: 3 tunnel added
Wed Sep 18 13:35:12 2019 daemon.warn mlvpn[15378]: Enabling automatic bandwidth adjustment
Wed Sep 18 13:35:12 2019 daemon.info mlvpn[15378]: 4 tunnel added
Wed Sep 18 13:35:12 2019 daemon.warn mlvpn[15378]: Enabling automatic bandwidth adjustment
Wed Sep 18 13:35:12 2019 daemon.info mlvpn[15378]: 5 tunnel added
Wed Sep 18 13:35:12 2019 daemon.warn mlvpn[15378]: Enabling automatic bandwidth adjustment
Wed Sep 18 13:35:12 2019 daemon.notice netifd: Interface 'omrvpn' is enabled
Wed Sep 18 13:35:12 2019 daemon.notice netifd: Network device 'mlvpn0' link is up
Wed Sep 18 13:35:12 2019 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Wed Sep 18 13:35:12 2019 daemon.notice netifd: Interface 'omrvpn' is setting up now
Wed Sep 18 13:35:12 2019 daemon.info mlvpn[15378]: created interface `mlvpn0'
Wed Sep 18 13:35:12 2019 daemon.info mlvpn[15378]: 3 bind to 192.168.2.108
Wed Sep 18 13:35:12 2019 daemon.notice netifd: Network device 'mlvpn0' link is down
Wed Sep 18 13:35:12 2019 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Wed Sep 18 13:35:12 2019 daemon.notice netifd: Interface 'omrvpn' is now down
Wed Sep 18 13:35:12 2019 daemon.notice netifd: Interface 'omrvpn' is disabled
Wed Sep 18 13:35:12 2019 user.notice post-tracking-post-tracking: Can't find MLVPN, restart it...
Wed Sep 18 13:35:13 2019 daemon.info mlvpn[15807]: new password set
Wed Sep 18 13:35:13 2019 daemon.info mlvpn[15807]: 3 tunnel added
Wed Sep 18 13:35:13 2019 daemon.warn mlvpn[15807]: Enabling automatic bandwidth adjustment
Wed Sep 18 13:35:13 2019 daemon.info mlvpn[15807]: 4 tunnel added
Wed Sep 18 13:35:13 2019 daemon.warn mlvpn[15807]: Enabling automatic bandwidth adjustment
Wed Sep 18 13:35:13 2019 daemon.info mlvpn[15807]: 5 tunnel added
Wed Sep 18 13:35:13 2019 daemon.warn mlvpn[15807]: Enabling automatic bandwidth adjustment
Wed Sep 18 13:35:13 2019 daemon.info mlvpn[15807]: created interface `mlvpn0'
Wed Sep 18 13:35:13 2019 daemon.notice netifd: Interface 'omrvpn' is enabled
Wed Sep 18 13:35:13 2019 daemon.notice netifd: Network device 'mlvpn0' link is up
Wed Sep 18 13:35:13 2019 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Wed Sep 18 13:35:13 2019 daemon.notice netifd: Interface 'omrvpn' is setting up now
Wed Sep 18 13:35:13 2019 daemon.info mlvpn[15807]: 3 bind to 192.168.2.108
Wed Sep 18 13:35:13 2019 daemon.notice netifd: Network device 'mlvpn0' link is down
Wed Sep 18 13:35:13 2019 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Wed Sep 18 13:35:13 2019 daemon.notice netifd: omrvpn (15818): Command failed: Permission denied
Wed Sep 18 13:35:13 2019 daemon.notice netifd: Interface 'omrvpn' is now down
Wed Sep 18 13:35:13 2019 daemon.notice netifd: Interface 'omrvpn' is disabled
Wed Sep 18 13:35:13 2019 user.notice omr-bypass: Starting OMR-ByPass...
Wed Sep 18 13:35:15 2019 user.notice omr-bypass: Restart dnsmasq...
Wed Sep 18 13:35:15 2019 daemon.info dnsmasq[15343]: exiting on receipt of SIGTERM
Wed Sep 18 13:35:15 2019 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Sep 18 13:35:15 2019 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Sep 18 13:35:15 2019 user.notice dnsmasq: Allowing RFC1918 responses for domain plex.direct
Wed Sep 18 13:35:17 2019 user.notice post-tracking-post-tracking: Can't find MLVPN, restart it...
Wed Sep 18 13:35:17 2019 daemon.info mlvpn[17124]: new password set
Wed Sep 18 13:35:17 2019 daemon.info mlvpn[17124]: 3 tunnel added
[... cut ...]
Wed Sep 18 13:35:33 2019 daemon.notice netifd: omrvpn (21281): udhcpc: sending discover
Wed Sep 18 13:35:33 2019 daemon.notice netifd: omrvpn (21281): udhcpc: sendto: Network is down
Wed Sep 18 13:35:33 2019 daemon.notice netifd: omrvpn (21281): udhcpc: read error: Network is down, reopening socket
Wed Sep 18 13:35:34 2019 daemon.notice netifd: omrvpn (21281): udhcpc: bind: No such device
Wed Sep 18 13:35:34 2019 daemon.notice netifd: Interface 'omrvpn' is now down
Wed Sep 18 13:35:34 2019 daemon.notice netifd: Interface 'omrvpn' is disabled

ccmks, 2019-09-19:
I am second to this. I am not able to use MLVPN at all through wizard. Also applies on DSVPN. The status page keep showing VPN is not running

Ysurac, 2019-09-19:
Do you both use DHCP for wan interfaces ?

Marctraider, 2019-09-19:
> 
> 
> I am second to this. I am not able to use MLVPN at all through wizard. Also applies on DSVPN. The status page keep showing VPN is not running

Strange. DSVPN does work for me, it just won't reconnect in certain circumstances but thats issued here: https://github.com/Ysurac/openmptcprouter/issues/661

I think @Ysurac is working on MLVPN fix (hopefully) next commit.

Marctraider, 2019-09-19:
> 
> 
> Do you both use DHCP for wan interfaces ?

All 3 WAN connection are DHCP.

Ysurac, 2019-09-19:
Did you test on 0.52.3 ?

ccmks, 2019-09-19:
> Do you both use DHCP for wan interfaces ?

Yes, I use DHCP also on all my WAN interface

ccmks, 2019-09-19:
> Did you test on 0.52.3 ?

I am on that version

Marctraider, 2019-09-19:
Not entirely sure but pretty sure 0.52.3 was already out when you said on irc you might have found something ;-)

Ysurac, 2019-09-19:
True :)
So you can try with this modified init: https://github.com/Ysurac/openmptcprouter-feeds/raw/149c30129d2684d725cd368c915388810f962e58/mlvpn/files/etc/init.d/mlvpn
Or wait for next release, this will maybe fix the problem...

Marctraider, 2019-09-19:
Ill try in 30 mins or so...

Marctraider, 2019-09-19:
> 
> 
> True :)
> So you can try with this modified init: https://github.com/Ysurac/openmptcprouter-feeds/raw/149c30129d2684d725cd368c915388810f962e58/mlvpn/files/etc/init.d/mlvpn
> Or wait for next release, this will maybe fix the problem...

No change.

Thu Sep 19 20:41:34 2019 user.notice post-tracking-post-tracking: Get status and settings for wan3...
Thu Sep 19 20:41:34 2019 daemon.info mlvpn[7976]: new password set
Thu Sep 19 20:41:34 2019 daemon.info mlvpn[7976]: 3 tunnel added
Thu Sep 19 20:41:34 2019 daemon.warn mlvpn[7976]: Enabling automatic bandwidth adjustment
Thu Sep 19 20:41:34 2019 daemon.info mlvpn[7976]: 4 tunnel added
Thu Sep 19 20:41:34 2019 daemon.warn mlvpn[7976]: Enabling automatic bandwidth adjustment
Thu Sep 19 20:41:34 2019 daemon.info mlvpn[7976]: 5 tunnel added
Thu Sep 19 20:41:34 2019 daemon.warn mlvpn[7976]: Enabling automatic bandwidth adjustment
Thu Sep 19 20:41:34 2019 daemon.info mlvpn[7976]: created interface `mlvpn0'
Thu Sep 19 20:41:34 2019 daemon.notice netifd: Interface 'omrvpn' is enabled
Thu Sep 19 20:41:34 2019 daemon.notice netifd: Network device 'mlvpn0' link is up
Thu Sep 19 20:41:34 2019 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Thu Sep 19 20:41:34 2019 daemon.notice netifd: Interface 'omrvpn' is setting up now
Thu Sep 19 20:41:34 2019 daemon.info mlvpn[7976]: 3 bind to 192.168.2.108
Thu Sep 19 20:41:34 2019 daemon.notice netifd: Network device 'mlvpn0' link is down
Thu Sep 19 20:41:34 2019 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Thu Sep 19 20:41:34 2019 daemon.notice netifd: omrvpn (7990): Command failed: Permission denied
Thu Sep 19 20:41:34 2019 daemon.notice netifd: Interface 'omrvpn' is now down
Thu Sep 19 20:41:34 2019 daemon.notice netifd: Interface 'omrvpn' is disabled
Thu Sep 19 20:41:35 2019 daemon.info omr-tracker-ss: Shadowsocks is up (can contact 1.0.0.1)
Thu Sep 19 20:41:35 2019 user.notice omr-bypass: Starting OMR-ByPass...
Thu Sep 19 20:41:36 2019 user.notice post-tracking-post-tracking: omrvpn down
Thu Sep 19 20:41:36 2019 user.notice post-tracking-post-tracking: omrvpn down. Replace default route by 192.168.2.254 dev eth0.1
Thu Sep 19 20:41:36 2019 user.notice post-tracking-post-tracking: eth0.2 switched off
Thu Sep 19 20:41:36 2019 user.notice post-tracking-post-tracking: Delete vps (89.207.131.164) route via  dev eth0.2
Thu Sep 19 20:41:37 2019 user.notice omr-bypass: Restart dnsmasq...
Thu Sep 19 20:41:37 2019 daemon.info dnsmasq[4663]: exiting on receipt of SIGTERM
Thu Sep 19 20:41:37 2019 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Thu Sep 19 20:41:37 2019 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Thu Sep 19 20:41:37 2019 user.notice dnsmasq: Allowing RFC1918 responses for domain plex.direct
Thu Sep 19 20:41:38 2019 user.notice post-tracking-post-tracking: Can't find MLVPN, restart it...
Thu Sep 19 20:41:38 2019 daemon.info mlvpn[9493]: new password set
Thu Sep 19 20:41:38 2019 daemon.info mlvpn[9493]: 3 tunnel added
Thu Sep 19 20:41:38 2019 daemon.warn mlvpn[9493]: Enabling automatic bandwidth adjustment
Thu Sep 19 20:41:38 2019 daemon.info mlvpn[9493]: 4 tunnel added
Thu Sep 19 20:41:38 2019 daemon.warn mlvpn[9493]: Enabling automatic bandwidth adjustment
Thu Sep 19 20:41:38 2019 daemon.info mlvpn[9493]: 5 tunnel added
Thu Sep 19 20:41:38 2019 daemon.warn mlvpn[9493]: Enabling automatic bandwidth adjustment
Thu Sep 19 20:41:38 2019 daemon.notice netifd: Interface 'omrvpn' is enabled
Thu Sep 19 20:41:38 2019 daemon.notice netifd: Network device 'mlvpn0' link is up
Thu Sep 19 20:41:38 2019 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Thu Sep 19 20:41:38 2019 daemon.notice netifd: Interface 'omrvpn' is setting up now
Thu Sep 19 20:41:38 2019 daemon.info mlvpn[9493]: created interface `mlvpn0'
Thu Sep 19 20:41:38 2019 daemon.info mlvpn[9493]: 3 bind to 192.168.2.108
Thu Sep 19 20:41:38 2019 daemon.notice netifd: Network device 'mlvpn0' link is down
Thu Sep 19 20:41:38 2019 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Thu Sep 19 20:41:38 2019 daemon.notice netifd: Interface 'omrvpn' is now down
Thu Sep 19 20:41:38 2019 daemon.notice netifd: Interface 'omrvpn' is disabled
Thu Sep 19 20:41:39 2019 user.notice post-tracking-post-tracking: Get status and settings for wan1... Done
Thu Sep 19 20:41:40 2019 user.notice omr-bypass: OMR-ByPass is running
Thu Sep 19 20:41:40 2019 daemon.info dnsmasq[9803]: started, version 2.80 cachesize 150
Thu Sep 19 20:41:40 2019 daemon.info dnsmasq[9803]: DNS service limited to local subnets
Thu Sep 19 20:41:40 2019 daemon.info dnsmasq[9803]: compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Thu Sep 19 20:41:40 2019 daemon.info dnsmasq-dhcp[9803]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Thu Sep 19 20:41:40 2019 daemon.info dnsmasq[9803]: using local addresses only for domain test
Thu Sep 19 20:41:40 2019 daemon.info dnsmasq[9803]: using local addresses only for domain onion
Thu Sep 19 20:41:40 2019 daemon.info dnsmasq[9803]: using local addresses only for domain localhost
Thu Sep 19 20:41:40 2019 daemon.info dnsmasq[9803]: using local addresses only for domain local
Thu Sep 19 20:41:40 2019 daemon.info dnsmasq[9803]: using local addresses only for domain invalid
[... cut ...]
Thu Sep 19 20:42:27 2019 daemon.notice netifd: Interface 'omrvpn' is setting up now
Thu Sep 19 20:42:27 2019 daemon.info mlvpn[20322]: created interface `mlvpn0'
Thu Sep 19 20:42:27 2019 daemon.info mlvpn[20322]: 3 bind to 192.168.2.108
Thu Sep 19 20:42:27 2019 daemon.notice netifd: Network device 'mlvpn0' link is down
Thu Sep 19 20:42:27 2019 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Thu Sep 19 20:42:27 2019 daemon.notice netifd: Interface 'omrvpn' is now down
Thu Sep 19 20:42:28 2019 daemon.notice netifd: Interface 'omrvpn' is disabled

ccmks, 2019-09-20:
> /usr/sbin/mlvpn --config "/tmp/etc/mlvpn0.conf" --user nobody
How did you able to get it working manually? I tried to put the command above on the SSH console client router but still not seeing any MLVPN up and running.

Marctraider, 2019-09-20:
Do you see the process loaded in top?

I'm not even sure whether it works or aggregates properly though, never tested that far (probably not)

Ysurac, 2019-09-20:
Can you paste "/tmp/etc/mlvpn0.conf" ?

Marctraider, 2019-09-20:
> 
> 
> > /usr/sbin/mlvpn --config "/tmp/etc/mlvpn0.conf" --user nobody
> 
> How did you able to get it working manually? I tried to put the command above on the SSH console client router but still not seeing any MLVPN up and running.

When I do manual command, and wait til it settles down, assuming aggregation works,
when i one by one turn down WAN interface like this:

tc qdisc replace dev eth0.1 root netem loss 100%
or
tc qdisc replace dev eth0.2 root netem loss 100%
or
tc qdisc replace dev eth0.3 root netem loss 100%

UDP stream only stops when i disconnect WAN3 (eth0.3)
So it seems this way, mlvpn only uses wan3 and aggregation not work. 

Fri Sep 20 18:26:57 2019 daemon.info mlvpn[23165]: 4 tunnel added
Fri Sep 20 18:26:57 2019 daemon.warn mlvpn[23165]: Enabling automatic bandwidth adjustment
Fri Sep 20 18:26:57 2019 daemon.info mlvpn[23165]: 5 tunnel added
Fri Sep 20 18:26:57 2019 daemon.warn mlvpn[23165]: Enabling automatic bandwidth adjustment
Fri Sep 20 18:26:57 2019 daemon.crit mlvpn[23165]: failed to open /dev/net/tun read/write
Fri Sep 20 18:26:59 2019 daemon.notice netifd: omrvpn (21899): udhcpc: sending discover
Fri Sep 20 18:26:59 2019 daemon.notice netifd: omrvpn (21899): udhcpc: sending select for 10.255.253.29
Fri Sep 20 18:26:59 2019 user.notice post-tracking-post-tracking: Get status and settings for wan3... Done
Fri Sep 20 18:27:00 2019 daemon.info dnsmasq[23484]: started, version 2.80 cachesize 150
Fri Sep 20 18:27:00 2019 daemon.info dnsmasq[23484]: DNS service limited to local subnets
Fri Sep 20 18:27:00 2019 daemon.info dnsmasq[23484]: compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Fri Sep 20 18:27:00 2019 daemon.info dnsmasq-dhcp[23484]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Fri Sep 20 18:27:00 2019 daemon.info dnsmasq[23484]: using local addresses only for domain test
Fri Sep 20 18:27:00 2019 daemon.info dnsmasq[23484]: using local addresses only for domain onion
Fri Sep 20 18:27:00 2019 daemon.info dnsmasq[23484]: using local addresses only for domain localhost
Fri Sep 20 18:27:00 2019 daemon.info dnsmasq[23484]: using local addresses only for domain local
Fri Sep 20 18:27:00 2019 daemon.info dnsmasq[23484]: using local addresses only for domain invalid
Fri Sep 20 18:27:00 2019 daemon.info dnsmasq[23484]: using local addresses only for domain bind
Fri Sep 20 18:27:00 2019 daemon.info dnsmasq[23484]: using nameserver 127.0.0.1#5353
Fri Sep 20 18:27:00 2019 daemon.info dnsmasq[23484]: using local addresses only for domain lan
Fri Sep 20 18:27:00 2019 daemon.info dnsmasq[23484]: read /etc/hosts - 4 addresses
Fri Sep 20 18:27:00 2019 daemon.info dnsmasq[23484]: read /tmp/hosts/dhcp.cfg01411c - 3 addresses
Fri Sep 20 18:27:00 2019 daemon.info dnsmasq-dhcp[23484]: read /etc/ethers - 0 addresses
Fri Sep 20 18:27:00 2019 user.notice omr-bypass: OMR-ByPass is running
Fri Sep 20 18:27:02 2019 daemon.notice netifd: omrvpn (21899): udhcpc: sending select for 10.255.253.29
Fri Sep 20 18:27:02 2019 daemon.info mlvpn[23689]: new password set
Fri Sep 20 18:27:02 2019 daemon.info mlvpn[23689]: 3 tunnel added
Fri Sep 20 18:27:02 2019 daemon.warn mlvpn[23689]: Enabling automatic bandwidth adjustment
Fri Sep 20 18:27:02 2019 daemon.info mlvpn[23689]: 4 tunnel added
Fri Sep 20 18:27:02 2019 daemon.warn mlvpn[23689]: Enabling automatic bandwidth adjustment
Fri Sep 20 18:27:02 2019 daemon.info mlvpn[23689]: 5 tunnel added
Fri Sep 20 18:27:02 2019 daemon.warn mlvpn[23689]: Enabling automatic bandwidth adjustment
Fri Sep 20 18:27:02 2019 daemon.crit mlvpn[23689]: failed to open /dev/net/tun read/write
Fri Sep 20 18:27:05 2019 daemon.notice netifd: omrvpn (21899): udhcpc: sending select for 10.255.253.29
Fri Sep 20 18:27:05 2019 daemon.notice netifd: omrvpn (21899): udhcpc: lease of 10.255.253.29 obtained, lease time 43200
Fri Sep 20 18:27:05 2019 daemon.notice netifd: Interface 'omrvpn' is now up
Fri Sep 20 18:27:05 2019 user.notice firewall: Reloading firewall due to ifup of omrvpn (mlvpn0)
Fri Sep 20 18:27:05 2019 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (mlvpn0)
Fri Sep 20 18:27:06 2019 daemon.notice miniupnpd[6564]: shutting down MiniUPnPd
Fri Sep 20 18:27:06 2019 daemon.err miniupnpd[24042]: could not open lease file: /var/run/miniupnpd.leases
Fri Sep 20 18:27:06 2019 daemon.notice miniupnpd[24042]: HTTP listening on port 5000
Fri Sep 20 18:27:06 2019 daemon.warn miniupnpd[24042]: no HTTP IPv6 address, disabling IPv6
Fri Sep 20 18:27:06 2019 daemon.notice miniupnpd[24042]: Listening for NAT-PMP/PCP traffic on port 5351
Fri Sep 20 18:27:06 2019 daemon.err miniupnpd[24042]: PCPSendUnsolicitedAnnounce() IPv6 sendto(): Bad file descriptor
Fri Sep 20 18:27:07 2019 daemon.info mlvpn[24102]: new password set
Fri Sep 20 18:27:07 2019 daemon.info mlvpn[24102]: 3 tunnel added
Fri Sep 20 18:27:07 2019 daemon.warn mlvpn[24102]: Enabling automatic bandwidth adjustment
Fri Sep 20 18:27:07 2019 daemon.info mlvpn[24102]: 4 tunnel added
Fri Sep 20 18:27:07 2019 daemon.warn mlvpn[24102]: Enabling automatic bandwidth adjustment
Fri Sep 20 18:27:07 2019 daemon.info mlvpn[24102]: 5 tunnel added
Fri Sep 20 18:27:07 2019 daemon.warn mlvpn[24102]: Enabling automatic bandwidth adjustment
Fri Sep 20 18:27:07 2019 daemon.crit mlvpn[24102]: failed to open /dev/net/tun read/write
[... 30 lines cut ...]
[... cut ...]
Fri Sep 20 18:30:17 2019 daemon.err /usr/bin/ss-local[3700]: getpeername: Socket not connected
Fri Sep 20 18:30:30 2019 daemon.notice netifd: wan3 (4212): udhcpc: sending renew to 192.168.3.3
Fri Sep 20 18:30:30 2019 daemon.notice netifd: wan3 (4212): udhcpc: lease of 91.141.134.155 obtained, lease time 300
Fri Sep 20 18:30:32 2019 daemon.err /usr/bin/ss-local[3700]: getpeername: Socket not connected
Fri Sep 20 18:31:01 2019 daemon.info mlvpn[21868]: 3 timeout
Fri Sep 20 18:31:01 2019 daemon.info mlvpn[21868]: 3 authenticated

ccmks, 2019-09-20:
> /tmp/etc/mlvpn0.conf

[general]
tuntap = "tun"
mode = "client"
interface_name = "mlvpn0"
timeout = 30
reorder_buffer = yes
reorder_buffer_size = 64
loss_tolerence = 50
password = "omitted"

[3]
bindhost = "eth1"
bindport = "65202"
remotehost = "45.77.41.111"
remoteport = "65202"

[4]
bindhost = "eth2"
bindport = "65203"
remotehost = "45.77.41.111"
remoteport = "65203"

[5]
bindhost = "eth3"
bindport = "65204"
remotehost = "45.77.41.111"
remoteport = "65204"
~

Ysurac, 2019-09-21:
No problem in the config.
You can try this image: https://www.openmptcprouter.com/release/4.19/
Maybe this will fix the problem, not sure...

ccmks, 2019-09-21:
I managed to flash it to my Banana PI R2 router and test it. In the initial setup on the wizard, I went straight through disabling Shadowsocks and enable MLVPN. Setup DHCP on all my WAN.

The router can establish MLVPN and I can do some testing and see the aggregation happening. However, the speed is awfully slow! I only get 6 Mbps at aggregation speed from both LTE modem. My LTE 1 get 7 Mbps bandwidth and my LTE 2 get around 20 Mbps bandwidth, but the MLVPN aggregation is below than any of my single LTE speed.

I have setup the speed on each WAN based on my LTE potential maximum speed, just in case if you ask for it.

But what makes even worse, the MLVPN won't re-establish the tunnel once I switched to another VPN protocol and come back to MLVPN. It doesn't matter if I reboot the router, still no MLVPN tunnel

Here is the log in case you ask for it
Sat Sep 21 14:07:16 2019 daemon.info dnsmasq-dhcp[5889]: read /etc/ethers - 0 addresses
Sat Sep 21 14:07:16 2019 user.notice omr-tracker: Launching...
Sat Sep 21 14:07:16 2019 user.notice omr-tracker: Launched
Sat Sep 21 14:07:17 2019 user.notice post-tracking-post-tracking: lan2 switched off
Sat Sep 21 14:07:17 2019 user.notice post-tracking-post-tracking: Delete vps (51.79.142.145) route via  dev lan2
Sat Sep 21 14:07:17 2019 user.notice post-tracking-post-tracking: wan3 down. Replace default route by 192.168.8.1 dev lan0
Sat Sep 21 14:07:17 2019 user.err vnstat_backup: cannot restore, backup file does not exist (yet)
Sat Sep 21 14:07:17 2019 user.notice adb-enablemodem: INFO: waiting for device
Sat Sep 21 14:07:17 2019 user.notice omr-bypass: Starting OMR-ByPass...
Sat Sep 21 14:07:17 2019 daemon.notice procd: /etc/rc.d/S99omr-bypass: 0+5 records in
Sat Sep 21 14:07:17 2019 daemon.notice procd: /etc/rc.d/S99omr-bypass: 0+5 records out
Sat Sep 21 14:07:17 2019 user.notice Shadowsocks: Rules UP
Sat Sep 21 14:07:18 2019 daemon.info mlvpn[6567]: new password set
Sat Sep 21 14:07:18 2019 daemon.info mlvpn[6567]: 3 tunnel added
Sat Sep 21 14:07:18 2019 daemon.warn mlvpn[6567]: Enabling automatic bandwidth adjustment
Sat Sep 21 14:07:18 2019 daemon.info mlvpn[6567]: 4 tunnel added
Sat Sep 21 14:07:18 2019 daemon.warn mlvpn[6567]: Enabling automatic bandwidth adjustment
Sat Sep 21 14:07:18 2019 daemon.info mlvpn[6567]: 5 tunnel added
Sat Sep 21 14:07:18 2019 daemon.warn mlvpn[6567]: Enabling automatic bandwidth adjustment
Sat Sep 21 14:07:18 2019 daemon.info mlvpn[6567]: 6 tunnel added
Sat Sep 21 14:07:18 2019 daemon.warn mlvpn[6567]: Enabling automatic bandwidth adjustment
Sat Sep 21 14:07:18 2019 daemon.info mlvpn[6567]: created interface `mlvpn0'
Sat Sep 21 14:07:18 2019 daemon.notice netifd: Interface 'omrvpn' is enabled
Sat Sep 21 14:07:18 2019 daemon.notice netifd: Network device 'mlvpn0' link is up
Sat Sep 21 14:07:18 2019 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Sat Sep 21 14:07:18 2019 daemon.notice netifd: Interface 'omrvpn' is setting up now
Sat Sep 21 14:07:18 2019 daemon.info mlvpn[6567]: 3 bind to 192.168.8.100
Sat Sep 21 14:07:18 2019 daemon.info mlvpn[6567]: 5 bind to lan2
Sat Sep 21 14:07:18 2019 daemon.info mlvpn[6567]: 6 bind to lan3
Sat Sep 21 14:07:18 2019 daemon.info mlvpn[6567]: 4 bind to 192.168.40.100
Sat Sep 21 14:07:18 2019 daemon.notice netifd: Network device 'mlvpn0' link is down
Sat Sep 21 14:07:18 2019 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Sat Sep 21 14:07:18 2019 daemon.notice netifd: Interface 'omrvpn' is now down
Sat Sep 21 14:07:18 2019 daemon.notice netifd: Interface 'omrvpn' is disabled
Sat Sep 21 14:07:20 2019 user.notice post-tracking-post-tracking: omrvpn down
Sat Sep 21 14:07:20 2019 user.notice post-tracking-post-tracking: lan3 switched off
Sat Sep 21 14:07:20 2019 user.notice post-tracking-post-tracking: Delete vps (51.79.142.145) route via  dev lan3
Sat Sep 21 14:07:20 2019 user.notice post-tracking-post-tracking: omrvpn down. Replace default route by 192.168.8.1 dev lan0
Sat Sep 21 14:07:20 2019 daemon.err uhttpd[2165]: luci: accepted login on / for root from 192.168.100.200
Sat Sep 21 14:07:20 2019 user.notice omr-bypass: Restart dnsmasq...
Sat Sep 21 14:07:21 2019 daemon.info dnsmasq[5889]: exiting on receipt of SIGTERM
Sat Sep 21 14:07:21 2019 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sat Sep 21 14:07:21 2019 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sat Sep 21 14:07:21 2019 user.notice dnsmasq: Allowing RFC1918 responses for domain plex.direct
Sat Sep 21 14:07:21 2019 daemon.notice procd: /etc/rc.d/S99omr-bypass: udhcpc: started, v1.31.0
Sat Sep 21 14:07:21 2019 daemon.notice procd: /etc/rc.d/S99omr-bypass: udhcpc: sending discover
Sat Sep 21 14:07:22 2019 user.notice Shadowsocks: Rules UP
Sat Sep 21 14:07:23 2019 daemon.info mlvpn[7981]: new password set
Sat Sep 21 14:07:23 2019 daemon.info mlvpn[7981]: 3 tunnel added
Sat Sep 21 14:07:23 2019 daemon.warn mlvpn[7981]: Enabling automatic bandwidth adjustment
Sat Sep 21 14:07:23 2019 daemon.info mlvpn[7981]: 4 tunnel added
Sat Sep 21 14:07:23 2019 daemon.warn mlvpn[7981]: Enabling automatic bandwidth adjustment
Sat Sep 21 14:07:23 2019 daemon.info mlvpn[7981]: 5 tunnel added
Sat Sep 21 14:07:23 2019 daemon.warn mlvpn[7981]: Enabling automatic bandwidth adjustment
Sat Sep 21 14:07:23 2019 daemon.info mlvpn[7981]: 6 tunnel added
Sat Sep 21 14:07:23 2019 daemon.warn mlvpn[7981]: Enabling automatic bandwidth adjustment
[... cut ...]
Sat Sep 21 14:09:30 2019 daemon.info mlvpn[27481]: 5 bind to lan2
Sat Sep 21 14:09:30 2019 daemon.info mlvpn[27481]: 6 bind to lan3
Sat Sep 21 14:09:30 2019 daemon.info mlvpn[27481]: 4 bind to 192.168.40.100
Sat Sep 21 14:09:30 2019 daemon.notice netifd: Network device 'mlvpn0' link is down
Sat Sep 21 14:09:30 2019 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Sat Sep 21 14:09:30 2019 daemon.notice netifd: Interface 'omrvpn' is now down
Sat Sep 21 14:09:30 2019 daemon.notice netifd: Interface 'omrvpn' is disabled

Thank you

Marctraider, 2019-09-21:
I will try this tonight as well.

Did you disable packet reordering and encryption?

More curious tho, why do you need such high speed on udp? Or do you not use mptcp at all for tcp?

ccmks, 2019-09-21:
> I will try this tonight as well.
> 
> Did you disable packet reordering and encryption?

Where can I find the settings?

> More curious tho, why do you need such high speed on udp? Or do you not use mptcp at all for tcp?

Well, if you read the other thread, I opened recently the new thread that one of my ISP doesn't support MPTCP. I figured out that all TCP based VPN won't aggregate my connection so I would need UPD based VPN for all my traffic. If this were not the case, I wouldn't bother to setup MLVPN, just use Shadowsocks instead.

Marctraider, 2019-09-21:
Luci -> VPN -> MLVPN -> Packet Reordering to 0 packets. 

Try that. As for disabling decryption there is no GUI setting but it can still be done with the mlvpn conf file. Look up the mlvpn documentation for the option. I forgot what it was called.

Marctraider, 2019-09-21:
Btw this firmware give same result sadly. I dont know how u even got it working.

Ysurac, 2019-09-21:
Working for me without problem but I don't use DHCP.
I never get good speed result with MLVPN...

Marctraider, 2019-09-21:
One shouldnt use it for TCP tho imho but only use it for time sensitive apps, where UDP was made for anyway.

Anyhow, Im forced to use DHCP as the modems are all giving out DHCP address and cannot be configured.

ccmks, 2019-09-21:
I just give up the idea of using MLVPN for now as it doesn't give any good performance plus bad stability. I decided to use Glorytun UPD instead since it gives me better result but still a lot slower than shadowsocks

Marctraider, 2019-09-21:
Yea me too. Glorytun UDP remains best for my use case scenario. I really wish there was solution for Packet Out of order other than add latency delay to wan interfaces.. even then its not failsafe.

TCP is no solution for UDP as if it has any delay or loss, time critical encapsulated UDP stream could potentially 'hang'. Detrimental for gaming etc.

ccmks, 2019-09-21:
Any idea on how to use shadowsocks on ISP which doesn't support MPTCP?

Ysurac, 2019-09-21:
Using UDP for time critical usage is quite strange. UDP is made as "send and forget", some packet can never go to destination and this shouldn't be a problem.

ccmks, 2019-09-21:
>Curious why are u using Glorytun UDP for all traffic?

The reason why I had to do this because one of my WAN ISP doesn't support MPTCP which the aggregation won't happen with shadowsocks. That's why I had to use Glorytun UDP to aggregate all traffic. Glorytun TCP and other TCP based VPN gives same behavior as shadowsocks which does not allow me to aggregate.

Marctraider, 2019-09-21:
> 
> 
> Using UDP for time critical usage is quite strange. UDP is made as "send and forget", some packet can never go to destination and this shouldn't be a problem.

Ofcourse, this is no problem. If game server maybe not receive 0.5% of the totality of packets, no problem. This is also why they use UDP, to counter TCP lag and retransmission, and have constant packet stream going.

But with Glorytun UDP, and 3 WAN connections with very close RTT, you easily get loads of packet out of order deliveries, which can easily go up to 5-10% packet loss, at least that is what i'm observing. In that case, not so good.

Delaying a connection RTT artificially can be a solution, but using 'netem' I think that only counts for egress packets, not ingress, so one would probably need to use veth or ifb to also delay incoming packets artificially.

Ysurac, 2019-09-21:
It's why I don't put Glorytun UDP as default, I didn't get good results for now. But I think it will improve, there is a new release that I will test for next OpenMPTCProuter release.

Marctraider, 2019-09-21:
Ah, we will wait and see then :-)

At least it does seem stable... with UDP CS6 DSCP and layer_cake.

Any reason why cake isnt the default QoS now btw? It seems much better and less overhead!

Ysurac, 2019-09-21:
Because it's still the default in OpenWrt, but I can try and change that.

Marctraider, 2019-09-23:
Is there a way to cheat and set static ip on all? and route to dhcp?

Like an interface in front of it?

Or is there an old version of vps script and firmware where MLVPN was working?

Marctraider, 2019-09-23:
Update: I just tried two WAN connection both with static IP, still not operable. 

So Its not related to DHCP after all?

Ysurac, 2019-09-23:
Maybe something strange in your config. Can you paste it here without VPS IP/Key ? It's available in System->OpenMPTCProuter and "Show all settings" tab.

closed 2019-12-20 by Ysurac